nkolban/esp32-snippets #707: SampleClient.cpp from BLETests not connecting

dkselw, 2018-10-26:
The client program I am running is identical to SampleClient.cpp from BLETests, except I have changed the UUID's and MyClient only prints the value of the characteristic, it does not try to write a value back.

I am using esp-idf v3.2-dev-1250-g221eced06

I have changed the UUID's to the following:
```
static BLEUUID serviceUUID("7506a198-431e-4a2a-aa36-9f32193691af");
static BLEUUID    charUUID("11d85759-ff31-4da6-b536-b383a7b28467");
```

The last version of BLE code that I have compiled with the client program and managed to get to work was esp32-snippets f8d6a65b6a06a0a8a5778279842678ca610834e3 from 15 Oct.

If you look at the log file "SampleClient log - f8d6a6.txt", you can see the scan finds the device 30:ae:a4:1c:63:8e and then sees the service UUID 7506a198-431e-4a2a-aa36-9f32193691af

However if you look at the log file "SampleClient log - master.txt" you can see the scan finds the device 30:ae:a4:1c:63:8e and then reports the invalid service UUID 32193691-af07-1103-0a02-31
7473744209. Part of this UUID "32193691-af" actually appears at the end of the valid UUID 7506a198-431e-4a2a-aa36-9f32193691af

Log files
[SampleClient log - master.txt](https://github.com/nkolban/esp32-snippets/files/2518238/SampleClient.log.-.master.txt)
[SampleClient log - f8d6a6.txt](https://github.com/nkolban/esp32-snippets/files/2518263/SampleClient.log.-.f8d6a6.txt)

dkselw, 2018-10-26:
> I think the problem might be not with SampleClient but with esp32
> advertising code. Can you confirm with nRF connect that your esp32 is
> advertising serviceUUID that you are looking for?

Yes I can connect to the BLE server with the nRF app on my phone.
Also the client program connects fine when compiled with older versions of
esp32-snippets

chegewara, 2018-10-26:
I know you can connect with nRF, but when you scan for devices in nRF connect you can see serviceUUID for that device. What i am asking is: is esp32 advertising serviceUUID that you want or is advertising `32193691-af07-1103-0a02-317473744209`?

Most likely its snippets library broken, unlucky i dont have time to do some tests, i am only trying to get as many information as you can provide.

dkselw, 2018-10-26:
> I know you can connect with nRF, but when you scan for devices in nRF connect you can see serviceUUID for that device. What i am asking is: is esp32 advertising serviceUUID that you want or is [...]

nRF reports the correct service UUID 7506a198-431e-4a2a-aa36-9f32193691af

chegewara, 2018-10-26:
This is reason i am asking those questions:
https://github.com/nkolban/esp32-snippets/pull/691#pullrequestreview-164519242

But from your answer i can say this is not the reason this issue.

chegewara, 2018-10-26:
@dkselw Could you change this line:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertisedDevice.cpp#L291
and this line:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertisedDevice.cpp#L296

last argument to true?

dkselw, 2018-10-26:
I found the problem in parseAdvertisement https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertisedDevice.cpp#L231

I changed
```
		length = *payload;          // Retrieve the length of the record.
		payload++;                  // Skip to type
	while (!finished) {
```
To
```
	while (!finished) {
		length = *payload;          // Retrieve the length of the record.
		payload++;                  // Skip to type
```

closed 2018-11-13 by chegewara