nkolban/esp32-snippets #708: Semaphore waiting stuck in getservices()

lvronghai, 2018-10-26:
Hello,

I was trying ESP32 acts as BLE clients to three BLE sensors.
It worked nicely until I started the long period testing. Sometimes, it seems it will get stuck with semaphore waiting.
Here is debug log. I am new to esp32 and arduino and studying the code as well. 

> = 
ESP32>      Retrieve data from BLE Sensors,
D (334836) FreeRTOS: Semaphore taking: name: RegEvt (0x3ffe84cc), owner: <N/A> for connect
D (334836) FreeRTOS: Semaphore taken:  name: RegEvt (0x3ffe84cc), owner: connect
V (334843) FreeRTOS: >> wait: Semaphore waiting: name: RegEvt (0x3ffe84cc), owner: connect for connect
V (334844) FreeRTOS: Semaphore giving: name: RegEvt (0x3ffe84cc), owner: connect
V (334859) FreeRTOS: << wait: Semaphore released: name: RegEvt (0x3ffe84cc), owner: connect
D (334867) FreeRTOS: Semaphore taking: name: OpenEvt (0x3ffe8980), owner: <N/A> for connect
D (334875) FreeRTOS: Semaphore taken:  name: OpenEvt (0x3ffe8980), owner: connect
V (334883) FreeRTOS: >> wait: Semaphore waiting: name: OpenEvt (0x3ffe8980), owner: connect for connect
V (335659) FreeRTOS: Semaphore giving: name: RssiCmplEvt (0x3ffe8734), owner: <N/A>
V (335660) FreeRTOS: Semaphore giving: name: SearchCmplEvt (0x3ffe86d4), owner: <N/A>
V (335664) FreeRTOS: Semaphore giving: name: OpenEvt (0x3ffe8980), owner: connect
V (335671) FreeRTOS: << wait: Semaphore released: name: OpenEvt (0x3ffe8980), owner: connect
            - Connected to server c4:7c:8d:64:72:df
D (340684) FreeRTOS: Semaphore taking: name: SearchCmplEvt (0x3ffe86d4), owner: <N/A> for getServices
D (340684) FreeRTOS: Semaphore taken:  name: SearchCmplEvt (0x3ffe86d4), owner: getServices
V (340690) FreeRTOS: >> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffe86d4), owner: getServices for getServices

There are three sensors, it stucked in the first sensor getServices.
The code:
 // Connect to the remove BLE Server.
    pClient->connect(pAddress);
    Serial.print("            - Connected to server ");
    Serial.println(sensor_mac);
    delay(5000);
    //Obtain a reference to the service we are after in the remote BLE server.
    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      Serial.print("            Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      return false;
    }
    Serial.println("            - Found our service");
   Serial.print("            - disConnected to server ");
    pClient->disconnect();
    return true;
Yonghai

chegewara, 2018-10-26:
I think the problem is not that semaphore is blocking (i may be wrong), the problem is where in your code is getServices() function called.

lvronghai, 2018-10-26:
Hello,

Thank very much for fast response. I provide more detail.
I have esp32 + A6 GSM + three BLE sensors. Since esp32 will go deep sleep and power off A6.

1. So I run BLE scan only in loop first time, to get all the sensor address.

void loop() {
  loop_count++;
  if (!BLUETOOTH_OK) {
    Serial.println(" -- search BLE sensors");
    BLEDevice::init(""); 
    BLEScan* pBLEScan = BLEDevice::getScan();
    pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
    pBLEScan->setActiveScan(true);
    pBLEScan->start(BLESCANTIMEOUT);
    BLUETOOTH_OK=1; 
  }

2. in loop, it continues with all mobile network registration and mqtt server connection with AT command calls.
    if it is failed, new loop run will retry.

3. if everything is ok. the connectToServer is called.
    a6_serial_response();
    CIPSEND_OK=1;
    Serial.println(" -- ready for sending data");
    delay(5000);
    pClient  = BLEDevice::createClient();
    for (int i=0; i<deviceAddress.size();i++){
      Serial.println(" ");
      if (!connectToServer(deviceAddress[i])) {
        Serial.println("failed to connect to the server; there is nothin more we will do.");
      } else {
        MQTTpublish();
        delay(2000);
      }
    }
    delay(5000);
    delete pClient;
    delay(5000);

4. getServices is called in connectToServer().
bool connectToServer(BLEAddress pAddress) {
 // Connect to the remove BLE Server.
    pClient->connect(pAddress);
    Serial.print("            - Connected to server ");
    Serial.println(sensor_mac);
    delay(5000);

    //Obtain a reference to the service we are after in the remote BLE server.
    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      Serial.print("            Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      return false;
    }
    Serial.println("            - Found our service");

    Serial.print("            - disConnected to server ");
    pClient->disconnect();
    return true;
    
5.

chegewara, 2018-10-26:
4. looks good, its like in examples so all should work; 

Sorry, but its hard to say what can cause this issue. Ive been working on library improvements, also semaphores, but at the moment i have no time. Maybe someone else could help with this.

lvronghai, 2018-10-26:
>    1. looks good, its like in examples so all should work;

Hi,

OK, thanks. At least I can reproduce the problem. I try to get familiar
with code and see if I can find root cause.

Best regards!
Yonghai.

zaindroid, 2020-05-31:
Hi,  I am not able to get services from client at all.  I have tried all of the tree methods 
1-getServices()        
2-getService(const char* uuid) 
3- getService(BLEUUID uuid)

In all of the cases esp32 gets stuck and then after a some time on Disconnect() callback is called from clientCallback class.

Can you please guide me here, i am using the latest build on arduino 1.8.12 as per your instruction in https://github.com/nkolban/esp32-snippets.git

lightingghost, 2020-09-01:
Hi @lvronghai, I am having the same issue, did you manage to solve the problem?

edge7, 2020-10-25:
same problem here, but it happens randomly

JoFRT, 2021-09-26:
Hello,
I face to the same problem, but unfortunately for me the BLE_Client sketch stucks always in getService. 
Has anyone solved this issue since?
I hope that you can help me.
Thanks for reading

ilektron, 2021-11-17:
I'm experiencing this, too. Unfortunately, it only happens when connecting to the BLE Uart example for esp32. When I connect to an NRF chip using their UART code it doen't hang. It appears to me like this might have something to do with the fact that the client isn't advertising the 128 bit primary service UUID where the NRF chip is.

ilektron, 2021-11-18:
Whoops, it looks like because I was using a random address to connect instead of public, connect was failing and a disconnect was happening before the getServices function was called.

sylque, 2024-01-07:
Same problem here. First, the semaphore is stuck:
```
[ 16999][V][BLEClient.cpp:412] getService(): >> getService: uuid: 00006a6a-0000-1000-8000-00805f9b34fb
[ 17002][V][BLEClient.cpp:448] getServices(): >> getServices
[ 17003][V][BLEClient.cpp:73] clearServices(): >> clearServices
[ 17004][V][BLEClient.cpp:80] clearServices(): << clearServices
[ 17006][V][FreeRTOS.cpp:189] take(): Semaphore taking: name: SearchCmplEvt (0x3fcb3048), owner: <N/A> for getServices
[ 17009][V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: SearchCmplEvt (0x3fcb3048), owner: getServices
[ 17011][V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: SearchCmplEvt (0x3fcb3048), owner: getServices for getServices
```
Then, after about 5 seconds, something happens, and the application finally hangs:
```
[ 22904][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_DISCONNECT_EVT
[ 22909][V][BLEUtils.cpp:1530] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_DISCONNECT_EVT
[ 22913][V][BLEUtils.cpp:1606] dumpGattServerEvent(): [conn_id: 0, remote_bda: 39:c7:6a:df:ae:03]
[ 22917][V][BLEServer.cpp:144] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
[ 22922][V][BLEDevice.cpp:584] startAdvertising(): >> startAdvertising
[ 22924][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[ 22927][V][BLEAdvertising.cpp:187] start(): >> start: customAdvData: 0, customScanResponseData: 0
[ 22931][D][BLEAdvertising.cpp:199] start(): - advertising service: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
[ 22934][V][BLEAdvertising.cpp:252] start(): << start
[ 22936][V][BLEDevice.cpp:586] startAdvertising(): << startAdvertising
[ 22938][D][BLEServer.cpp:379] onDisconnect(): BLEServerCallbacks
[ 22939][D][BLEServer.cpp:380] onDisconnect(): BLEServerCallbacks
[ 22941][D][BLEServer.cpp:381] onDisconnect(): BLEServerCallbacks
[ 22943][V][BLECharacteristic.cpp:205] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
[ 22947][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: ConfEvt (0x3fcb1df0), owner: <N/A>
[ 22951][V][BLECharacteristic.cpp:465] handleGATTServerEvent(): << handleGATTServerEvent
[ 22955][V][BLECharacteristic.cpp:205] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
[ 22960][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: ConfEvt (0x3fcb26b8), owner: <N/A>
[ 22964][V][BLECharacteristic.cpp:465] handleGATTServerEvent(): << handleGATTServerEvent
[ 22968][V][BLEServer.cpp:281] handleGATTServerEvent(): << handleGATTServerEvent
[ 22971][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_DISCONNECT_EVT
[ 22977][V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: ESP_GATTC_DISCONNECT_EVT
[ 22981][V][BLEUtils.cpp:1326] dumpGattClientEvent(): [reason: ESP_GATT_CONN_TIMEOUT, conn_id: 0, remote_bda: 39:c7:6a:df:ae:03]
[ 22985][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_DISCONNECT_EVT
[ 22988][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: OpenEvt (0x3fcb2fe4), owner: <N/A>
[ 22990][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: RssiCmplEvt (0x3fcb30ac), owner: <N/A>
[ 22993][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: SearchCmplEvt (0x3fcb3048), owner: getServices
```
Any idea what causes this?