nkolban/esp32-snippets #709: More semaphor issues - WiFi stuck in taken in connectAP

wegunterjr, 2018-10-29:
I am sort of stuck. The first time I run though, I can do a wifi->scan() and get a list of available AP's.   Then I try to connect, but get stuck in the Semaphore.  Any thoughts?

I am using Platformio IDE, esp-idf 3.1, esp32-snippets v3.0.

Thoughts?
```
[0;31mE (57239) FreeRTOS: Semaphore NOT taken:  name: ConnectFinished (0x3ffde8c8), owner: connectAP␛[0m
D (57239) WiFi: esp_wifi_connect␛[0m
V (57239) FreeRTOS: Semaphore taking: name: ConnectFinished (0x3ffde8c8), owner: connectAP for connectAP␛[0m
I (57359) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (57359) wifi: state: init -> auth (b0)
I (58359) wifi: state: auth -> init (200)
I (58359) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
␛[0;31mE (62249) FreeRTOS: Semaphore NOT taken:  name: ConnectFinished (0x3ffde8c8), owner: connectAP␛[0m
D (62249) WiFi: esp_wifi_connect␛[0m
V (62249) FreeRTOS: Semaphore taking: name: ConnectFinished (0x3ffde8c8), owner: connectAP for connectAP␛[0m
I (62369) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (62369) wifi: state: init -> auth (b0)
I (63369) wifi: state: auth -> init (200)
I (63369) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
␛[0;31mE (67259) FreeRTOS: Semaphore NOT taken:  name: ConnectFinished (0x3ffde8c8), owner: connectAP␛[0m
D (67259) WiFi: esp_wifi_connect␛[0m
V (67259) FreeRTOS: Semaphore taking: name: ConnectFinished (0x3ffde8c8), owner: connectAP for connectAP␛[0m
I (67379) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (67379) wifi: state: init -> auth (b0)
I (68379) wifi: state: auth -> init (200)
I (68379) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
␛[0;31mE (72269) FreeRTOS: Semaphore NOT taken:  name: ConnectFinished (0x3ffde8c8), owner: connectAP␛[0m
D (72269) WiFi: esp_wifi_connect␛[0m
```

chegewara, 2018-11-15:
Hi
try to change this line:
https://github.com/nkolban/esp32-snippets/blob/v3.0/cpp_utils/WiFi.cpp#L197
qith:
`m_connectFinished.take(5000, "connectAP")`
Thanks

wegunterjr, 2018-11-17:
Made the suggested change and testing it now.

wegunterjr, 2018-11-18:
it seems that I get the semaphor problem after it actually connects to an AP. 
and then every 5 seconds, outputs the same line about the semaphor...
scratching my head....wondering what fix there might be...
working feverishly to fix this...
```
D (341968) WiFi: esp_wifi_connect␛[0m
V (341968) FreeRTOS: Semaphore taking: name: ConnectFinished (0x3ffdec98), owner: connectAP for connectAP␛[0m
␛[0;31mE (346978) FreeRTOS: Semaphore NOT taken:  name: ConnectFinished (0x3ffdec98), owner: connectAP␛[0m
D (346978) WiFi: esp_wifi_connect␛[0m
V (346978) FreeRTOS: Semaphore taking: name: ConnectFinished (0x3ffdec98), owner: connectAP for connectAP␛[0m
␛[0;31mE (351988) FreeRTOS: Semaphore NOT taken:  name: ConnectFinished (0x3ffdec98), owner: connectAP␛[0m
D (351988) WiFi: esp_wifi_connect␛[0m
V (351988) FreeRTOS: Semaphore taking: name: ConnectFinished (0x3ffdec98), owner: connectAP for connectAP␛[0m
␛[0;31mE (356998) FreeRTOS: Semaphore NOT taken:  name: ConnectFinished (0x3ffdec98), owner: connectAP␛[0m
D (356998) WiFi: esp_wifi_connect␛[0m
V (356998) FreeRTOS: Semaphore taking: name: ConnectFinished (0x3ffdec98), owner: connectAP for connectAP␛[0m
␛[0;31mE (362008) FreeRTOS: Semaphore NOT taken:  name: ConnectFinished (0x3ffdec98), owner: connectAP␛[0m
D (362008) WiFi: esp_wifi_connect␛[0m
V (362008) FreeRTOS: Semaphore taking: name: ConnectFinished (0x3ffdec98), owner: connectAP for connectAP␛[0m
␛[0;31mE (367018) FreeRTOS: Semaphore NOT taken:  name: ConnectFinished (0x3ffdec98), owner: connectAP␛[0m
D (367018) WiFi: esp_wifi_connect␛[0m
```

chegewara, 2018-11-18:
It is waiting on this line to be called:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/WiFi.cpp#L262

wegunterjr, 2018-11-18:
Hmmm...  right. makes sense, but do I need to do anything special to get that event to fire.  meaning, do i need to adjust the WiFiEventhandler I have in my code:

```
class  MyWiFiEventHandler: public WiFiEventHandler 
{
	esp_err_t staGotIp(system_event_sta_got_ip_t event_sta_got_ip) 
	{
		ESP_LOGI(LOG_TAG, "MyWiFiEventHandler(Class): staGotIp");
		// Do something
		return ESP_OK;
	}

};
```

chegewara, 2018-11-18:
@wegunterjr Could you add some logs to this event handler to confirm it is called at all:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/WiFi.cpp#L238
Thanks

wegunterjr, 2018-11-18:
I added logs to right before and after line 256...

```
ESP_LOGI(LOG_TAG, " I am in the wifi Eventhanlder of WiFicpp: ");
	// Invoke the event handler.
	esp_err_t rc;
	if (pWiFi->m_pWifiEventHandler != nullptr) {
		rc = pWiFi->m_pWifiEventHandler->getEventHandler()(pWiFi->m_pWifiEventHandler, event);
	} else {
		rc = ESP_OK;
	}

    ESP_LOGI(LOG_TAG, " I am in the wifi Eventhanlder of WiFicpp: ESP_OK %i ", ESP_OK);
	// If the event we received indicates that we now have an IP address or that a connection was disconnected then unlock the mutex that
	// indicates we are waiting for a connection complete.
	if (event->event_id == SYSTEM_EVENT_STA_GOT_IP || event->event_id == SYSTEM_EVENT_STA_DISCONNECTED) {
		 ESP_LOGI(LOG_TAG, " SYSTEM_EVENT_STA_GOT_IP :%i  ", event->event_id);
		if (event->event_id == SYSTEM_EVENT_STA_GOT_IP) {  // If we connected and have an IP, change the status to ESP_OK.  Otherwise, change it to the reason code.
			pWiFi->m_apConnectionStatus = ESP_OK;
		} else {
			pWiFi->m_apConnectionStatus = event->event_info.disconnected.reason;
		}
		pWiFi->m_connectFinished.give();
	}
```

```
V (23488) event: exit default callback␛[0m
␛[0;32mI (23488) WiFi:  I am in the wifi Eventhanlder of WiFicpp: ␛[0m
D (23488) WiFiEventHandler: >> getEventHandler()␛[0m
D (23498) WiFiEventHandler: << getEventHandler: 0x400dc688␛[0m
D (23508) WiFiEventHandler: >> eventHandler called: ctx=0x3ffd7924, event=0x3ffe0ea0␛[0m
D (23508) WiFiEventHandler: default staDisconnected␛[0m
␛[0;32mI (23518) WiFi:  I am in the wifi Eventhanlder of WiFicpp: ESP_OK 0 ␛[0m
␛[0;32mI (23518) WiFi:  SYSTEM_EVENT_STA_GOT_IP :5  ␛[0m
V (23528) FreeRTOS: Semaphore giving: name: ConnectFinished (0x3ffdf530), owner: connectAP␛[0m
V (23538) FreeRTOS: Semaphore taken:  name: ConnectFinished (0x3ffdf530), owner: <N/A>␛[0m
V (23548) FreeRTOS: Semaphore giving: name: ConnectFinished (0x3ffdf530), owner: <N/A>␛[0m
D (23548) WiFi: << connectAP␛[0m
␛[0;32mI (23558) IoTGrill:  WiFi Initialized: 2␛[0m
```

chegewara, 2018-11-18:
Looks like now it works?

wegunterjr, 2018-11-18:
It's not keeping the connection... I am gonna simplify my setup, make sure I am not overwriting something...

wegunterjr, 2018-11-18:
Not staying connected, I made sure I was just using the following to connect, and it is connecting and disconnection similar to what is shown above.  The 5 indicates it is disconnect.  If it was 7 it would indicate it has an IP address, but it isn't getting one.

```

WiFi *wifi;
wifi = new WiFi();
wifi->connectAP(tempSid,tempPwd);
```

chegewara, 2018-11-18:
Then try to print this to see the reason:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/WiFi.cpp#L260

wegunterjr, 2018-11-18:
Here is my complete log from when there is 

```

I (2453) wifi: n:9 0, o:1 0, ap:255 255, sta:9 0, prof:1
I (3423) wifi: state: init -> auth (b0)
I (3423) wifi: state: auth -> init (8a0)
I (3423) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
␛[0;31mE (6393) FreeRTOS: Semaphore NOT taken:  name: ConnectFinished (0x3ffde954), owner: connectAP␛[0m
D (6393) WiFi: esp_wifi_connect␛[0m
V (6393) FreeRTOS: Semaphore taking: name: ConnectFinished (0x3ffde954), owner: connectAP for connectAP␛[0m
I (6513) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (6513) wifi: state: init -> auth (b0)
I (7513) wifi: state: auth -> init (200)
I (7513) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (7513) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (7513) wifi: state: init -> auth (b0)
I (7523) wifi: state: auth -> assoc (0)
I (7533) wifi: state: assoc -> run (10)
I (7623) wifi: connected with Guest, channel 9
I (7633) wifi: pm start, type: 1

D (7633) event: SYSTEM_EVENT_STA_CONNECTED, ssid:Guest, ssid_len:15, bssid:bb:aa:bb:aa:bb:aa, channel:9, authmode:3␛[0m
V (7633) event: enter default callback␛[0m
V (7643) tcpip_adapter: check: local, if=0 fn=0x4013b818
␛[0m
V (7643) tcpip_adapter: call api in lwip: ret=0x0, give sem␛[0m
V (7653) tcpip_adapter: check: remote, if=0 fn=0x4013b818
␛[0m
V (7653) tcpip_adapter: check: local, if=0 fn=0x4013be14
␛[0m
D (7663) tcpip_adapter: dhcp client init ip/mask/gw to all-0␛[0m
D (7673) tcpip_adapter: if0 start ip lost tmr: enter␛[0m
D (7673) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffe3e70 interval=120 ip=0␛[0m
D (7683) tcpip_adapter: dhcp client start successfully␛[0m
V (7693) tcpip_adapter: call api in lwip: ret=0x0, give sem␛[0m
V (7693) tcpip_adapter: check: remote, if=0 fn=0x4013be14
␛[0m
V (7703) event: exit default callback␛[0m
␛[0;32mI (7703) WiFi:  I am in the wifi Eventhanlder of WiFicpp: ␛[0m
D (7713) WiFiEventHandler: >> getEventHandler()␛[0m
D (7713) WiFiEventHandler: << getEventHandler: 0x400db5c0␛[0m
D (7723) WiFiEventHandler: >> eventHandler called: ctx=0x3ffd7784, event=0x3ffe0ea0␛[0m
D (7723) WiFiEventHandler: default staConnected␛[0m
␛[0;32mI (7733) WiFi:  I am in the wifi Eventhanlder of WiFicpp: ESP_OK 0 ␛[0m
␛[0;31mE (11403) FreeRTOS: Semaphore NOT taken:  name: ConnectFinished (0x3ffde954), owner: connectAP␛[0m
D (11403) WiFi: esp_wifi_connect␛[0m
V (11403) FreeRTOS: Semaphore taking: name: ConnectFinished (0x3ffde954), owner: connectAP for connectAP␛[0m
I (11523) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (11523) wifi: state: run -> auth (b0)
I (12523) wifi: state: run -> init (200)
I (12523) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (12523) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (12523) wifi: state: init -> auth (b0)
I (13533) wifi: state: auth -> init (200)
I (13533) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (13533) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (13533) wifi: state: init -> auth (b0)
I (14543) wifi: state: auth -> init (200)
I (14543) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
D (14543) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:Guest, ssid_len:15, bssid:bb:aa:bb:aa:bb:aa, reason:2␛[0m
V (14543) event: enter default callback␛[0m
V (14553) tcpip_adapter: check: local, if=0 fn=0x4013b91c
␛[0m
D (14553) tcpip_adapter: if0 start ip lost tmr: enter␛[0m
D (14563) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x3ffe3e70 interval=120 ip=0␛[0m
V (14573) tcpip_adapter: call api in lwip: ret=0x0, give sem␛[0m
V (14573) tcpip_adapter: check: remote, if=0 fn=0x4013b91c
␛[0m
V (14583) event: exit default callback␛[0m
␛[0;32mI (14583) WiFi:  I am in the wifi Eventhanlder of WiFicpp: ␛[0m
D (14593) WiFiEventHandler: >> getEventHandler()␛[0m
D (14593) WiFiEventHandler: << getEventHandler: 0x400db5c0␛[0m
D (14603) WiFiEventHandler: >> eventHandler called: ctx=0x3ffd7784, event=0x3ffe0ea0␛[0m
D (14613) WiFiEventHandler: default staDisconnected␛[0m
␛[0;32mI (14613) WiFi:  I am in the wifi Eventhanlder of WiFicpp: ESP_OK 0 ␛[0m
␛[0;32mI (14623) WiFi:  SYSTEM_EVENT_STA_GOT_IP :5  ␛[0m
V (14633) FreeRTOS: Semaphore giving: name: ConnectFinished (0x3ffde954), owner: connectAP␛[0m
V (14633) FreeRTOS: Semaphore taken:  name: ConnectFinished (0x3ffde954), owner: connectAP␛[0m
V (14643) FreeRTOS: Semaphore giving: name: ConnectFinished (0x3ffde954), owner: connectAP␛[0m
D (14653) WiFi: << connectAP␛[0m
␛[0;32mI (14653) IoTGrill:  high WaterMark in wifi 33168␛[0m
␛[0;32mI (14663) subpub: WiFiAPRecord RAM left 61688␛[0m
␛[0;32mI (14663) subpub: After SID Scan and Compare RAM left 61688␛[0m
␛[0;32mI (14673) subpub: Setting up UART left 61688␛[0m
D (14673) Task: << runTask: taskName=WiFi Task␛[0m
```