nkolban/esp32-snippets #710: reading BLE control remote buttons on ESP32 when pressed.

MeddahAbdellah, 2018-10-30:
First of all, thank you to all the contributors for their work, it made using the ESP32 very easy.

I am currently trying to control an ESP from a remote controller https://www.ebay.com/itm/R1-Remote-Controller-Game-Mini-Wireless-Bluetooth-For-IOS-Android-PC-Gamepad-VR/162674671259 ,
I would like to be able to read the buttons pressed on the ESP32. I sent the screenshot from the NRF app that shows the services and the characteristics. Please if you could show me where to start that would be greatly appreciated.

Thanks in advance

chegewara, 2018-10-30:
From screenshots you sent me i can say it wont be easy. There is 4 input report characteristics. 
You have 2 options:
- learn hid and write driver that will parse report map, then notify to characteristic that you need
- reverse engineer gamepad and then register for notify (most likely by characteristic handle); this will require to find characteristic that you are interested (1 or 2) and check how many bytes are sent and what each byte means

MeddahAbdellah, 2018-10-30:
Thank you for the quick response, I have seen this issue: https://github.com/nkolban/esp32-snippets/issues/660 and it seems very similar to mine, I wish to find a quick solution as it is part of a bigger project. I have basic knowledge of BLE and GATT protocol, I also read the BLEHIDDevice.h but I didn't seem to be able to figure out a way to use it correctly. I like the first solution, what do I need to learn about HID and what does the driver do? maybe at the end of this post, we will be able to make a ready to use code for future users as it seems that this is a known protocol (my phone could read the device button's easily this should be applicable to the ESP32)

chegewara, 2018-11-15:
Yes, it is what you are looking for.

hansmbakker, 2018-12-04:
@MeddahAbdellah is your issue solved? If so, please close your issue to keep the issue list clean

cc @chegewara

closed 2018-12-09 by chegewara