nkolban/esp32-snippets #712: BLE Server stop to work after several connetions

mpbejo, 2018-11-07:
Hello everyone, first of all thanks for the work done.
This is my problem, I have two ESP32 one as BLE server and one as BLE client. The client connects to the server reads the value of a characteristic (always the same value for test) and disconnects.
This every 5 seconds, works for many times but then freezes. When this happens I tried to connect with my smartphone using nRF Connect but the server is not visible. I set up Arduino IDE in DEBUG and this is the log:

1 when everything works!

DevConn: 0
[D][BLEServer.cpp:342] startAdvertising(): >> startAdvertising
[D][BLEAdvertising.cpp:174] start(): >> start: customAdvData: 0, customScanResponseData: 0
[D][BLEAdvertising.cpp:194] start(): - no services advertised
[D][BLEAdvertising.cpp:231] start(): << start
[D][BLEServer.cpp:344] startAdvertising(): << startAdvertising
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1099] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1285] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[D][BLEUtils.cpp:1123] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEServer.cpp:342] startAdvertising(): >> startAdvertising
[D][BLEAdvertising.cpp:174] start(): >> start: customAdvData: 0, customScanResponseData: 0
[D][BLEAdvertising.cpp:194] start(): - no services advertised
[D][BLEAdvertising.cpp:231] start(): << start
[D][BLEServer.cpp:344] startAdvertising(): << startAdvertising
start advertising
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1099] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1285] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[D][BLEUtils.cpp:1123] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONNECT_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLEUtils.cpp:1707] dumpGattServerEvent(): [conn_id: 0, remote_bda: b4:e6:2d:85:72:27]
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
DevConn: 1
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEServer.cpp:342] startAdvertising(): >> startAdvertising
[D][BLEAdvertising.cpp:174] start(): >> start: customAdvData: 0, customScanResponseData: 0
[D][BLEAdvertising.cpp:194] start(): - no services advertised
[D][BLEAdvertising.cpp:231] start(): << start
[D][BLEServer.cpp:344] startAdvertising(): << startAdvertising
restart advertising after connection
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1099] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1285] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[D][BLEUtils.cpp:1123] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_READ_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_READ_EVT
[D][BLEUtils.cpp:1777] dumpGattServerEvent(): [conn_id: 0, trans_id: 1, bda: b4:e6:2d:85:72:27, handle: 0x2a, is_long: 0, need_rsp:1]
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_READ_EVT
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_READ_EVT
[D][BLECharacteristic.cpp:369] handleGATTServerEvent(): Sending a response (esp_ble_gatts_send_response)
----------------------------- RETRY N: 150 -----------------------------
[D][BLECharacteristic.cpp:664] setValue(): >> setValue: length=4, data=00000000, characteristic UUID=dc23888a-90fa-480a-bb1f-b7585881b249
[D][BLECharacteristic.cpp:671] setValue(): << setValue
[D][BLECharacteristic.cpp:413] handleGATTServerEvent():  - Data: length=4, data=00000000, offset=0
[... 155 lines cut ...]
        Serial.println("start advertising");
        //oldDeviceConnected = deviceConnected;
        PrevDevConn = DevConn;
        if (DevConn == 0) {
          deviceConnected = false;
          digitalWrite(LED_BUILTIN, LOW);
        }
        
    }
    // connecting
    
    if (PrevDevConn < DevConn) {
        // do stuff here on connecting
        PrevDevConn = DevConn;
        delay(500); // give the bluetooth stack the chance to get things ready
        pServer->startAdvertising();
        Serial.println("restart advertising after connection");
        digitalWrite(LED_BUILTIN, HIGH);     
    }    
}

I hope you can help me to understand where is the problem
Thank You 
Best Regards 
Marco

chegewara, 2018-11-07:
Can you check if there is no memory leaks anywhere? try this line of code in your loop:
`		heap_caps_print_heap_info(MALLOC_CAP_INTERNAL);`

It will print few lines every time when is called. I need only last few lines before code stuck.

mpbejo, 2018-11-07:
Hi chegewara,
thank You for your support.
I supposed is a memory leak but i did not how to check.

This is the status at the begin:
Heap summary for capabilities 0x00000800:
  At 0x3ffaff10 len 240 free 0 allocated 180 min_free 0
    largest_free_block 0 alloc_blocks 1 free_blocks 1 total_blocks 2
  At 0x3ffce350 len 72880 free 0 allocated 70220 min_free 0
    largest_free_block 0 alloc_blocks 163 free_blocks 2 total_blocks 165
  At 0x3ffe0440 len 15296 free 0 allocated 15040 min_free 0
    largest_free_block 0 alloc_blocks 14 free_blocks 0 total_blocks 14
  At 0x3ffe4350 len 113840 free 93760 allocated 18860 min_free 90784
    largest_free_block 92328 alloc_blocks 72 free_blocks 6 total_blocks 78
  At 0x40093784 len 51324 free 51276 allocated 0 min_free 51276
    largest_free_block 51276 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 145036 allocated 104300 min_free 142060 largest_free_block 92328
	
And this is at the end	
Heap summary for capabilities 0x00000800:
  At 0x3ffaff10 len 240 free 0 allocated 180 min_free 0
    largest_free_block 0 alloc_blocks 1 free_blocks 1 total_blocks 2
  At 0x3ffce350 len 72880 free 0 allocated 70220 min_free 0
    largest_free_block 0 alloc_blocks 163 free_blocks 2 total_blocks 165
  At 0x3ffe0440 len 15296 free 0 allocated 15040 min_free 0
    largest_free_block 0 alloc_blocks 14 free_blocks 0 total_blocks 14
  At 0x3ffe4350 len 113840 free 17128 allocated 79720 min_free 13520
    largest_free_block 2044 alloc_blocks 966 free_blocks 373 total_blocks 1339
  At 0x40093784 len 51324 free 51276 allocated 0 min_free 51276
    largest_free_block 51276 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 68404 allocated 165160 min_free 64796 largest_free_block 51276

I hope this help

chegewara, 2018-11-07:
It help a bit, we can confirm memory leak:
`free 145036 allocated 104300 min_free 142060 largest_free_block 92328`
vs
`free 68404 allocated 165160 min_free 64796 largest_free_block 51276`
Now is the hard part. Track where we have memory leak.
There is many ways how to do it, if i would do this i would start from very simple example:
- start advertising,
- stop advertising
- repeat.
Then i would change it to:
- start advertising,
- connect and disconnect,
- start advertising.
Next would be to add notification procedure.

mpbejo, 2018-11-07:
Ok, tomorrow I will be out office, next day I will start the checks.
I did it a quick check I tried the BLE wirte sample to see if there are memory leak ( not found)
The main difference is that the start advertising it is only in the setup.
I will check and I will tell You.
Thank You 
Best Regards 
Marco

mpbejo, 2018-11-14:
Hi chegewara,
I found that if I set on Arduino IDE Core Debug Level: DEBUG I have the memory leak.
If I set on NOTHING the memory allocated is released after the disconnection I tried several times.
Thank You for help

mpbejo, 2018-11-15:
Well I supposed my problem was DEBUG that do not release the memory (and this is one problem).
But I saw that after some hours the sketch stop to work, in the serial monitor i saw that this happen always after pServer->startAdvertising();
I also added a whatchdog as suggested to reset the board and restart: 
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Timer/WatchdogTimer/WatchdogTimer.ino
But nothing happens, any suggestion?
Thank You Best Regards
Marco

chegewara, 2018-11-15:
I am doing test now with esp-idf with startAdvertising -> wait 5 sec ->stopAdvertising loop. I dont see memory leak:
```
Heap summary for capabilities 0x00000005:
  At 0x3ffe0440 len 15072 free 15036 allocated 0 min_free 15036
    largest_free_block 15036 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 113804 allocated 0 min_free 113804
    largest_free_block 113804 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 128840 allocated 0 min_free 128840 largest_free_block 113804
W (15748) SampleServer: custom gap event handler, event: 6
W (20751) SampleServer: custom gap event handler, event: 17
W (20754) SampleServer: custom gap event handler, event: 6
Heap summary for capabilities 0x00000005:
  At 0x3ffe0440 len 15072 free 15036 allocated 0 min_free 15036
    largest_free_block 15036 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 113804 allocated 0 min_free 113804
    largest_free_block 113804 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 128840 allocated 0 min_free 128840 largest_free_block 113804
W (20793) SampleServer: custom gap event handler, event: 6
W (25796) SampleServer: custom gap event handler, event: 17
W (25799) SampleServer: custom gap event handler, event: 6
Heap summary for capabilities 0x00000005:
  At 0x3ffe0440 len 15072 free 15036 allocated 0 min_free 15036
    largest_free_block 15036 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 113804 allocated 0 min_free 113804
    largest_free_block 113804 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 128840 allocated 0 min_free 128840 largest_free_block 113804
W (25838) SampleServer: custom gap event handler, event: 6
W (30841) SampleServer: custom gap event handler, event: 17
W (30844) SampleServer: custom gap event handler, event: 6
Heap summary for capabilities 0x00000005:
  At 0x3ffe0440 len 15072 free 15036 allocated 0 min_free 15036
    largest_free_block 15036 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 113804 allocated 0 min_free 113804
    largest_free_block 113804 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 128840 allocated 0 min_free 128840 largest_free_block 113804
W (30883) SampleServer: custom gap event handler, event: 6
W (35886) SampleServer: custom gap event handler, event: 17
```
The fact is i am using modified BLE library, especially in comparison to arduino library. Its possible there is some memory leak in arduino library. I am also doing test with debug log level warning, but i will change it to debug and run for next few hours.

chegewara, 2018-11-15:
Over 3 hours and still no change. Im suspecting there is something else involved in this issue, not only start/stop advertising:
```
D (12843842) BLEDevice: get advertising
D (12843842) BLEAdvertising: >> stop
I (12843842) BT_APPL: BTA_DmBleBroadcast: start = 0

D (12843845) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT
D (12843851) BLEUtils: [status: 0]
D (12843855) BLEServer: BLEServer ... handling GAP event!
D (12843860) BLEDevice: get advertising
D (12843864) BLEAdvertising: gapEventHandler [event no: 17]
I (12843870) BLEAdvertising: STOP advertising
D (12843875) BLEAdvertising: >> start: customAdvData: 0, customScanResponseData: 0
D (12843882) BLEAdvertising: - advertising service: 00001234-0000-1000-8000-00805f9b34fb
D (12843890) BLEAdvertising: - advertising service: 4fafc202-0000-1000-8000-00805f9b34fb
D (12843899) BLEAdvertising: - advertising service: 4fafc222-1fb5-459e-8fcc-c5c9c331914b
D (12843907) BLEAdvertising: << start
W (12843910) SampleServer: custom gap event handler, event: 17
I (12843917) BT_APPL: BTA_DmSetBleAdvParamsAll: 32, 64

I (12843923) BT_APPL: adv_type = 0, addr_type_own = 3, chnl_map = 7, adv_fil_pol = 0

D (12843934) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (12843939) BLEUtils: [status: 0]
D (12843942) BLEServer: BLEServer ... handling GAP event!
D (12843948) BLEDevice: get advertising
D (12843952) BLEAdvertising: gapEventHandler [event no: 6]
W (12843957) SampleServer: custom gap event handler, event: 6
D (12843964) BLEAdvertising: << stop
Heap summary for capabilities 0x00000005:
  At 0x3ffe0440 len 15072 free 15036 allocated 0 min_free 15036
    largest_free_block 15036 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 113804 allocated 0 min_free 113804
    largest_free_block 113804 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 128840 allocated 0 min_free 128840 largest_free_block 113804
D (12844003) BLEDevice: >> startAdvertising
D (12844007) BLEDevice: get advertising
D (12844011) BLEAdvertising: >> start: customAdvData: 0, customScanResponseData: 0
D (12844019) BLEAdvertising: - advertising service: 00001234-0000-1000-8000-00805f9b34fb
D (12844027) BLEAdvertising: - advertising service: 4fafc202-0000-1000-8000-00805f9b34fb
D (12844035) BLEAdvertising: - advertising service: 4fafc222-1fb5-459e-8fcc-c5c9c331914b
I (12844043) BT_APPL: BTA_DmSetBleAdvParamsAll: 32, 64

I (12844049) BT_APPL: adv_type = 0, addr_type_own = 3, chnl_map = 7, adv_fil_pol = 0

D (12844061) BLEUtils: Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
D (12844065) BLEUtils: [status: 0]
D (12844068) BLEServer: BLEServer ... handling GAP event!
D (12844074) BLEDevice: get advertising
D (12844078) BLEAdvertising: gapEventHandler [event no: 6]
W (12844083) SampleServer: custom gap event handler, event: 6
D (12844090) BLEAdvertising: << start
D (12844094) BLEDevice: << startAdvertising
```

mpbejo, 2018-11-19:
Hi chegewara,
I think I found a workaround for my problem.
Doing some test I saw that when the startAdvertising(); does not work correctly my sketch is not hang, only the BLE functions are not working correctly, 
I can say this because serial monitor of the server continue to print information, but the client cannot connect via BLE (using address) and a scan cannot find the server.
For workaround  I set a timer and a counter on the BLE server if I don't have connection for a while I call 
stopAdvertising() and then startAdvertising() again, if the server continue to not receive connection  I reset the ESP32. 
With the stop and start the client start to connect again I don't see reset in the log
I hope this is useful.
Best Regards
Marco

chegewara, 2018-11-19:
@mpbejo Thanks for info, i am working currently on this library and i think i have faster and more stable code which should be compatible with arduino. I have last one thing to fix now and will update github repo.

closed 2018-12-29 by chegewara