nkolban/esp32-snippets #717: New branch with added

chegewara, 2018-11-14:
Hi, i am creating new branch in which i will add new features like multi connect and some others. Because all this code has not been fully tested it is not merged to master yet. There is also small possibility i have made some mistakes during copying code from my WIP repository or not all files has been updated. 

I would like to ask to test it and give me a feedback, so i can fix all issues in code.
Thanks and have a nice codding.

chegewara, 2018-11-22:
Here is beta version for arduino with all changes i made so far, it should works with arduino but can be very unstable in some situations:

[ESP32_ble_library.zip](https://github.com/nkolban/esp32-snippets/files/2606187/ESP32_ble_library.zip)

mpbejo, 2018-11-22:
Hi chegewara,
I tried this version, I do not have problem during compile, but when I call:
pRemoteCharacParam = pRemoteService->getCharacteristic(charUUID_Param);
I have this error
E (23847) boot: Assert failed in heap_caps_free, /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c:261 (heap != NULL && "free() target pointer is outside heap areas")
the same sketch works with Esp32 BLE standard library

mpbejo, 2018-11-22:
Now I changed the  core debug level to DEBUG to ha have more information and with this option the compile fails:
In file included from C:\Users\Marco\Documents\Arduino\libraries\ESP32_ble_library\BLECharacteristic.cpp:23:0:

C:\Users\Marco\Documents\Arduino\libraries\ESP32_ble_library\BLECharacteristic.cpp: In member function 'void BLECharacteristic::handleGATTServerEvent(esp_gatts_cb_event_t, esp_gatt_if_t, esp_ble_gatts_cb_param_t*)':

C:\Users\Marco\Documents\Arduino\libraries\ESP32_ble_library\BLECharacteristic.cpp:433:80: error: 'struct esp_ble_gatts_cb_param_t::gatts_conf_evt_param' has no member named 'handle'

    ESP_LOGD(LOG_TAG, "m_handle = %d, conf->handle = %d", m_handle, param->conf.handle);

                                                                                ^

C:\Users\Marco\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32/esp32-hal-log.h:87:72: note: in definition of macro 'log_d'

 #define log_d(format, ...) log_printf(ARDUHAL_LOG_FORMAT(D, format), ##__VA_ARGS__)

                                                                        ^

C:\Users\Marco\Documents\Arduino\libraries\ESP32_ble_library\BLECharacteristic.cpp:433:4: note: in expansion of macro 'ESP_LOGD'

    ESP_LOGD(LOG_TAG, "m_handle = %d, conf->handle = %d", m_handle, param->conf.handle);

    ^

exit status 1
Errore durante la compilazione per la scheda DOIT ESP32 DEVKIT V1.

I hope thi is useful
Marco

chegewara, 2018-11-22:
this you can just remove:
`ESP_LOGD(LOG_TAG, "m_handle = %d, conf->handle = %d", m_handle, param->conf.handle);`

the other one i have to investigate.

chegewara, 2018-11-24:
For compatibility purpose i will switch and do all changes in library based on esp-idf release/v3.1 branch. I will update library soon.

chegewara, 2018-11-27:
I pushed code with few useful changes i hope. There was many time problem with advertising 128 bit UUID along with long device name. I changed a bit BLEAdvertising to allow this. I was even able to advertise long name and 16, 32 and 128 bit UUIDs one of each at the same. This is procedure how to achieve this:
- there is function setScanResponse(bool),
- if this function is with parameter true, then name and TX power are advertised in scan response only and 128 bit UUID can be advertised in packet,
- we have also 2 additional functions that can remove another advertised data from payload, its setMinPreferred() and setMaxPreferred(); if parameter in one of those is set to 0 then it wont be in payload; those 2 functions are important for some other reason, they are used when we need compatibility with Apple devices (iPhone, iPad, mac);

We have now option to get raw payload from BLEAdvertisedDevice. Its 2 new functions to get raw data and to get length of raw data. Fixed parsing payload to parse also scan response.

BLEscan class: fixed memory leak, added few functions to control stored discovered devices. We can now retrieve address type from BLEAdvertisedDevice that allow us to connect with peer device that is having private address. This lead us to BLEClient class where we will use this info, but to make it easier for us there is overridden function connect() where we can pass BLEAdvertisedDevice.

I encourage to check all changes and have a nice codding time with as few bugs as possible.

mpbejo, 2018-11-28:
HI chegewara
After added the new library,( downloaded from github branch:master created a.zip file of the cpp_utils folder, and from Arduino IDE, #include library, Add library from .ZIP file)
I have the same error, here the log:

Forming a connection to 24:0a:c4:9c:ba:b2
 - Created client
 - Connecting to server
[D][BLEClient.cpp:100] connect(): >> connect(24:0a:c4:9c:ba:b2)
[I][BLEDevice.cpp:594] addPeerDevice(): add conn_id: 0, GATT role: client
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:578] updatePeerDevice(): update conn_id: 4, GATT role: client
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:133] connect(): << connect(), rc=1
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
 - Connected to server
[D][BLEClient.cpp:380] getService(): >> getService: uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b
[D][BLEClient.cpp:416] getServices(): >> getServices
[D][BLEClient.cpp:75] clearServices(): >> clearServices
[D][BLEClient.cpp:82] clearServices(): << clearServices
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLERemoteService.cpp:29] BLERemoteService(): >> BLERemoteService()
[D][BLERemoteService.cpp:37] BLERemoteService(): << BLERemoteService()
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLERemoteService.cpp:29] BLERemoteService(): >> BLERemoteService()
[D][BLERemoteService.cpp:37] BLERemoteService(): << BLERemoteService()
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLERemoteService.cpp:29] BLERemoteService(): >> BLERemoteService()
[D][BLERemoteService.cpp:37] BLERemoteService(): << BLERemoteService()
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:433] getServices(): << getServices
[D][BLEClient.cpp:393] getService(): << getService: found the service with uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b
 - Found our service
pRemoteCharacAddr
[D][BLERemoteService.cpp:165] retrieveCharacteristics(): >> getCharacteristics() for service: 4fafc201-1fb5-459e-8fcc-c5c9c331914b
[D][BLERemoteService.cpp:196] retrieveCharacteristics(): Found a characteristic: Handle: 42, UUID: beb5483e-36e1-4688-b7f5-ea07361b26a8
[D][BLERemoteCharacteristic.cpp:41] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 42 0x42, uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8
[D][BLERemoteCharacteristic.cpp:254] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: beb5483e-36e1-4688-b7f5-ea07361b26a8
[D][BLERemoteCharacteristic.cpp:284] retrieveDescriptors(): Found a descriptor: Handle: 43, UUID: 00002902-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:298] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[D][BLERemoteCharacteristic.cpp:49] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:196] retrieveCharacteristics(): Found a characteristic: Handle: 45, UUID: dc23888a-90fa-480a-bb1f-b7585881b249
[D][BLERemoteCharacteristic.cpp:41] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 45 0x45, uuid: dc23888a-90fa-480a-bb1f-b7585881b249
[D][BLERemoteCharacteristic.cpp:254] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: dc23888a-90fa-480a-bb1f-b7585881b249
[D][BLERemoteCharacteristic.cpp:284] retrieveDescriptors(): Found a descriptor: Handle: 46, UUID: 00002902-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:298] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[D][BLERemoteCharacteristic.cpp:49] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:196] retrieveCharacteristics(): Found a characteristic: Handle: 48, UUID: db7ee806-20e8-4a7b-ba95-993e6b1651f7
[D][BLERemoteCharacteristic.cpp:41] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 48 0x48, uuid: db7ee806-20e8-4a7b-ba95-993e6b1651f7
[D][BLERemoteCharacteristic.cpp:254] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: db7ee806-20e8-4a7b-ba95-993e6b1651f7
[D][BLERemoteCharacteristic.cpp:284] retrieveDescriptors(): Found a descriptor: Handle: 49, UUID: 00002902-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:298] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[... cut ...]
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
E (60634) boot: Assert failed in heap_caps_free, /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c:261 (heap != NULL && "free() target pointer is outside heap areas")
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: Btc_task
CPU 1: IDLE

chegewara, 2018-11-29:
This is example what we can do now with advertising in library. Very long name and 3 services UUID, one of each type (16, 32 and 128 bits):
[image]

closed 2021-04-29 by chegewara