nkolban/esp32-snippets #718: ESP32 lib or nkolban ESP32

czuvich, 2018-11-15:
I’ve been working on a BLE WiFi project on the ESP32 for a few months using Arduino.

I’ve noticed the official ESP32 BLE library is very different from this one.  Which one should I be using?  Thanks!

chegewara, 2018-11-15:
ESP32 BLE library for arduino has not been updated for few months. You can try this:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#replacing-the-version-that-comes-with-arduino-esp32

Most likely it will works, but there is always small chance that some API has been implemented that will cause compilation errors because arduino-esp32 is not updated to latest esp-idf.

chegewara, 2018-11-19:
@czuvich I just pushed changes to features branch, i didnt test it very extensively but BLE_uart is working without issues. I was able to connect **9** client apps.

czuvich, 2018-11-20:
Excellent!  I'm a little green with github.  Should I follow the instructions in your link to grab your changes?  Or did you update the ESP-IDF Arduino BLE branch?

chegewara, 2018-11-20:
I will prepare zip file with library to replace it in arduino. But i have to say it may be not stable yet even if i performed quick tests.

czuvich, 2018-11-20:
I'm ok waiting until a stable release can be published.  So far, I've only run into the free RAM issue which is directly related to IDF.  Are you the only person testing the changes before publishing them?

chegewara, 2018-11-29:
Yes, I am the only one. I created external repo that anyone could test my changes before I merge it, but not many people are willing to help.
Now all changes I made so far are merged and I will fix all introduced and reported bugs.