nkolban/esp32-snippets #719: u8g2_esp32_hal

billcollis, 2018-11-15:


kifantidis, 2019-01-09:
Ive got the same problem. And I've noticed it today. I can't figure exactly when the problem occurs but with backtrace I found that it starts after using u8g2_SendBuffer.

gaw123, 2019-05-12:
I have the same issue

When I use in the same program DS3231 library from UncleRus https://github.com/UncleRus/esp-idf-lib and the u8g2 from nkolban im getting the same error.

For me it looks like conflict of two i2c libraries: 
"i2cdev.h" from UncleRus 
and 
"driver/i2c.h" from espidf team

But I can be wrong... no solution till now

denruss, 2019-06-23:
I have the same issue

no solution till now

Devenda, 2019-11-10:
I had the same issue before, sometimes it worked, most of the time it didn't.
Turned out I had an issue with the SCL line, so I2C communication could not be established.

This returns an error which is trying to be checked by the above code, resulting in a hard error and a reset. 

After fixing the SCL issue the above did not happen any more.

Have you been able to fix the issue?

arturfreitas, 2020-07-03:
Hi all,
I'm having the same error also. Was not able to fix it yet.
How did you solve the issue?

arturfreitas, 2020-07-03:
Hi all,
I'm having the same issue.
How did you solve it?

arturfreitas, 2020-07-03:
Hi all,
I'm having the same issue.
How did you solve it?

Thanks

arturfreitas, 2020-07-04:
Hi all, I was able to solve the issue at my end by replacing the function **'u8g2_Setup_ssd1306_128x32_noname_f**' with **'u8g2_Setup_ssd1306_i2c_128x64_noname_f'** and **setting the reset pin** along with SDA_PIN and SCL_PIN as shown below: 

`#define RST_PIN GPIO_NUM_16
u8g2_esp32_hal.sda   = SDA_PIN;
u8g2_esp32_hal.scl  = SCL_PIN ;
**u8g2_esp32_hal.reset = RST_PIN;**
u8g2_esp32_hal_init(u8g2_esp32_hal);`

**Please note:** I'm using Heltec esp32 board.
**For your reference:** https://www.gitmemory.com/issue/nkolban/esp32-snippets/872/597721315

Regards,
Artur

arturfreitas, 2020-07-04:
Hi all, I was able to solve the issue at my end by replacing the function 'u8g2_Setup_ssd1306_128x32_noname_f' with 'u8g2_Setup_ssd1306_i2c_128x64_noname_f' and setting the reset pin along with SDA_PIN and SCL_PIN as shown below:

```
#define RST_PIN GPIO_NUM_16 
[...]
u8g2_esp32_hal.sda = SDA_PIN;
u8g2_esp32_hal.scl = SCL_PIN ; 
u8g2_esp32_hal.reset = RST_PIN; 
u8g2_esp32_hal_init(u8g2_esp32_hal);
```

**Please note:** I'm using Heltec esp32 board.
**For your reference:** https://www.gitmemory.com/issue/nkolban/esp32-snippets/872/597721315

Regards,
Artur

arturfreitas, 2020-07-04:
Hi all, I was able to solve the issue at my end by replacing the function 'u8g2_Setup_ssd1306_128x32_noname_f' with 'u8g2_Setup_ssd1306_i2c_128x64_noname_f' and setting the reset pin along with SDA_PIN and SCL_PIN as shown below:

```
#define RST_PIN GPIO_NUM_16 
[...]
void task_test_SSD1306i2c(void *ignore) {
	u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	u8g2_esp32_hal.sda   = SDA_PIN;
	u8g2_esp32_hal.scl  = SCL_PIN ;
	**u8g2_esp32_hal.reset = RST_PIN;**
	u8g2_esp32_hal_init(u8g2_esp32_hal);

	u8g2_t u8g2; // a structure which will contain all the data for one display
	**u8g2_Setup_ssd1306_i2c_128x64_noname_f**(
		&u8g2,
		U8G2_R0,
		//u8x8_byte_sw_i2c,
		u8g2_esp32_i2c_byte_cb,
		u8g2_esp32_gpio_and_delay_cb);  // init u8g2 structure
	u8x8_SetI2CAddress(&u8g2.u8x8,0x78);
[...]
}
```
Please note: I'm using Heltec esp32 board.
For your reference: https://www.gitmemory.com/issue/nkolban/esp32-snippets/872/597721315

Regards,
Artur