nkolban/esp32-snippets #721: BLEAdvertisedDevice getPayload() how do I get total_len of payload

saknarak, 2018-11-19:
I want raw payload to parse it myself.
I found there is no total_len in BLEAdvertisedDevice

```
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    uint8_t *payload = advertisedDevice.getPayload();
    // I can't get total_len so I can't loop,
    // or may be I should limit total_len to 62?
  }
}
```

chegewara, 2018-11-19:
You can use total_len = 62 but raw payload usually is not as long and you also cant check against 0x00 value because some parameters can have value 0x00.
I will add this soon.

saknarak, 2018-11-19:
Thank you very much!

chegewara, 2018-11-27:
Check last commit please.