nkolban/esp32-snippets #723: Monitoring error: "BT_HCI: hci write adv params error"

jamesgz, 2018-11-21:
Hey all,

I am trying to use nkolban's SampleRead.cpp on esp-idf-template on Ubuntu.

I have done the following:
1. Cloned esp-idf-template to "ble-test" folder
2. Created components folder in "ble-test" and copied cpp_utils to components folder
3. Edited and copied main.cpp & SampleRead.cpp into main folder
4. deleted main.c
5. modified set(COMPONENT_SRCS "main.c") to set(COMPONENT_SRCS "main.cpp") in CMakeList.txt in main folder
6. ran make menuconfig and enabled bluetooth & c++ exceptions

There was an error while compiling. I noticed the extra brackets at line 99 & 100 in BLEUUID.cpp and deleted them. Then I was able to compile successfully.

I used nRF on my android phone and was able to connect to my ESP32 board and see that the time increases every time I read on my phone. 

Doing 'make monitor' returns the following error:

```
Toolchain path: /home/james/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /home/james/esp/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6164
load:0x40078000,len:10168
load:0x40080400,len:6552
entry 0x40080764
I (28) boot: ESP-IDF v3.2-beta1-151-g960c24057 2nd stage bootloader
I (28) boot: compile time 17:11:40
I (28) boot: Enabling RNG early entropy source...
I (34) boot: SPI Speed      : 40MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3c5f4 (247284) map
I (180) esp_image: segment 1: paddr=0x0004c61c vaddr=0x3ff80000 size=0x00000 (     0) load
I (180) esp_image: segment 2: paddr=0x0004c624 vaddr=0x3ff80000 size=0x00000 (     0) load
I (186) esp_image: segment 3: paddr=0x0004c62c vaddr=0x3ffbdb60 size=0x02dd0 ( 11728) load
I (200) esp_image: segment 4: paddr=0x0004f404 vaddr=0x3ffc0930 size=0x00000 (     0) load
I (204) esp_image: segment 5: paddr=0x0004f40c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/james/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (213) esp_image: segment 6: paddr=0x0004f814 vaddr=0x40080400 size=0x007fc (  2044) load
I (223) esp_image: segment 7: paddr=0x00050018 vaddr=0x400d0018 size=0xa44d0 (672976) map
0x400d0018: _flash_cache_start at ??:?

I (466) esp_image: segment 8: paddr=0x000f44f0 vaddr=0x40080bfc size=0x11060 ( 69728) load
I (496) esp_image: segment 9: paddr=0x00105558 vaddr=0x400c0000 size=0x00000 (     0) load
I (496) esp_image: segment 10: paddr=0x00105560 vaddr=0x50000000 size=0x00000 (     0) load
I (503) esp_image: segment 11: paddr=0x00105568 vaddr=0x50000000 size=0x00000 (     0) load
I (522) boot: Loaded app from partition at offset 0x10000
I (522) boot: Disabling RNG early entropy source...
I (523) cpu_start: Pro cpu up.
I (527) cpu_start: Starting app cpu, entry point is 0x4008103c
0x4008103c: call_start_cpu1 at /home/james/esp/esp-idf/components/esp32/cpu_start.c:245

I (0) cpu_start: App cpu up.
I (537) heap_init: Initializing. RAM available for dynamic allocation:
I (544) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (550) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (556) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (562) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (569) heap_init: At 3FFCAB08 len 000154F8 (85 KiB): DRAM
I (575) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (581) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (587) heap_init: At 40091C5C len 0000E3A4 (56 KiB): IRAM
I (594) cpu_start: Pro cpu start user code
I (277) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (347) BTDM_INIT: BT controller compile version [ac67b34]
I (357) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (427) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
W (967) BT_BTC: btc_gatts_arg_deep_copy 8, NULL value
I (967) BLEDevice: create advertising
E (977) BT_HCI: hci write adv params error 0x30
```

Any suggestions on how to fix this error are appreciated.

chegewara, 2018-11-21:
Already answered in esp-idf. You can change code like this in BLEAdvertising.cpp:
```
	m_advParams.adv_int_min       = 0x20;
	m_advParams.adv_int_max       = 0x40;
	m_advParams.adv_type          = ADV_TYPE_IND;
	m_advParams.own_addr_type     = BLE_ADDR_TYPE_PUBLIC;
	m_advParams.channel_map       = ADV_CHNL_ALL;
	m_advParams.adv_filter_policy = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY;
	m_advParams.peer_addr_type    = BLE_ADDR_TYPE_PUBLIC;  // <-- this fix that bug
```

jamesgz, 2018-11-21:
> Already answered in esp-idf. You can change code like this in BLEAdvertising.cpp:
>
> ```
> 	m_advParams.adv_int_min       = 0x20;
> 	m_advParams.adv_int_max       = 0x40;
> 	m_advParams.adv_type          = [...]

Thanks alot! Running make monitor does not return error now. 

However, the monitor does not get updated when I connect to the ESP32 board on my android phone or when I READ/WRITE value to it.

The last line of the monitor is always `I (976) BLEDevice: create advertising`

chegewara, 2018-11-21:
Without more info i have no idea.

jamesgz, 2018-11-22:
Monitoring is not being updated because I had 'Info' for log verbosity. I see now that nkolban had 'Verbose' as his log verbosity. So I changed the log verbosity, but I have the following error keep repeating now:

`
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6164
load:0x40078000,len:10168
load:0x40080400,len:6552
entry 0x40080764
I (56) boot: ESP-IDF v3.2-beta1-151-g960c24057 2nd stage bootloader
I (56) boot: compile time 16:27:37
I (57) boot: Enabling RNG early entropy source...
I (63) boot: SPI Speed      : 40MHz
I (67) boot: SPI Mode       : DIO
I (71) boot: SPI Flash Size : 4MB
I (76) boot: Partition Table:
I (79) boot: ## Label            Usage          Type ST Offset   Length
I (86) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (94) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (101) boot:  2 factory          factory app      00 00 00010000 00100000
I (109) boot: End of partition table
I (113) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x4ef54 (323412) map
I (235) esp_image: segment 1: paddr=0x0005ef7c vaddr=0x3ff80000 size=0x00000 (     0) load
I (236) esp_image: segment 2: paddr=0x0005ef84 vaddr=0x3ff80000 size=0x00000 (     0) load
I (242) esp_image: segment 3: paddr=0x0005ef8c vaddr=0x3ffbdb60 size=0x01084 (  4228) load
I (253) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xb57b0 (743344) map
0x400d0018: _flash_cache_start at ??:?

I (520) esp_image: segment 5: paddr=0x001157d0 vaddr=0x3ffbebe4 size=0x01d48 (  7496) load
I (523) esp_image: segment 6: paddr=0x00117520 vaddr=0x3ffc092c size=0x00000 (     0) load
I (527) esp_image: segment 7: paddr=0x00117528 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/james/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (536) esp_image: segment 8: paddr=0x00117930 vaddr=0x40080400 size=0x11878 ( 71800) load
I (575) esp_image: segment 9: paddr=0x001291b0 vaddr=0x400c0000 size=0x00000 (     0) load
I (575) esp_image: segment 10: paddr=0x001291b8 vaddr=0x50000000 size=0x00000 (     0) load
I (581) esp_image: segment 11: paddr=0x001291c0 vaddr=0x50000000 size=0x00000 (     0) load
E (591) esp_image: Image length 1151472 doesn't fit in partition length 1048576
E (598) boot: Factory app partition is not bootable
E (604) boot: No bootable app partitions in the partition table
ets Jun  8 2016 00:22:57
`

chegewara, 2018-11-22:
You need to change partition in options. Most likely to `large (No OTA)`.

jamesgz, 2018-11-22:
You mean 'Single factory app, no OTA'? My partition table was already set to that.

chegewara, 2018-11-22:
Sorry, i assumed you are working with arduinom i should look more thorough on logs. In such case you need custom partitions file. But before that check if you have in menuconfig set flash size to 4MB. Its by default 2MB.

If that wont help here is simple partitions.csv:
https://gist.github.com/chegewara/68bf2b24fc12def3fcaa01ca549149fe

jamesgz, 2018-11-22:
Thanks a lot. You partition table worked.

closed 2018-11-22 by jamesgz