nkolban/esp32-snippets #724: MyCharactristicCallback causing esception when from client

vbshightime, 2018-11-22:
I am using two callbacks simultaneously. one for characteristic and one for a server. the server callback is working fine but when I am trying to write from the client and using characteristic callback for it, it is causing exception

here is my code 

       `          
             class MyCharacteristicCallback: public BLECharacteristicCallbacks

             {
                 void on  Write(BLECharacteristic* pCharactristic){

                std::string getValue = pCharacteristic->getValue();

                 Serial.println(getValue.c_str());

                  if(getValue.length()> 0){
                        clientWrite =true;
                }  
               }
            };`
             
getting this exception
        `    
                 Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
                 Core 0 register dump:
                 PC      : 0x40171da3  PS      : 0x00060430  A0      : 0x800d5cee  A1      : 0x3ffd7d90  
                 A2      : 0x3ffd7df4  A3      : 0x00000074  A4      : 0x3ffd5eb0  A5      : 0x00000000   
                 A6      : 0x00000000  A7      : 0x7f000001  A8      : 0x3ffd7dfc  A9      : 0x3ffd7d30   
                 A10     : 0x00000000  A11     : 0x3ffd6008  A12     : 0x3ffd6098  A13     : 0x00000001  
                 A14     : 0x00060220  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
                 EXCVADDR: 0x00000074  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

            Backtrace: 0x40171da3:0x3ffd7d90 0x400d5ceb:0x3ffd7db0 0x400d23d8:0x3ffd7dd0 
            0x400d133e:0x3ffd7df0 0x400d2742:0x3ffd7e30 0x400d2c25:0x3ffd80e0 
             0x400d5069:0x3ffd8100 0x400d5491:0x3ffd8150 0x400d4879:0x3ffd8170 
             0x400d364d:0x3ffd81b0 0x400e1f7d:0x3ffd81d0 0x400dd2ce:0x3ffd8210

            Rebooting...
            ets Jun  8 2016 00:22:57

             rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
             configsip: 0, SPIWP:0xee
             clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
             mode:DIO, clock div:1
             load:0x3fff0018,len:4
             load:0x3fff001c,len:952
             load:0x40078000,len:6084
             load:0x40080000,len:7936 
             entry 0x40080310
       `

chegewara, 2018-11-22:
Hi @vbshightime 
yes, we are having some issues with read/write operations from callback. I didnt check it with my latest changes yet, but if you wish you can see how it works. Its branch features from this repo or zip file from this post for arduino:
https://github.com/nkolban/esp32-snippets/issues/717#issuecomment-440858956

closed 2019-02-01 by vbshightime

W-M-D, 2021-11-24:
be me
having the same issue on a totally different repo that used this one code
????
they changed a lot
says to go here to ask questions even though the project is dead lol

chegewara, 2021-11-24:
Who sent you here and why?

This repo is dead indeed, because arduino-esp32 is now maintaining own version. Also there is plenty forks.
Anyway, it is hard to help when you just write i have the same problem on issue which has been solved long time ago.

W-M-D, 2021-11-24:
oh nevermind it was from the arduino-esp32 readme but I figured it out I forgot to pass the reference to the BLE characteristic in a function i made lol 

`uint8_t BLUETOOTH_CONTROL::update_characteristic(uint8_t & in_value, uint8_t &  value, BLECharacteristic *&c_chara)
    {
                uint8_t rc = 0;
                uint8_t * data; 
                data = c_chara->getData();
                int current_value = 0;
                current_value = *data;
                if(current_value != in_value)
                    {
                        if(value != current_value)
                            {
                                in_value = current_value;
                                value = in_value;
                                rc = 1;
                            }
                        else
                            {
                                value = in_value;
                                c_chara->setValue((uint8_t*)&in_value,4);
                            }
                    }
                return(rc);
    }
`