nkolban/esp32-snippets #725: Can't compile head of repository

fbeaulier, 2018-11-23:
There are syntax errors on code, at least one error have been introduced by the last commit.
Compilation fails.

chegewara, 2018-11-23:
@fbeaulier Can you provide some logs?

fbeaulier, 2018-11-23:
`CXX build/cpp_utils/BLEUUID.o
esp32/BLE_test/components/cpp_utils/BLEUUID.cpp:100:2: error: expected declaration before '}' token
  }
  ^
esp32/BLE_test/components/cpp_utils/BLEUUID.cpp:17:20: warning: 'LOG_TAG' defined but not used [-Wunused-variable]
 static const char* LOG_TAG = "BLEUUID";
                    ^
esp32/BLE_test/components/cpp_utils/BLEUUID.cpp:42:13: warning: 'void memrcpy(uint8_t*, uint8_t*, uint32_t)' defined but not used [-Wunused-function]
 static void memrcpy(uint8_t* target, uint8_t* source, uint32_t size) {
             ^
`

chegewara, 2018-11-23:
Just delete those 2 lines, i will fix it soon:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEUUID.cpp#L99-L100

Thanks.

fbeaulier, 2018-11-23:
there are other errors after:
`CXX build/cpp_utils/MMU.o
esp32/BLE_test/components/cpp_utils/MMU.cpp: In static member function 'static void MMU::dump()':
esp32/BLE_test/components/cpp_utils/MMU.cpp:50:24: warning: comparison is always true due to limited range of data type [-Wtype-limits]
  for (uint8_t i = 0; i < 256; i++) {
                        ^
esp32/BLE_test/components/cpp_utils/MMU.cpp:62:24: warning: comparison is always true due to limited range of data type [-Wtype-limits]
  for (uint8_t i = 0; i < 256; i++) {
                        ^
CXX build/cpp_utils/BLEUtils.o
esp32/BLE_test/components/cpp_utils/BLEUtils.cpp: In static member function 'static std::__cxx11::string BLEUtils::gattServerEventTypeToString(esp_gatts_cb_event_t)':
esp32/BLE_test/components/cpp_utils/BLEUtils.cpp:1004:8: error: 'ESP_GATTS_SEND_SERVICE_CHANGE_EVT' was not declared in this scope
   case ESP_GATTS_SEND_SERVICE_CHANGE_EVT:
        ^
`

chegewara, 2018-11-23:
Are you working with esp-idf or arduino?
Either way, just delete this:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEUtils.cpp#L1004-L1005

fbeaulier, 2018-11-23:
Esp-idf !

chegewara, 2018-11-23:
Then you should not have the last error. `ESP_GATTS_SEND_SERVICE_CHANGE_EVT` unless its dependent on ble client cache option in menuconfig.

fbeaulier, 2018-11-23:
Strangely in System.cpp 6 lines are missing in io_mux_translate array.
See how line 112 is commented as index 21 and line 113 as index 28.
Also at line 404 we have:
**const int numPins = 40;**
the number of pins is still set at 40 an that leads to reading  io_mux_translate out of its bound in the loop.

chegewara, 2018-11-23:
Yes, it would be best option since it's not related to ble.
System.cpp is part of code that I didn't even read, it's been created by @nkolban and I believe there is reason why we have gap between index 21 and 28.

closed 2019-01-22 by chegewara