nkolban/esp32-snippets #726: cpp_utils/BLEAdvertising ctor missing peer_addr_type for esp-idf v3.2

rkone, 2018-11-23:
I was getting errors using the BLE classes after updating to esp-idf 3.2 ("Invalid advertisting peer address type parameters").

it's solved by adding the following line in BLEAdvertising::BLEAdvertising(): 
m_advParams.peer_addr_type = BLE_ADDR_TYPE_PUBLIC;

chegewara, 2018-11-23:
Thanks for info, looks like its been implemented as per request:
https://github.com/espressif/esp-idf/issues/2714

chegewara, 2018-11-27:
Should be fixed in last commit.

rkone, 2018-11-27:
I think you missed BLEScan.h from the commit. I'm missing definitions for BLEScan::getResults() and BLEScan::clearResults()

chegewara, 2018-11-27:
@rkone Fixed, thanks.

rkone, 2018-11-27:
Thank you for the updates! Now to figure out what else got broken migrating to 3.2...

closed 2018-11-27 by rkone