nkolban/esp32-snippets #730: Undertanding BLE_server Example

lcaggio, 2018-11-27:
Testing the BLE_server example. It compile fine and it works on my ESP.

Checking the advertised information with nRF Connect Android app, I see the following information:

[image]

Some questions:
 1. Who can I change Device Type to: LE (not Classic and LE)?
 2. Why 2 Flags?
 3. Why 2 Complete Local Name?
 4. Who can I change intervals frequency? (at the moment is 1.2s)

chegewara, 2018-11-27:
1. to be honest i dont know how to change it, but i know one thing for sure: in arduino its 99% not possible, because both classic and ble are precompiled in library.
2. it it set by default in BLE library for one reason, it is most common option; BeEdr flag informs that only ble is supported, first flag informs how long advertising is turned on; it can be GeneralDiscoverable or LimitedDiscoverable which means device is advertising only for some short time (usually no longer than 30 sec (bluetooth specs))
3. its because ble library is build this way (this will be changed in next release), both advertising and scan response payloads are exactly the same.
4. it is caused by some bug in library which also will be fixed in next release; because some value in advertising params is not initialized it is getting random value (usually 0x02) but allowed values are 0x00 or 0x01; because bt stack cant parse it then default advertising values are used (not ble library) and it is about 1280 ms

lcaggio, 2018-11-27:
Thanks for the answer!

On the second point, let me clarify the question: why the Flags section appear twice?

chegewara, 2018-11-27:
Same reason as 2.

lcaggio, 2018-11-28:
Actually in the BLE_server example, there are no AdvertisementData or ResponseData set. I do not see why flags and complete name should be present twice: probably the library set some default behavior?

Adding the following code (empty AdvertisementData):

```
  BLEAdvertisementData oAdvertisementData;
  pAdvertising->setAdvertisementData(oAdvertisementData);
```

make the adverting as expected (please correct me if my expectation are wrong :-) ):

[image]

chegewara, 2018-11-28:
AdvertisementData and ResponseData has been introduced in library to address users requirements to customize advertising payload, especially in beacon devices.
As you suspect there is default advertising behaviour which will be changed a bit in next release. It was very simple, include all default values in advertising data and do the same in scan response.

lcaggio, 2018-11-30:
thanks, it makes sense.

About the 4th point: when the fix will be ready? is there a way to quickly fix it on the current version hardcoding and initialising the variable in some part of the code?

chegewara, 2018-11-30:
Actually it is merged to https://github.com/nkolban/ESP32_BLE_Arduino with all changes in library etc and new features are included in examples.

lcaggio, 2018-11-30:
Nice! Thanks!

closed 2018-11-30 by lcaggio

wanqbabay, 2019-01-07:
@lcaggio  @chegewara  
Hello ! The question i have met is the same as the first ONE:  TYPE: CLASSICAL AND BLE.

[image]

What is more, the NRF in an android can not pair with the ESP32.  
But iPhone can pair, I have test other esp32, and the same question.
I want to konw if the TYPE  can be set :BLE?
PS : the latest version arduino library.