nkolban/esp32-snippets #731: Timeout on BLE

msj121, 2018-11-28:
I got most of the code from nkolban with minimal changes for reading an analog value and relaying to a cordova app.

I can connect and get values; however, what seems like after 5 seconds to 2 minutes, I get an error on the cordova app that "The connection has timed out". I similarly get the nRF Connect app to connect and read values but all of a sudden it disconnects in a similar way so I don't think it's the phone software.

I have tried updating the arduino library as stated by replacing this repository (nkolban cpp_utils), restarting arduino, but it didn't work. 

Any ideas? Did I do something stupid or should I test anything? I would like to avoid constantly reconnecting if possible.

```
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLECharacteristic* pCharacteristic = NULL;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/
#define SERVICE_UUID        "F8D61858-DD19-4CED-B921-6E6CCE526D48"
#define CHARACTERISTIC_UUID "9F576249-8BE9-40B0-B009-C31FB8781207"
#define DESCRIPTOR_UUID     "a928d2aa-d48e-45aa-a8c1-51e2b44a7c45"

int analogPin = 34; //For Adafruit Feather

bool deviceConnected = false;
bool oldDeviceConnected = false;
//uint8_t value = 0;

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

void setup() {
  pinMode(analogPin, INPUT);

  BLEServer* pServer = NULL;
  BLEDescriptor* pDescriptor = NULL;
  // Create the BLE Device
  BLEDevice::init("ESP32");

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID,
                      BLECharacteristic::PROPERTY_NOTIFY 
                    );

  // https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml
  // Create a BLE Descriptor
  pCharacteristic->addDescriptor(new BLE2902());

  // Start the service
  pService->start();

  // Start advertising
  pServer->getAdvertising()->addServiceUUID(pService->getUUID());
  pServer->getAdvertising()->start();
}

void loop() {
    int analogValue = 0;
    // notify changed value
    if (deviceConnected) {
        analogValue = analogRead(analogPin);
        pCharacteristic->setValue(analogValue);
        pCharacteristic->notify();
        delay(250); // bluetooth stack will go into congestion, if too many packets are sent
    }
    // disconnecting
    if (!deviceConnected && oldDeviceConnected) {
        delay(500); // give the bluetooth stack the chance to get things ready
        pServer->startAdvertising(); // restart advertising
        oldDeviceConnected = deviceConnected;
    }
    // connecting
    if (deviceConnected && !oldDeviceConnected) {
        // do stuff here on connecting
        oldDeviceConnected = deviceConnected;
    }
}
```

Here is the verbose log from arduino
> [D][BLECharacteristic.cpp:664] setValue(): >> setValue: length=4, data=ae070000, characteristic UUID=9f576249-8be9-40b0-b009-c31fb8781207
> [D][BLECharacteristic.cpp:671] setValue(): << setValue
> [D][BLECharacteristic.cpp:524] notify(): >> notify: length: 4
> [D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONF_EVT
> [D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CONF_EVT
> [D][BLEUtils.cpp:1692] dumpGattServerEvent(): [status: ESP_GATT_OK, conn_id: 0x00]
> [D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONF_EVT
> [D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONF_EVT
> [D][BLECharacteristic.cpp:566] notify(): << notify
> [D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
> [D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
> [D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_DISCONNECT_EVT
> [D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_DISCONNECT_EVT
> [D][BLEUtils.cpp:1723] dumpGattServerEvent(): [conn_id: 0, remote_bda: 65:c3:41:74:59:c6]
> [D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
> [D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
> [D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
>

msj121, 2018-11-28:
This now works without disconnect when using @chegewara repository.

https://github.com/chegewara/esp32-snippets-enchancements-test

msj121, 2018-11-28:
Thank you everyone for this library, and Thanks @chegewara  for your work on improving the code

msj121, 2018-11-28:
Hmmm, if I change delay from 250 to 100 I am back where I started. I know you mentioned you had 70Hz, but this seems like it is only 4Hz. Is it because I am sending an int?

chegewara, 2018-11-28:
Hi @msj121 
I will try to test it asap with arduino, because my high performance was from esp-idf.

chegewara, 2018-11-28:
@msj121 Here is log from nRF connect. First time ive got situation when my samsung S9+ got problem with copy/paste nRF connect logs after few seconds connection (so many logs):
```
nRF Connect, 2018-11-29
No name (30:AE:A4:43:A1:72)
D	00:25:31.330	gatt.close()
D	00:25:31.340	wait(200)
V	00:25:31.545	Connecting to 30:AE:A4:43:A1:72...
D	00:25:31.546	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	00:25:31.852	[Server callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	00:25:31.852	[Server] Device with address 30:AE:A4:43:A1:72 connected
D	00:25:31.872	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	00:25:31.872	Connected to 30:AE:A4:43:A1:72
D	00:25:31.874	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V	00:25:31.894	Discovering services...
D	00:25:31.894	gatt.discoverServices()
I	00:25:32.347	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	00:25:32.553	[Callback] Services discovered with status: 0
I	00:25:32.553	Services discovered
V	00:25:32.561	Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
 Client Characteristic Configuration (0x2902)
Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Central Address Resolution [R] (0x2AA6)
Unknown Service (4fafc201-1fb5-459e-8fcc-c5c9c331914b)
- Unknown Characteristic [I N R W] (beb5483e-36e1-4688-b7f5-ea07361b26a8)
 Client Characteristic Configuration (0x2902)
D	00:25:32.561	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	00:25:32.563	gatt.setCharacteristicNotification(beb5483e-36e1-4688-b7f5-ea07361b26a8, true)
I	00:25:32.576	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 01
A	00:25:32.576	"(0x) 01" received
I	00:25:32.584	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 02
A	00:25:32.584	"(0x) 02" received
I	00:25:32.592	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 03
A	00:25:32.592	"(0x) 03" received
I	00:25:32.600	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 04
A	00:25:32.600	"(0x) 04" received
I	00:25:32.607	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 05
A	00:25:32.607	"(0x) 05" received
I	00:25:32.667	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 06
A	00:25:32.667	"(0x) 06" received
I	00:25:32.675	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 07
A	00:25:32.675	"(0x) 07" received
I	00:25:32.680	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 08
A	00:25:32.680	"(0x) 08" received
I	00:25:32.689	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 09
A	00:25:32.689	"(0x) 09" received
I	00:25:32.695	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 0A
A	00:25:32.695	"(0x) 0A" received
I	00:25:32.703	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 0B
A	00:25:32.703	"(0x) 0B" received
I	00:25:32.710	Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 5000ms)
I	00:25:32.719	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 0C
A	00:25:32.719	"(0x) 0C" received
I	00:25:32.727	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 0D
A	00:25:32.727	"(0x) 0D" received
I	00:25:32.735	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 0E
A	00:25:32.735	"(0x) 0E" received
I	00:25:32.741	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 0F
A	00:25:32.741	"(0x) 0F" received
I	00:25:32.750	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 10
A	00:25:32.750	"(0x) 10" received
I	00:25:32.764	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 11
A	00:25:32.764	"(0x) 11" received
I	00:25:32.764	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 12
A	00:25:32.764	"(0x) 12" received
I	00:25:32.770	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 13
A	00:25:32.770	"(0x) 13" received
I	00:25:32.776	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 14
[... 1131 lines cut ...]
A	00:25:39.094	"(0x) 7C, "|"" received
I	00:25:39.094	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 7D, "}"
A	00:25:39.094	"(0x) 7D, "}"" received
I	00:25:39.094	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 7E, "~"
A	00:25:39.094	"(0x) 7E, "~"" received
I	00:25:39.094	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 7F
A	00:25:39.094	"(0x) 7F" received
I	00:25:39.094	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 80
A	00:25:39.094	"(0x) 80" received
I	00:25:39.094	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 81
A	00:25:39.094	"(0x) 81" received
I	00:25:39.094	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 82
A	00:25:39.094	"(0x) 82" received
I	00:25:39.094	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 83
A	00:25:39.094	"(0x) 83" received
I	00:25:39.094	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 84
A	00:25:39.094	"(0x) 84" received
I	00:25:39.094	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) 85
A	00:25:39.094	"(0x) 85" received
D	00:25:39.116	[Callback] Connection state changed with status: 22 and new state: DISCONNECTED (0)
I	00:25:39.125	Disconnected
E	00:25:39.125	Error 22 (0x16): GATT CONN TERMINATE LOCAL HOST
```

I am using standard BLE_notify example from library.

chegewara, 2018-11-28:
Here is "pimp my code" version with logs:
```
/*
    Video: https://www.youtube.com/watch?v=oCMOYS71NIU
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleNotify.cpp
    Ported to Arduino ESP32 by Evandro Copercini

   Create a BLE server that, once we receive a connection, will send periodic notifications.
   The service advertises itself as: 4fafc201-1fb5-459e-8fcc-c5c9c331914b
   And has a characteristic of: beb5483e-36e1-4688-b7f5-ea07361b26a8

   The design of creating the BLE server is:
   1. Create a BLE Server
   2. Create a BLE Service
   3. Create a BLE Characteristic on the Service
   4. Create a BLE Descriptor on the characteristic
   5. Start the service.
   6. Start advertising.

   A connect hander associated with the server starts a background task that performs notification
   every couple of seconds.
*/
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLEServer* pServer = NULL;
BLECharacteristic* pCharacteristic = NULL;
bool deviceConnected = false;
bool oldDeviceConnected = false;
uint8_t value = 0;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param) {
      pServer->updateConnParams(param->connect.remote_bda, 0x06, 0x06, 0, 100);
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("MyESP32");

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID,
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_WRITE  |
                      BLECharacteristic::PROPERTY_NOTIFY |
                      BLECharacteristic::PROPERTY_INDICATE
[... 660 lines cut ...]
D	00:39:03.177	[Callback] Services discovered with status: 0
I	00:39:03.177	Services discovered
V	00:39:03.183	Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
 Client Characteristic Configuration (0x2902)
Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Central Address Resolution [R] (0x2AA6)
Unknown Service (4fafc201-1fb5-459e-8fcc-c5c9c331914b)
- Unknown Characteristic [I N R W] (beb5483e-36e1-4688-b7f5-ea07361b26a8)
 Client Characteristic Configuration (0x2902)
D	00:39:03.183	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	00:39:03.184	[Server callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	00:39:03.184	[Server] Device disconnected
D	00:39:03.197	gatt.setCharacteristicNotification(beb5483e-36e1-4688-b7f5-ea07361b26a8, true)
D	00:39:03.214	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
I	00:39:03.214	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) C4
A	00:39:03.214	"(0x) C4" received
I	00:39:03.215	Indication received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) C5
A	00:39:03.215	"(0x) C5" received
D	00:39:03.225	[Callback] Connection state changed with status: 22 and new state: DISCONNECTED (0)
E	00:39:03.225	Error 22 (0x16): GATT CONN TERMINATE LOCAL HOST
I	00:39:03.225	Disconnected
```

chegewara, 2018-11-29:
Here is version with `delay(1)` which is causing congest from time to time(often). As you can see there is about 150 ms gap (congestion) and then is madness, even 12 notifications in the same millisecond :
```
I	00:52:42.084	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) A8
A	00:52:42.084	"(0x) A8" received
I	00:52:42.084	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) A9
A	00:52:42.084	"(0x) A9" received
I	00:52:42.231	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) AA
A	00:52:42.231	"(0x) AA" received
I	00:52:42.231	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) AB
A	00:52:42.231	"(0x) AB" received
I	00:52:42.231	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) AC
A	00:52:42.231	"(0x) AC" received
I	00:52:42.231	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) AD
A	00:52:42.231	"(0x) AD" received
I	00:52:42.231	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) AE
A	00:52:42.231	"(0x) AE" received
I	00:52:42.232	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) AF
A	00:52:42.232	"(0x) AF" received
I	00:52:42.232	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) B0
A	00:52:42.232	"(0x) B0" received
I	00:52:42.232	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) B1
A	00:52:42.232	"(0x) B1" received
I	00:52:42.232	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) B2
A	00:52:42.232	"(0x) B2" received
I	00:52:42.232	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) B3
A	00:52:42.232	"(0x) B3" received
I	00:52:42.232	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) B4
A	00:52:42.232	"(0x) B4" received
I	00:52:42.232	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) B5
A	00:52:42.232	"(0x) B5" received
I	00:52:42.232	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) B6
A	00:52:42.232	"(0x) B6" received
I	00:52:42.232	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) B7
A	00:52:42.232	"(0x) B7" received
I	00:52:42.232	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) B8
A	00:52:42.232	"(0x) B8" received
I	00:52:42.232	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) B9
A	00:52:42.232	"(0x) B9" received
I	00:52:42.232	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) BA
A	00:52:42.232	"(0x) BA" received
I	00:52:42.233	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) BB
A	00:52:42.233	"(0x) BB" received
I	00:52:42.233	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) BC
A	00:52:42.233	"(0x) BC" received
I	00:52:42.233	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) BD
A	00:52:42.233	"(0x) BD" received
I	00:52:42.233	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) BE
A	00:52:42.233	"(0x) BE" received
I	00:52:42.233	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) BF
A	00:52:42.233	"(0x) BF" received
I	00:52:42.233	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) C0
A	00:52:42.233	"(0x) C0" received
I	00:52:42.233	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) C1
A	00:52:42.233	"(0x) C1" received
I	00:52:42.233	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) C2
A	00:52:42.233	"(0x) C2" received
I	00:52:42.233	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) C3
A	00:52:42.233	"(0x) C3" received
I	00:52:42.233	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) C4
A	00:52:42.233	"(0x) C4" received
I	00:52:42.233	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) C5
A	00:52:42.233	"(0x) C5" received
I	00:52:42.233	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) C6
A	00:52:42.233	"(0x) C6" received
I	00:52:42.233	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) C7
A	00:52:42.233	"(0x) C7" received
I	00:52:42.234	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) C8
A	00:52:42.234	"(0x) C8" received
I	00:52:42.234	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) C9
A	00:52:42.234	"(0x) C9" received
[... 23 lines cut ...]
A	00:52:42.235	"(0x) D5" received
I	00:52:42.235	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) D6
A	00:52:42.235	"(0x) D6" received
I	00:52:42.235	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) D7
A	00:52:42.235	"(0x) D7" received
I	00:52:42.235	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) D8
A	00:52:42.235	"(0x) D8" received
I	00:52:42.235	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) D9
A	00:52:42.235	"(0x) D9" received
I	00:52:42.235	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) DA
A	00:52:42.235	"(0x) DA" received
I	00:52:42.235	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) DB
A	00:52:42.235	"(0x) DB" received
[... cut ...]
A	00:52:42.235	"(0x) DD" received
I	00:52:42.235	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) DE
A	00:52:42.235	"(0x) DE" received
I	00:52:42.235	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) DF
A	00:52:42.235	"(0x) DF" received
I	00:52:42.235	Notification received from beb5483e-36e1-4688-b7f5-ea07361b26a8, value: (0x) E0
```

[Log 2018-11-29 00_57_29.txt](https://github.com/nkolban/esp32-snippets/files/2627028/Log.2018-11-29.00_57_29.txt)

msj121, 2018-11-29:
@chegewara Thanks, I will try to root down the problem on my code then. I fear it might be iOS because as I search through similar errors with other boards people sometimes mention that they don't have issues with android. SO I'll try to find an Android phone and see if it works there. Thank you! Your library definitely helped improve the connection and I am sure a lot of other bugs. At the very least I seem to have a stable 4Hz on iOS with your code and I didn't have even 1Hz stability before.

For now I will close, I will try to find and Android.

chegewara, 2018-11-29:
I performed long run test. After over 6 hours test with `delay(3)` I've got not even one packet drop or congest event.

To work with iOS you have 3 new functions. 
- setMinInterval (advertising),
- setMaxInterval (advertising), 
- updateConnParams (BLEServer->onConnect())

msj121, 2019-02-08:
@chegewara I still have not been able to find out why I get random disconnects on my iphone.... I have updated to the new library, and added updateConnParams:

`      pServer->updateConnParams(param->connect.remote_bda, 0x06, 0x06, 0, 100);
`

Do you or anyone know of any fixes for this? It looks like you solved it for Android, I will include a log for most current example of disconnect:

I wrote some code to write out the malloc when "disconnect" is called from the callbacks. I am hoping this might help. Any ideas how to try and narrow down the problem would also be great, happy to try things in the .cpp files.

> 01:17:25.371 -> [D][BLECharacteristic.cpp:533] notify(): << notify
> 01:17:25.371 -> [D][BLEDevice.cpp:108] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
> 01:17:25.371 -> [V][BLEUtils.cpp:1531] dumpGattServerEvent(): GATT ServerEvent: Unknown
> 01:17:25.405 -> [V][BLEUtils.cpp:1721] dumpGattServerEvent(): dumpGattServerEvent: *** NOT CODED ***
> 01:17:25.405 -> [D][BLEServer.cpp:153] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
> 01:17:25.405 -> [D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
> 01:17:25.405 -> [V][FreeRTOS.cpp:120] give(): Semaphore giving: name: ConfEvt (0x3ffe881c), owner: <N/A>
> 01:17:25.405 -> [D][BLECharacteristic.cpp:459] handleGATTServerEvent(): << handleGATTServerEvent
> 01:17:25.405 -> [D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
> 01:17:25.438 -> [D][BLECharacteristic.cpp:631] setValue(): >> setValue: length=2, data=5b05, characteristic UUID=9f576249-8be9-40b0-b009-c31fb8781207
> 01:17:25.438 -> [V][FreeRTOS.cpp:120] give(): Semaphore giving: name: ConfEvt (0x3ffe90b4), owner: <N/A>
> 01:17:25.438 -> [D][BLECharacteristic.cpp:638] setValue(): << setValue
> 01:17:25.438 -> [D][BLECharacteristic.cpp:459] handleGATTServerEvent(): << handleGATTServerEvent
> 01:17:25.438 -> [D][BLECharacteristic.cpp:484] notify(): >> notify: length: 2
> 01:17:25.438 -> [D][BLEServer.cpp:280] handleGATTServerEvent(): << handleGATTServerEvent
> 01:17:25.505 -> [V][GeneralUtils.cpp:305] hexDump():      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
> 01:17:25.505 -> [V][GeneralUtils.cpp:306] hexDump():      -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 01:17:25.505 -> [V][GeneralUtils.cpp:332] hexDump(): 0000 5b 05                                            [.
> 01:17:25.505 -> [D][BLECharacteristic.cpp:533] notify(): << notify
> 01:17:25.505 -> [D][BLEDevice.cpp:108] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
> 01:17:25.505 -> [V][BLEUtils.cpp:1531] dumpGattServerEvent(): GATT ServerEvent: Unknown
> 01:17:25.539 -> [V][BLEUtils.cpp:1721] dumpGattServerEvent(): dumpGattServerEvent: *** NOT CODED ***
> 01:17:25.539 -> [D][BLEServer.cpp:153] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
> 01:17:25.539 -> [D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
> 01:17:25.539 -> [V][FreeRTOS.cpp:120] give(): Semaphore giving: name: ConfEvt (0x3ffe881c), owner: <N/A>
> 01:17:25.539 -> [D][BLECharacteristic.cpp:459] handleGATTServerEvent(): << handleGATTServerEvent
> 01:17:25.539 -> [D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
> 01:17:25.573 -> [D][BLECharacteristic.cpp:631] setValue(): >> setValue: length=2, data=6005, characteristic UUID=9f576249-8be9-40b0-b009-c31fb8781207
> 01:17:25.573 -> [V][FreeRTOS.cpp:120] give(): Semaphore giving: name: ConfEvt (0x3ffe90b4), owner: <N/A>
> 01:17:25.573 -> [D][BLECharacteristic.cpp:638] setValue(): << setValue
> 01:17:25.573 -> [D][BLECharacteristic.cpp:459] handleGATTServerEvent(): << handleGATTServerEvent
> 01:17:25.573 -> [D][BLECharacteristic.cpp:484] notify(): >> notify: length: 2
> 01:17:25.573 -> [D][BLEServer.cpp:280] handleGATTServerEvent(): << handleGATTServerEvent
> 01:17:25.640 -> [V][GeneralUtils.cpp:305] hexDump():      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
> 01:17:25.640 -> [V][GeneralUtils.cpp:306] hexDump():      -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 01:17:25.640 -> [V][GeneralUtils.cpp:332] hexDump(): 0000 60 05                                            `.
> 01:17:25.640 -> [D][BLECharacteristic.cpp:533] notify(): << notify
> 01:17:25.640 -> [D][BLEDevice.cpp:108] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
> 01:17:25.640 -> [V][BLEUtils.cpp:1531] dumpGattServerEvent(): GATT ServerEvent: Unknown
> 01:17:25.673 -> [V][BLEUtils.cpp:1721] dumpGattServerEvent(): dumpGattServerEvent: *** NOT CODED ***
> 01:17:25.673 -> [D][BLEServer.cpp:153] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
> 01:17:25.673 -> [D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
> 01:17:25.673 -> [V][FreeRTOS.cpp:120] give(): Semaphore giving: name: ConfEvt (0x3ffe881c), owner: <N/A>
> 01:17:25.673 -> [D][BLECharacteristic.cpp:459] handleGATTServerEvent(): << handleGATTServerEvent
> 01:17:25.673 -> [D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
> 01:17:25.706 -> [D][BLECharacteristic.cpp:631] setValue(): >> setValue: length=2, data=5c05, characteristic UUID=9f576249-8be9-40b0-b009-c31fb8781207
> 01:17:25.706 -> [V][FreeRTOS.cpp:120] give(): Semaphore giving: name: ConfEvt (0x3ffe90b4), owner: <N/A>
[... cut ...]
> 01:17:27.404 -> [D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 1]
> 01:17:27.404 -> [V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 6 0x06
> 01:17:27.404 -> [V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
> 01:17:27.404 -> [V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
> 01:17:27.404 -> [D][BLEDevice.cpp:563] getAdvertising(): get advertising
> 01:17:27.404 -> [D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 6]

chegewara, 2019-02-08:
Hi,
for sure there is no issue with heap. You have plenty of it free:
`free 141428 allocated 106252 min_free 136944 largest_free_block 86612`

Problem with iPhone connection is that i dont know and understand some very specific and restrictive connection requirements. Because of that some library functions can  cause issues. From time to time when such issues are reported and i find time i am trying to read apple specs again and again.

I will try to run some tests with iPhone when i have time.

msj121, 2019-02-08:
@chegewara Thanks for all the help! Really appreciate it. From my end, what I see everytime before disconnection event is that it looks like there is some sort of stuttering in the bluetooth data. So that it will delay like 200ms a few times and clump data together.

Then will disconnect with an error peripheral disconnected from the apple side. I have tried writing my own apple software in SWIFT thinking that it might be an apple issue and the cordova library, but bug persisted.

msj121, 2019-02-08:
playing with it for a while I have noticed setting different "delays" in the loop cycle for sending data creates different timelines for disconnection event. For example, 7ms disconnects readily, 120 ms also causes some disconnects though fewer, 300ms more, and 40 ms less than all above. I am not sure if I am imagining this, but I have tried to repeat the experiment a few times on different adafruit esp32 feathers.

msj121, 2019-02-08:
I did expand looking at other ble modules to see if they have the same problem and some do have public issues, but someone posted a solution for an NRF module I'd like to try.

https://devzone.nordicsemi.com/f/nordic-q-a/11837/automatically-disconnection-ios

@chegewara Do you know where these values can be set in esp32 arduino? I have gone through the current arduino code and have had trouble finding this in bleserver or other files I have been perusing.

chegewara, 2019-02-08:
Solution from this link are those 2 functions:
```
 setMinInterval (advertising),
 setMaxInterval (advertising),
```
https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/examples/BLE_server/BLE_server.ino#L36-L37
also try to set timeout in this function for few seconds 6-10s, or even more:
` updateConnParams()`
https://github.com/nkolban/ESP32_BLE_Arduino/blob/934702b6169b92c71cbc850876fd17fb9ee3236d/src/BLEServer.cpp#L415

btw i should fix that timeout probably

msj121, 2019-02-13:
I have tried all three options:
```
void onConnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param) {
      pServer->updateConnParams(param->connect.remote_bda, 0x01, 0x90, 0, 800);
      deviceConnected = true;
    };
```

I also put the following at the end of the setup and got rid of pServer->getAdvertising() code:

```
  // Start the service
  pService->start();

  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(pService->getUUID());
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x12);
  pAdvertising->setMaxPreferred(0x16);
  BLEDevice::startAdvertising();
```

@chegewara Still same issues, but takes longer to connect, which makes sense. Did I use large enough values? Thanks for all the help, let me know if you have any other ideas to try.

Really appreciate all the help.

chegewara, 2019-02-13:
You should read this, especially 11.6:
https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf

msj121, 2019-02-13:
@chegewara Changed the settings to match as far as I can tell with no more success. I did notice that everytime I use the "reset" button or software "reset", the next connection tends to last longer like 5 minutes without interruption. 

But once there is an interruption it doesn't tend to last more than a minute, sometimes only 10 seconds. So I reset each time right now on disconnection no matter what the reason for disconnect.

This is a "bad" workaround, but it might be useful to know that something in the code on the esp32 is fixed somewhat on reset. Not sure what changes though. Let me know if I can be of any help tracking this down.

edwardocano, 2023-12-18:
@msj121 I have the same problem trying to upload the firmware using ios, Did when i send the first package the cellphone and esp32 disconnects, could you find a solution?

italocjs, 2024-01-04:
Similar issues here, almost giving up on esp32.  Connects on android but disconnect without any reason or message (even verbose activated).  Its quite sad that BLE has so little support and people sharing its code.