nkolban/esp32-snippets #732: Can't compile with latest version of ESP32

vtomanov, 2018-11-28:
I'm really stuck as a critical wifi client connect freezing bug has been fixed in the latest espressif/arduino-esp32 1.0.1 , but unfortunately with the latest  version the BLE library refuse to compile - any suggestion are very welcome. ( actually the compiling error is with any EPS32 arduino release after May/2018)
--------------------------------------------
C:\Users\admin\Desktop\arduino-1.8.5-windows\arduino-1.8.5\libraries\ESP32_BLE_Arduino\src\BLEServer.cpp:21:22: fatal error: gatt_api.h: No such file or directory

compilation terminated.

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\admin\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Not used: C:\Users\admin\Desktop\arduino-1.8.5-windows\arduino-1.8.5\libraries\WiFi

chegewara, 2018-11-28:
There is no need to yell with gig bold font.
First of all you have installed 2 arduino ide, which are causing conflict. Its not hard to read this:
```
Multiple libraries were found for "WiFi.h"
Used: C:\Users\admin\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
Not used: C:\Users\admin\Desktop\arduino-1.8.5-windows\arduino-1.8.5\libraries\WiFi
```

vtomanov, 2018-11-28:
Sorry for all - your are 100% right - all fixed now.

Thank you!

closed 2018-11-28 by vtomanov