nkolban/esp32-snippets #734: Detect if service is still available

pixelpress, 2018-11-28:
I have a server (iOS) running a service only at certain times. I am able to connect to this service and get notifications when it's running. However, if it stops and restarts I would need  to reconnect. I am wondering how I can detect if a service is still available. Do I need to look for the service UUID in each loop I am running? If there is no service anymore, do I need to establish a new connection or just turn on notifications again?

I used 

`pClient->isConnected()`

which always tells me I'm connected even if there is no service anymore. So I assume the devices stay connected independent of services running?

chegewara, 2018-11-28:
Bluetooth specs says that: if server can change services by change characteristics, add or delete service etc. then needs to implement SERVICE CHANGE characteristic with indication. 
I can think of two scenarios. 
1. Its rare that peripheral device is changing services and you can assume that each time you connect the same peripheral you find the same services and characteristics; thats why ble central devices are using `cache gatt attributes` for power save (it is implemented in esp-idf, but not yet in this library)
2. When you receive SERVICE CHANGE event you have to read all attributes from peer device after reconnect.
SERVICE CHANGE event is not implemented in this library, but in latest version you can register custom callback an add logic to your app.

For notifications you have to register on every reconnect (assumption) because it is not only write message to CCC descriptor, but also internal message to bt stack which says `i am expecting notifications`.

pixelpress, 2018-11-29:
Thanks for your response. The developer of the app confirmed that services are turned of if not needed. At the moment I'm still confused about connections. If I put the following code in the loop

`  
if(pClient->isConnected()){
  Serial.println("Client is connected");
}

pRemoteService_ = pClient->getService(serviceUUID);
pRemoteCharacteristic_ = pRemoteService_->getCharacteristic(charUUID);
  if (pRemoteCharacteristic_ == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
      Serial.println(charUUID.toString().c_str());
    }
  Serial.println("Found our characteristic and service");`

the Serial output confirms me that connection to the device, service and characteristic still exist all the time. This is hard to believe since nRF Connect, the developer and the missing notifications clearly tell me that there is no service anymore.

Furthermore, I'm also running a manual read of the characteristic all the time, even if BLE server is clearly turned off. Should this not lead to any exceptions or errors?

`std::string value_ = pRemoteCharacteristic->readValue();`

For my problem I found a workaround by turning on notifications in every loop, which is of course non-sense if the service does not even exist. But again no errors.

`   pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notificationOn, 2, true);
pRemoteCharacteristic->registerForNotify(notifyCallback);`

Is there any possibility/callback to detect if the service/characteristic exists again or at least disappeared? Or another nice way not to send registerForNotify() continuously?

chegewara, 2018-11-29:
You can try point 2 from my earlier answer but this requires some knowledge. In latest version i added option to register custom gap, gattc and gatts event handlers:
```
		BLEDevice::setCustomGapHandler(my_gap_event_handler);
		BLEDevice::setCustomGattsHandler(my_gatts_event_handler);
		BLEDevice::setCustomGattcHandler(my_gattc_event_handler);
```
In your case you will need only gattc event handler:
```
static void my_gattc_event_handler(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gattc_cb_param_t* param) {
	ESP_LOGW(LOG_TAG, "custom gattc event handler, event: %d", (uint8_t)event);
}
```
There is one issue i didnt know about. Just found that there is no SERVICE_CHANGED_EVT in gattc, it is only in gatts to inform peer device when esp32 service is changed. I have to ask in esp-idf repo why, but before i do that i will see what will happen when i change service in peer device.

jstiefel, 2018-11-29:
I start to understand, but I first need to get deeper into BLE to implement this. But I'm still wondering about the following two points:

- Why can I find service and characteristics with my code even they do not exist anymore?
- Why is there no error if I read a non-existent characteristic value?

pixelpress, 2018-11-29:
I start to understand, but I first need to get deeper into BLE to implement this. But I'm still wondering about the following two points:

- Why can I find service and characteristics with my code even they do not exist anymore?
- Why is there no error if I read a non-existent characteristic value?

chegewara, 2018-11-29:
This is real riddle. I don't know the other device you are connecting to, but from your words I am assuming it is custom code. There is one thing I can think of.
Like I said, Bluetooth specs has some rules about service change situation. There is small chance that even if not required service has been deleted, esp32 still have cached old structure. So, when you search for services esp32 is using cache data. 

I have no explanation why there is no error when you read such characteristics other than that peer device still responding on requests. Unless read value is wrong then we have bug in library.

chegewara, 2018-11-29:
I completely forgot, there is service changed in gattc:
`ESP_GATTC_SRVC_CHG_EVT`
It is also in BLEClient.cpp but only for logging.

pixelpress, 2018-11-29:
I was thinking about it again. And I realized that even I get this service change when turning off the service, there would be no way to know at what time the service exists again without continuously searching for it (I assume). So the solution is easy: continuously searching for new services. Not really satisfying to me.

chegewara, 2018-11-29:
Like i said, bluetooth specs says it very clearly:
if bluetooth le device is supporting service change at any time then `SERVICE CHANGE` characteristic is mandatory, otherwise its optional.

Of course service can change when you are not connected to peer device and you wont receive indication. 

In your situation i would consider to search for services and characteristics on connect and then write code that will listen to mentioned indications. It is already implemented in form of log message, so you should be able to test your configuration without write additional code.

pixelpress, 2018-11-29:
Nearly managed to make it work just by waiting some more seconds and then it gets disconnected, which I can easily detect. However, I'm struggling with reconnecting. I'm doing the whole scanning and connecting to the server again. Only difference is that I define "static BLEClient*  pClient;" as static. My code gets stuck at pClient->connect(pAddress) with no errors or exception (it is still running but gets stuck in the connect function). The address from scanning is right. Do I miss something? I'm doing

```
BLEDevice::init("");
pClient = BLEDevice::createClient()
```

again before scanning.

chegewara, 2018-11-29:
I see 2 possible "issues". I changed BLEClient->connect() a bit and I didn't test it enough. Now you can pass pointer to BLEAdvertsedDevice or address like before and type of address. Type of address can be taken from BLEAdvertsedDevice object and can be BLE_ADDR_TYPE_PUBLIC or BLE_ADDR_TYPE_RANDOM.

pixelpress, 2018-12-01:
A simple disconnect() before connect() solved this issue. 

I'm trying to reconnect to the same pAddress as from initial scanning (in case of an unexpected disconnect in background). And since the server runs in iOS background, it is not advertised anymore. Connecting again works fine. But in case my server deleted the service, my code crashes at `BLERemoteService* pRemoteService = pClient->getService(serviceUUID);`. According to client sketch I would expect that it outputs a nullptr if there is no service at all. What else can I do to check if service exists without using BLEAdvertisedDevice?

chegewara, 2018-12-01:
I think i did one assumption which in this case was not wise. I assumed that calling by connect() we have always new BLEClient instance or old instance with known services. Just uncomment clearServices() in BLEClient::connect() and should be fine.

chegewara, 2018-12-01:
```
But in case my server deleted the service, my code crashes at BLERemoteService* pRemoteService = pClient->getService(serviceUUID);. According to client sketch I would expect that it outputs a nullptr if there is no service at all.
```

pixelpress, 2018-12-02:
I'm using development release link and 1.0.1-rc1 esp32. But how can I make changes like trying to comment out? I do not have the esp32 folder.

pixelpress, 2018-12-02:
Lost the overview a little bit. But what I can say is that with 1.0.1-rc1 I get stuck in connect() after scanning and in 1.0.0 this works just fine. Furthermore, scanning started to not find my service sometimes in 1.0.0 which worked just fine yesterday with the same code and same release.

This is my reconnect function:

```
bool reconnect(){
    if(_initialConnect==true){
    //only do reconnect if we have address from a previous connection
      Serial.print("Reconnecting to ");
      Serial.println(pServerAddress->toString().c_str());
  
      // Connect to the remote BLE Server.
      if(!(pClient->connect(*pServerAddress))){
        Serial.print("Failed to connect to same server again");
        return false;
      }
      Serial.println(" - Connected to server again");
      
      //this expression works without any errors
      std::map<std::string, BLERemoteService*>* services_list = pClient->getServices();

      Serial.println("Debug 0");
  
      // Obtain a reference to the service we are after in the remote BLE server.
      BLERemoteService* pRemoteService = pClient->getService(serviceUUID); //this leads to crash
      Serial.println("Debug 1");
      if (pRemoteService == nullptr) {
        Serial.print("Failed to find our service UUID while reconnect: ");
        Serial.println(serviceUUID.toString().c_str());
        return false;
      }
      Serial.println(" - Found our service in reconnect");
  
  
      // Obtain a reference to the characteristic in the service of the remote BLE server.
      pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
      if (pRemoteCharacteristic == nullptr) {
        Serial.print("Failed to find our characteristic UUID while reconnect: ");
        Serial.println(charUUID.toString().c_str());
        return false;
      }
      Serial.println(" - Found our characteristic while reconnect");

      pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notificationOn, 2, true);
      Serial.println("Debug3");
      pRemoteCharacteristic->registerForNotify(notifyCallback);
      Serial.println("Debug4");
    }

}
```

chegewara, 2018-12-02:
> I'm using development release link and 1.0.1-rc1 esp32.
> clearServices() is not commented here.

https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/src/BLEClient.cpp#L111

pixelpress, 2018-12-02:
Ok, made a new try. I'm now using ESP32 1.0.1-rc1 and ESP32_BLE_Arduino commit  11a0aa6. I also adapted my code to your new example.

1. Adding "clearServices()" always leads to crashes when reconnecting (BLEClient::connect()): "ORRUPT HEAP: Bad head" crash. If I comment it, using connect() again works just fine.

2. **Sometimes** pRemoteCharacteristic->readValue() crashes even I am connected and there was a notify about new data: assertion "heap != NULL && "free() target pointer is outside heap areas"". After some resets it suddenly works or fails again. Very irregular. This was never a problem before.

3. If using exactly the same function for reconnecting as for connect(), except leaving out the scan, it gets stuck in connect(). No crash. If I do scan before, it works fine. So what does pBLEScan change?

4. Disconnect reason 19 or 62 during reconnection and it just stops there, but only rarely. What does this mean? Why does connect() stop and not just return "false"?

chegewara, 2018-12-02:
Yes, im having some problems with reason 62 and i am working on it past 2 days, but i might have solution or at least workaround. I have to clean code and prepare to update library. This is my currently running test:
```
I (5272357) TEST: custom gattc event handler, event: 40
I (5272359) TEST: custom gattc event handler, event: 2
I (5272361) TEST: custom gattc event handler, event: 38
I (5272486) TEST: custom gap event handler, event: 20
I (5272676) TEST: custom gattc event handler, event: 18
I (5272676) TEST: custom gattc event handler, event: 9
I (5272678) TEST: custom gattc event handler, event: 4
I (5272686) TEST: custom gattc event handler, event: 4
I (5272691) TEST: custom gattc event handler, event: 4
I (5272697) TEST: custom gattc event handler, event: 4
I (5272702) TEST: custom gattc event handler, event: 4
W (5272708) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016
I (5272717) TEST: custom gattc event handler, event: 5
I (5272721) TEST: custom gattc event handler, event: 41
W (5272726) my_gattc_event_handler: total disconnect count: 9199, disconnect 0x16: 9182, disconnect 0x3e: 17
```
The problem is it works only in one case:
- scan,
- start loop,
- connect,
- do things,
- disconnect,
- end loop

In this case crash with disconnect reason 62:
- start loop
- scan, 
- connect,
- do things,
- disconnect,
- end loop

I am starting to think about changing design and switch to events with only 1 callback, but this very drastically.

pixelpress, 2018-12-02:
Which BLEClient.cpp?

For me it's exactly the opposite of what you're describing. 

- scan,
- start loop,
- connect,
- do things,
- disconnect,
- end loop

Just stops in the connect() function forever. Not sure if this makes a difference, but I'm connecting to my server from before even it does not offer any services right now.

chegewara, 2018-12-02:
Sorry, this one:
https://gist.github.com/chegewara/5f23b6cf0eaa33176da2a333c1449b0b

There is also my loop included.

jstiefel, 2018-12-02:
I think I got the problem of the connect() in reconnection! I have a pClient->disconnect() before my pClient->connect(). * And this disconnect seems to have a delay and leads to a disconnection right during execution of pClient->connect(). Adding a delay between solved this.

* Because the iOS app I connect to still keeps connected for several seconds, but if service starts again I can not registerForNotify() again for whatever reason. So I just disconnect() before connecting again.

pixelpress, 2018-12-02:
I think I got the problem of the connect() in reconnection! I have a pClient->disconnect() before my pClient->connect() (1). And this disconnect seems to have a delay and leads to a disconnection right during execution of pClient->connect(). Adding a delay between solved this.

(1) Because the iOS app I connect to still keeps connected for several seconds, but if service starts again I can not registerForNotify() again for whatever reason. So I just disconnect() before connecting again.

chegewara, 2018-12-02:
I have still running test with loop code from above link and here is log:
```
W (13275737) my_gattc_event_handler: total disconnect count: 22991, disconnect 0x16: 22956, disconnect 0x3e: 34
```
almost 23000 connections with the same esp32 server and only 34 dropped connections.

chegewara, 2018-12-02:
Here is server part:
```
W (19571084) SampleServer: custom gap event handler, event: 0
W (19571086) SampleServer: custom gap event handler, event: 1
W (19571092) SampleServer: custom gap event handler, event: 6
I (19571207) SampleServer: onConnect
W (19571207) SampleServer: custom gatts event handler, event: 14
W (19571384) SampleServer: custom gap event handler, event: 20
W (19571549) SampleServer: custom gatts event handler, event: 4
W (19571567) SampleServer: custom gatts event handler, event: 2
I (19571571) SampleServer: write to char:
W (19571571) SampleServer: custom gatts event handler, event: 2
W (19571571) task: send notification 135121

I (19571584) SampleServer: write to char:
W (19571585) SampleServer: custom gatts event handler, event: 2
W (19571585) task: send notification 135122

I (19571596) SampleServer: write to char:
W (19571603) SampleServer: custom gatts event handler, event: 2
W (19571603) task: send notification 135123

I (19571610) SampleServer: write to char:
W (19571617) SampleServer: custom gatts event handler, event: 2
W (19571618) task: send notification 135124

I (19571626) SampleServer: write to char:
W (19571634) SampleServer: custom gatts event handler, event: 2
W (19571634) task: send notification 135125

I (19571640) SampleServer: onDisconnect
```

pixelpress, 2018-12-02:
I see you also have a delay in the loop: vTaskDelay(50);

Switched to your posted BLEClient file and reconnecting fails again by just stopping in connect function without error...

chegewara, 2018-12-02:
I have delay because i delete pClient object at the end, without delay it will crash because not all events are consumed yet.

pixelpress, 2018-12-02:
One of the problems seems to be that connect() does not return a bool if no connection is possible it just stops somewhere. Is there a list with disconnect reason meanings?

chegewara, 2018-12-02:
https://github.com/chegewara/esp32-ble-wiki/issues/5

chegewara, 2018-12-02:
@pixelpress because i am still experimenting with this class, there was small bug here:
https://gist.github.com/chegewara/5f23b6cf0eaa33176da2a333c1449b0b#file-bleclient-cpp-L199-L200

those 2 should be commented out or deleted.

pixelpress, 2018-12-03:
Thanks for your help. I will come back to this.

chegewara, 2018-12-03:
Hi @pixelpress 
could you test this BLEClinet.cpp?

There is also included my test code, so you could see how i do it. I have disconnect reason 62 (0x3e) and my app is still running with no problems.

pixelpress, 2018-12-06:
Hi @chegewara 
I'm a little bit busy right now. But I will test it extensively when I work on this project next time.

chegewara, 2018-12-07:
https://gist.github.com/chegewara/d3ed0e3365407f6d3abcffc486b2c462