nkolban/esp32-snippets #735: Working combination of arduino-esp32 and ESP32_BLE_Arduino versions?

an-erd, 2018-11-28:
I'm going to read values from a BLE fitness device. I used pretty old versions some time ago (BLE 0.4.9 and some version of the arduino-esp32 tree I don't have available anymore).

Now I upgraded to 
- arduino-esp32 1.0.0 and BLE 0.4.16
- arduino-esp32 1.0.1-rc1 and current version of nkolban/esp32-snippets tree (#45fcc07)
- and several other combinations 

Which all these combinations a got 
- disconnects as described in #731 
- problem as in #733 
- problems with WiFi and BLE enabled (BLE does not connect to the server or lost connection)
- diverse compile issues.

I found all these issues in this and in the arduino-esp32 issue list.

My question is, what is a good, stable working combination of arduino-esp32 and BLE which I can use to complete my sketch? It would be great to have a proven version available?

chegewara, 2018-11-28:
Current library version is after many drastic but important changes and as you can imagine it was not tested with arduino 1.0.1-rc1 because arduino has been released less than 24 hours ago.

I will do my best to fix all issues, but I need users help. I can't test most of possible usage cases so if it's possible it would very helpful to have example code posted with issue. This way I can find problem quick and try to fix it asap.

an-erd, 2018-11-28:
Versions used: arduino 1.0.1-rc1 with nkolban/esp32-snippets
 commit #45fcc07
Device used: M5Stack-Fire

Logfile with verbose messages is
`

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8412
load:0x40080400,len:5768
entry 0x40080698
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
M5Stack initializing...OK
M5 screen w 320, h 240
M5 screen textHeight 22
updateDisplayWithStatus(): status line x 160, y 157
updateStatusScreenData_Wifi(): new SSID: ...
M5 screen textHeight 22
updateDisplayWithStatus(): status line x 160, y 157
I (3094) wifi: wifi driver task: 3ffd3d94, prio:23, stack:3584, core=0
I (6296) wifi: wifi firmware version: abbd416
I (6296) wifi: config NVS flash: enabled
I (6298) wifi: config nano formating: disabled
I (6321) wifi: Init dynamic tx buffer num: 32
I (6321) wifi: Init data frame dynamic rx buffer num: 10
I (6321) wifi: Init management frame dynamic rx buffer num: 10
I (6325) wifi: Init static rx buffer size: 1600
I (6329) wifi: Init static rx buffer num: 4
I (6333) wifi: Init dynamic rx buffer num: 10
I (6418) wifi: mode : sta (84:0d:8e:25:92:34)
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
I (6562) wifi: n:8 0, o:1 0, ap:255 255, sta:8 0, prof:1
.I (7299) wifi: state: init -> auth (b0)
I (7303) wifi: state: auth -> assoc (0)
I (7308) wifi: state: assoc -> run (10)
I (7324) wifi: connected with W12, channel 8
I (7329) wifi: pm start, type: 1

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:386] _eventCallback(): STA IP: 192.168.2.154, MASK: 255.255.255.0, GW: 192.168.2.1
.Connection to WiFi done!
SSID: W12
IP Adress: 192.168.2.154
updateStatusScreenData_Wifi(): new SSID: W12
M5 screen textHeight 22
updateDisplayWithStatus(): status line x 160, y 157
OTA started!
M5 screen textHeight 22
updateDisplayWithStatus(): status line x 160, y 157
Starting disPOD application...
Using predefined device!
M5 screen textHeight 22
updateDisplayWithStatus(): status line x 160, y 157
Forming a connection to 50:33:8b:1c:f0:12
[D][BLEDevice.cpp:60] createClient(): >> createClient
[D][BLEDevice.cpp:66] createClient(): << createClient
 - Created client
[D][BLEClient.cpp:100] connect(): >> connect(50:33:8b:1c:f0:12)
[I][BLEDevice.cpp:594] addPeerDevice(): add conn_id: 0, GATT role: client
[V][BLEUtils.cpp:950] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:950] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[... 101 lines cut ...]

	DEBUGLOG(" - Found both of our characteristics\n");

	delay(100);		// TODO check to wait for param update

	// for both services: turn notifications on, turn indications off
	const uint8_t v[] = { 0x1,0x0 };
	pRemoteCharacteristicRscMeasurement->registerForNotify(notifyCallbackRsc);
	pRemoteCharacteristicRscMeasurement->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)v, 2, true);
	pRemoteCharacteristicCustom->registerForNotify(notifyCallbackCustom);
	pRemoteCharacteristicCustom->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)v, 2, true);

	delay(100);
	DEBUGLOG("Preparation done, waiting for NOTIFY\n");

	// TODO: cleanup needed in case one of the steps failed
	// TODO: check return values
	// TODO: check that everything is up and running
	return true;
}
`

If WiFI is disabled in the code, I got the same messages.

Please advice if you want me to test different versions.

chegewara, 2018-11-28:
Im going to ask you to add few lines of code to provide some debug info:
- in setup()
```
BLEDevice::setCustomGattcHandler(my_gattc_event_handler);  // before BLEDevice::init();
```
- this is custom event handler that can be used to add functionality into your code, but it is very handy to debug ble:
```
// before setup()
static void my_gattc_event_handler(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gattc_cb_param_t* param) {
	ESP_LOGW(LOG_TAG, "custom gattc event handler, event: %d", (uint8_t)event);
        if(event == ESP_GATTC_DISCONNECT_EVT) Serial.print("Disconnect reason: "); Serial.println((int)param->disconnect.reason);
}
```
This should tell us why you have disconnections. Thanks.

chegewara, 2018-11-28:
@an-erd I think i know where is the problem. To make library easier to use and compatible with more peripherals i added some option in BLEClient->connect(). I didnt test it with old examples tho.
This is is how you need to change your code:
`    pClient->connect(pAddress, type);`
In scan callback add this line:
`      type = advertisedDevice.getAddressType();`
also add global variable:
`static esp_ble_addr_type_t type;`

chegewara, 2018-11-28:
```
00:06:53.369 -> [W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 120"
[W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 121"
[W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 122"
00:06:56.436 -> [W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 123"
[W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 124"
00:06:58.476 -> [W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 125"
00:06:59.493 -> [W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 126"
[W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 127"
[W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 128"
[W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 129"
[W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 130"
[W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 131"
[W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 132"
[W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 133"
[W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
Setting new characteristic value to "Time since boot: 134"
00:07:08.677 -> [W][BLE_client.ino:96] my_gattc_event_handler(): custom gattc event handler, event: 4
```

an-erd, 2018-11-29:
@chegewara I added in setup()
`BLEDevice::setCustomGattcHandler(my_gattc_event_handler);  // before BLEDevice::init();`
and 
`static void my_gattc_event_handler()` and also changed the connect() call to use the type
`static esp_ble_addr_type_t type;` as you proposed.

To get the type variable filled correctly, I changed the code not to connect to a known predefined device but to scan first.

Results are: 
```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8412
load:0x40080400,len:5768
entry 0x40080698
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
M5Stack initializing...OK
M5 screen w 320, h 240
M5 screen textHeight 22
updateDisplayWithStatus(): status line x 160, y 157
M5 screen textHeight 22
updateDisplayWithStatus(): status line x 160, y 157
Starting disPOD application...
[D][BLEScan.cpp:201] start(): >> start(duration=30)
[D][BLEScan.cpp:233] start(): << start()
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 2 0x02
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 7 0x07
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEAdvertisedDevice.cpp:422] setRSSI(): - setRSSI(): rssi: -52
[V][BLEUtils.cpp:745] advTypeToString():  adv data type: 0x1
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[V][BLEUtils.cpp:745] advTypeToString():  adv data type: 0xff
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0xff (), length: 9, data: 4c001005031cc5e8e7
[D][BLEAdvertisedDevice.cpp:399] setManufacturerData(): - manufacturer data: 4c001005031cc5e8e7
BLE Advertised Device found: Name: , Address: 42:91:96:a3:9c:c4, manufacturer data: 4c001005031cc5e8e7
Not a MilestonePod
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:103] handleGAPEvent(): Ignoring 42:91:96:a3:9c:c4, already seen it.
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEAdvertisedDevice.cpp:422] setRSSI(): - setRSSI(): rssi: -63
[V][BLEUtils.cpp:745] advTypeToString():  adv data type: 0x1
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[V][BLEUtils.cpp:745] advTypeToString():  adv data type: 0x2
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x02 (), length: 4, data: 1418f0ff
[D][BLEAdvertisedDevice.cpp:451] setServiceUUID(): - addServiceUUID(): serviceUUID: 00001814-0000-1000-8000-00805f9b34fb
[D][BLEAdvertisedDevice.cpp:451] setServiceUUID(): - addServiceUUID(): serviceUUID: 0000fff0-0000-1000-8000-00805f9b34fb
[V][BLEUtils.cpp:745] advTypeToString():  adv data type: 0x9
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x09 (), length: 15, data: 4d696c6573746f6e65506f64203132
[D][BLEAdvertisedDevice.cpp:411] setName(): - setName(): name: MilestonePod 12
[V][BLEUtils.cpp:745] advTypeToString():  adv data type: 0x12
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x12 (), length: 4, data: 10002000
[D][BLEAdvertisedDevice.cpp:347] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[V][BLEUtils.cpp:745] advTypeToString():  adv data type: 0xa
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x0a (), length: 1, data: 00
[D][BLEAdvertisedDevice.cpp:482] setTXPower(): - txPower: 0
[... 259 lines cut ...]
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEClient.cpp:461] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEScan.cpp:103] handleGAPEvent(): Ignoring 42:91:96:a3:9c:c4, already seen it.
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEClient.cpp:461] handleGAPEvent(): BLEClient ... handling GAP event!
[W][BLEScan.cpp:75] handleGAPEvent(): ESP_GAP_SEARCH_INQ_CMPL_EVT

Stack smashing protect failure!

abort() was called at PC 0x4015fec3 on core 1

Backtrace: 0x400902f4:0x3ffcb530 0x40090525:0x3ffcb550 0x4015fec3:0x3ffcb570 0x400d2833:0x3ffcb590 0x401a1c90:0x3ffcb5b0 0x40092815:0x3ffd83a4
```
I didn't had the chance to decode the Backtrace and PC, will do this afternoon (German time) and post results.

Also, I don't understand why this line
`Starting disPOD application...`
is printed twice in the logs, but the code to print the log entry is only hit once? Also, the scan() is done twice.

So at this point of debugging we're not at the **disconnects** but at an earlier step. 

Or did I miss something?

an-erd, 2018-11-29:
Also, what does this event mean and how to deal with them?
```
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 2 0x02
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
```

an-erd, 2018-11-29:
Ok, here's another log, same code but with an disconnect.

```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8412
load:0x40080400,len:5768
entry 0x40080698
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
M5Stack initializing...OK
M5 screen w 320, h 240
M5 screen textHeight 22
updateDisplayWithStatus(): status line x 160, y 157
M5 screen textHeight 22
updateDisplayWithStatus(): status line x 160, y 157
Starting disPOD application...
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 14 0x0e
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[I][BLEDevice.cpp:194] gapEventHandler(): ESP_GAP_BLE_LOCAL_IR_EVT
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 15 0x0f
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[I][BLEDevice.cpp:197] gapEventHandler(): ESP_GAP_BLE_LOCAL_ER_EVT
[D][BLEScan.cpp:201] start(): >> start(duration=30)
[D][BLEScan.cpp:233] start(): << start()
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 2 0x02
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 7 0x07
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEAdvertisedDevice.cpp:422] setRSSI(): - setRSSI(): rssi: -52
[V][BLEUtils.cpp:745] advTypeToString():  adv data type: 0x1
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[V][BLEUtils.cpp:745] advTypeToString():  adv data type: 0xff
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0xff (), length: 9, data: 4c001005031cc5e8e7
[D][BLEAdvertisedDevice.cpp:399] setManufacturerData(): - manufacturer data: 4c001005031cc5e8e7
BLE Advertised Device found: Name: , Address: 42:91:96:a3:9c:c4, manufacturer data: 4c001005031cc5e8e7
Not a MilestonePod
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:103] handleGAPEvent(): Ignoring 42:91:96:a3:9c:c4, already seen it.
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:103] handleGAPEvent(): Ignoring 42:91:96:a3:9c:c4, already seen it.
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:103] handleGAPEvent(): Ignoring 42:91:96:a3:9c:c4, already seen it.
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:103] handleGAPEvent(): Ignoring 42:91:96:a3:9c:c4, already seen it.
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:103] handleGAPEvent(): Ignoring 42:91:96:a3:9c:c4, already seen it.
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[... 36 lines cut ...]
[V][BLEUtils.cpp:950] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[W][disPOD.ino:602] my_gattc_event_handler(): custom gattc event handler, event: 0
[V][BLEUtils.cpp:950] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:950] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[V][BLEUtils.cpp:1282] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:950] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[I][BLEDevice.cpp:605] removePeerDevice(): remove: 0, GATT role client
[... cut ...]
[W][disPOD.ino:602] my_gattc_event_handler(): custom gattc event handler, event: 2
[V][BLEUtils.cpp:950] gattClientEventTypeToString(): Unknown GATT Client event type: 1
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:950] gattClientEventTypeToString(): Unknown GATT Client event type: 1
[V][BLEUtils.cpp:1282] dumpGattClientEvent(): GATT Event: Unknown
[W][disPOD.ino:602] my_gattc_event_handler(): custom gattc event handler, event: 1
```

It looks like that the connect did not complete?

chegewara, 2018-11-29:
Yes, you are right. Disconnect reason is 62 (0x3E):
`0x3E Connection Failed to be Established`

Let me think what else i can do, because i have no issues with connection and i did test in which client (esp32) is connecting, write and receive notifications from server (esp32) over 1400 in about 35-40 minutes (1 connection per second).

an-erd, 2018-11-29:
Maybe we can postpone the "disconnect" issue and first understand the issue in the post above (Stack smashing protect failure! and the scan() function called twice)?

chegewara, 2018-11-29:
Ok, i have to read all logs one more time and i have to find stack smashing option in menuconfig to see what is this and why it is happening. I need some time, sorry.

chegewara, 2018-11-29:
From logs i think you have some issue in your app. Why do you have twice this:
```
Starting disPOD application...
[D][BLEScan.cpp:201] start(): >> start(duration=30)
[D][BLEScan.cpp:233] start(): << start()
```

an-erd, 2018-11-29:
Actually, I don't call it twice. In the code there's only one block to write the log and to start the scan, and its only called once.
```
DEBUGLOG("Starting disPOD application...\n");
BLEDevice::init("disPOD");

BLEScan* pBLEScan = BLEDevice::getScan();
	pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
	pBLEScan->setActiveScan(true);
	pBLEScan->start(30);
```

I recompiled from scratch and currently I don't have an idea why this happens. Need maybe some debugging...

chegewara, 2018-11-29:
> Also, what does this event mean and how to deal with them?
>
> ```
> [V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 2 0x02
> [V][BLEUtils.cpp:1048] dumpGapEvent(): Received [...]

I thought i will add some preprocessor that will make binary file smaller in case of using log level DEBUG or less, but something goes wrong and VERBOSE is also stripped. 
GAP event 2(0x02) is `ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT`.

chegewara, 2018-11-29:
Stack smashing:
https://github.com/espressif/arduino-esp32/issues/1455#issuecomment-394165007

an-erd, 2018-11-29:
I got the chance to decode the stack trace. It seems that some delay() functions caused some problems. I now removed all delay() for the moment.
The issue is not resolved yet. 
Current log (same versions as above)
```

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8412
load:0x40080400,len:5768
entry 0x40080698
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
M5Stack initializing...OK
M5 screen w 320, h 240
M5 screen textHeight 22
updateDisplayWithStatus(): status line x 160, y 157
M5 screen textHeight 22
updateDisplayWithStatus(): status line x 160, y 157
Starting disPOD application...
[D][BLEScan.cpp:201] start(): >> start(duration=30)
[D][BLEScan.cpp:233] start(): << start()
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 2 0x02
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 7 0x07
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEAdvertisedDevice.cpp:422] setRSSI(): - setRSSI(): rssi: -60
[V][BLEUtils.cpp:745] advTypeToString():  adv data type: 0x1
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[V][BLEUtils.cpp:745] advTypeToString():  adv data type: 0xff
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0xff (), length: 13, data: 4c000f0200001005031cf75dab
[D][BLEAdvertisedDevice.cpp:399] setManufacturerData(): - manufacturer data: 4c000f0200001005031cf75dab
BLE Advertised Device found: Name: , Address: 6b:4f:56:2b:cd:68, manufacturer data: 4c000f0200001005031cf75dab
Not a MilestonePod
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEAdvertisedDevice.cpp:422] setRSSI(): - setRSSI(): rssi: -46
[V][BLEUtils.cpp:745] advTypeToString():  adv data type: 0x1
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[V][BLEUtils.cpp:745] advTypeToString():  adv data type: 0xff
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0xff (), length: 9, data: 4c001005031c0bc45c
[D][BLEAdvertisedDevice.cpp:399] setManufacturerData(): - manufacturer data: 4c001005031c0bc45c
BLE Advertised Device found: Name: , Address: 66:03:27:0a:0b:a8, manufacturer data: 4c001005031c0bc45c
Not a MilestonePod
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:103] handleGAPEvent(): Ignoring 6b:4f:56:2b:cd:68, already seen it.
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:103] handleGAPEvent(): Ignoring 6b:4f:56:2b:cd:68, already seen it.
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:103] handleGAPEvent(): Ignoring 6b:4f:56:2b:cd:68, already seen it.
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1048] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1263] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEScan.cpp:103] handleGAPEvent(): Ignoring 6b:4f:56:2b:cd:68, already seen it.
[V][BLEUtils.cpp:1816] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[... 332 lines cut ...]
0x40092815: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```
The Code in disPod.ino, line 616 is
```
void setup() {
	// initialize the M5Stack object
	M5.begin(true, false, true);
	esp_log_level_set("*", ESP_LOG_VERBOSE);

	M5.Lcd.fillScreen(BLACK);
	DEBUGLOG("M5 screen w %u, h %u\n", M5.Lcd.width(), M5.Lcd.height());

	// show splash screen
	// TODO
	//delay(1000);

	initializeStatusScreenData();
	updateDisplayWithStatus();

```

Line 616 is the `DEBUGLOG("M5 screen w %u, h %u\n", M5.Lcd.width(), M5.Lcd.height());`

I don't know on how to follow up now. Setup has been completed, all the connection to the BLE server, etc. is done in the loop() function. The PC has passed setup(), but the stack trace shows PC in setup().
Any ideas on how to proceed?

chegewara, 2018-11-29:
Comment out this line:
`DEBUGLOG("M5 screen w %u, h %u\n", M5.Lcd.width(), M5.Lcd.height());`

an-erd, 2018-11-29:
Good idea, the PC then is located somewhere in preparing some values - need to check for wrong memory access. Will take me some time to find.

an-erd, 2018-11-30:
Didn't make any progress in finding the issue, since the code used is pretty easy, it maybe is consequence from another problem.

Do we have an example code available on how to scan, connect and enable notify for this updated tree?

chegewara, 2018-11-30:
Examples are WIP now, but actually i do have BLE_client ready:
```
/**
 * A BLE client example that is rich in capabilities.
 * There is a lot new capabilities implemented.
 * author unknown
 * updated by chegewara
 */

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("4fafc201-1fb5-459e-8fcc-c5c9c331914b");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("beb5483e-36e1-4688-b7f5-ea07361b26a8");

static boolean doConnect = false;
static boolean connected = false;
static doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
    Serial.print("data: ");
    Serial.println((char*)pData);
}

class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
  }

  void onDisconnect(BLEClient* pclient) {
    connected = false;
    Serial.println("onDisconnect");
  }
};

bool connectToServer() {
    Serial.print("Forming a connection to ");
    Serial.println(myDevice->getAddress().toString().c_str());
    
    BLEClient*  pClient  = BLEDevice::createClient();
    Serial.println(" - Created client");

    pClient->setClientCallbacks(new MyClientCallback());

    // Connect to the remove BLE Server.
    pClient->connect(myDevice);  // if you pass BLEAdvertisedDevice instead of address, it will be recognized type of peer device address (public or private)
    Serial.println(" - Connected to server");

    // Obtain a reference to the service we are after in the remote BLE server.
    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      pClient->disconnect();
      return false;
    }
    Serial.println(" - Found our service");

    // Obtain a reference to the characteristic in the service of the remote BLE server.
    pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
    if (pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
      Serial.println(charUUID.toString().c_str());
      pClient->disconnect();
      return false;
    }
    Serial.println(" - Found our characteristic");

    // Read the value of the characteristic.
    if(pRemoteCharacteristic->canRead()) {
      std::string value = pRemoteCharacteristic->readValue();
      Serial.print("The characteristic value was: ");
      Serial.println(value.c_str());
    }

    if(pRemoteCharacteristic->canNotify())
      pRemoteCharacteristic->registerForNotify(notifyCallback);

    connected = true;
}
/**
 * Scan for BLE servers and find the first one that advertises the service we are looking for.
 */
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
 /**
   * Called for each advertising BLE server.
   */
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    Serial.print("BLE Advertised Device found: ");
    Serial.println(advertisedDevice.toString().c_str());

    // We have found a device, let us now see if it contains the service we are looking for.
    if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID)) {

      BLEDevice::getScan()->stop();
      myDevice = new BLEAdvertisedDevice(advertisedDevice);
      doConnect = true;
      doScan = true;

    } // Found our server
  } // onResult
}; // MyAdvertisedDeviceCallbacks

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");
  BLEDevice::init("");

  // Retrieve a Scanner and set the callback we want to use to be informed when we
  // have detected a new device.  Specify that we want active scanning and start the
  // scan to run for 5 seconds.
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setInterval(1349);
  pBLEScan->setWindow(449);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(5, false);
} // End of setup.

// This is the Arduino main loop function.
void loop() {

  // If the flag "doConnect" is true then we have scanned for and found the desired
  // BLE Server with which we wish to connect.  Now we connect to it.  Once we are 
  // connected we set the connected flag to be true.
  if (doConnect == true) {
    if (connectToServer()) {
      Serial.println("We are now connected to the BLE Server.");
    } else {
      Serial.println("We have failed to connect to the server; there is nothin more we will do.");
    }
    doConnect = false;
  }

  // If we are connected to a peer BLE Server, update the characteristic each time we are reached
  // with the current time since boot.
  if (connected) {
    String newValue = "Time since boot: " + String(millis()/1000);
    Serial.println("Setting new characteristic value to \"" + newValue + "\"");
    
    // Set the characteristic's value to be the array of bytes that is actually a string.
    pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());
  }else if(doScan){
    BLEDevice::getScan()->start(0);  // this is just eample to start scan after disconnect, most likely there is better way to do it in arduino
  }
  
  delay(1000); // Delay a second between loops.
} // End of loop
```

an-erd, 2018-11-30:
Ok, thanks, I'll try tonight.

an-erd, 2018-11-30:
I compiled the BLE_client example you posted (with to minor corrections - it must read `static boolean doScan = false;` and `bool connectToServer() `should `return true; ` at the end.
Log file (same arduino-esp32 1.0.1-rc1 and BLE #45fcc07), output level set to `DEBUG`:

```

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8412
load:0x40080400,len:5768
entry 0x40080698
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
Starting Arduino BLE Client application...
[D][BLEScan.cpp:201] start(): >> start(duration=15)
[D][BLEScan.cpp:233] start(): << start()
[D][BLEAdvertisedDevice.cpp:422] setRSSI(): - setRSSI(): rssi: -64
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x02 (), length: 4, data: 1418f0ff
[D][BLEAdvertisedDevice.cpp:451] setServiceUUID(): - addServiceUUID(): serviceUUID: 00001814-0000-1000-8000-00805f9b34fb
[D][BLEAdvertisedDevice.cpp:451] setServiceUUID(): - addServiceUUID(): serviceUUID: 0000fff0-0000-1000-8000-00805f9b34fb
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x09 (), length: 15, data: 4d696c6573746f6e65506f64204636
[D][BLEAdvertisedDevice.cpp:411] setName(): - setName(): name: MilestonePod F6
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x12 (), length: 4, data: 10002000
[D][BLEAdvertisedDevice.cpp:347] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x0a (), length: 1, data: 00
[D][BLEAdvertisedDevice.cpp:482] setTXPower(): - txPower: 0
BLE Advertised Device found: Name: MilestonePod F6, Address: 50:33:8b:1c:ef:f6, serviceUUID: 00001814-0000-1000-8000-00805f9b34fb, txPower: 0
[D][BLEScan.cpp:256] stop(): >> stop()
[D][BLEScan.cpp:268] stop(): << stop()
Forming a connection to 50:33:8b:1c:ef:f6
[D][BLEDevice.cpp:60] createClient(): >> createClient
[D][BLEDevice.cpp:66] createClient(): << createClient
 - Created client
[D][BLEClient.cpp:100] connect(): >> connect(50:33:8b:1c:ef:f6)
[I][BLEDevice.cpp:594] addPeerDevice(): add conn_id: 0, GATT role: client
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:578] updatePeerDevice(): update conn_id: 4, GATT role: client
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:133] connect(): << connect(), rc=1
 - Connected to server
[D][BLEClient.cpp:380] getService(): >> getService: uuid: 00001814-0000-1000-8000-00805f9b34fb
[D][BLEClient.cpp:416] getServices(): >> getServices
[D][BLEClient.cpp:75] clearServices(): >> clearServices
[D][BLEClient.cpp:82] clearServices(): << clearServices
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLERemoteService.cpp:29] BLERemoteService(): >> BLERemoteService()
[D][BLERemoteService.cpp:37] BLERemoteService(): << BLERemoteService()
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLERemoteService.cpp:29] BLERemoteService(): >> BLERemoteService()
[D][BLERemoteService.cpp:37] BLERemoteService(): << BLERemoteService()
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLERemoteService.cpp:29] BLERemoteService(): >> BLERemoteService()
[D][BLERemoteService.cpp:37] BLERemoteService(): << BLERemoteService()
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLERemoteService.cpp:29] BLERemoteService(): >> BLERemoteService()
[D][BLERemoteService.cpp:37] BLERemoteService(): << BLERemoteService()
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[... 72 lines cut ...]

Decoding stack results
0x400d1eff: loop() at C:\Users\AKAEM\Documents\Arduino\MyTests\BLEtest1.ino/BLEtest1.ino.ino line 145
0x40193844: loopTask(void*) at C:\Users\AKAEM\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 17
0x40095065: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

```
Line 145 in BLEtest1.ino points to `doConnect = false;` in here.

```
// If the flag "doConnect" is true then we have scanned for and found the desired
  // BLE Server with which we wish to connect.  Now we connect to it.  Once we are 
  // connected we set the connected flag to be true.
  if (doConnect == true) {
    if (connectToServer()) {
      Serial.println("We are now connected to the BLE Server.");
    } else {
      Serial.println("We have failed to connect to the server; there is nothin more we will do.");
    }
    doConnect = false;
  }

```

Please advice on what you want to see? I can compile with a different arduino-esp32 commit, too!

an-erd, 2018-11-30:
Next time I started ends here:
```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8412
load:0x40080400,len:5768
entry 0x40080698
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
Starting Arduino BLE Client application...
[D][BLEScan.cpp:201] start(): >> start(duration=15)
[D][BLEScan.cpp:233] start(): << start()
[D][BLEAdvertisedDevice.cpp:422] setRSSI(): - setRSSI(): rssi: -68
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x02 (), length: 4, data: 1418f0ff
[D][BLEAdvertisedDevice.cpp:451] setServiceUUID(): - addServiceUUID(): serviceUUID: 00001814-0000-1000-8000-00805f9b34fb
[D][BLEAdvertisedDevice.cpp:451] setServiceUUID(): - addServiceUUID(): serviceUUID: 0000fff0-0000-1000-8000-00805f9b34fb
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x09 (), length: 15, data: 4d696c6573746f6e65506f64204636
[D][BLEAdvertisedDevice.cpp:411] setName(): - setName(): name: MilestonePod F6
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x12 (), length: 4, data: 10002000
[D][BLEAdvertisedDevice.cpp:347] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x0a (), length: 1, data: 00
[D][BLEAdvertisedDevice.cpp:482] setTXPower(): - txPower: 0
BLE Advertised Device found: Name: MilestonePod F6, Address: 50:33:8b:1c:ef:f6, serviceUUID: 00001814-0000-1000-8000-00805f9b34fb, txPower: 0
[D][BLEScan.cpp:256] stop(): >> stop()
[D][BLEScan.cpp:268] stop(): << stop()
Forming a connection to 50:33:8b:1c:ef:f6
[D][BLEDevice.cpp:60] createClient(): >> createClient
[D][BLEDevice.cpp:66] createClient(): << createClient
 - Created client
[D][BLEClient.cpp:100] connect(): >> connect(50:33:8b:1c:ef:f6)
[I][BLEDevice.cpp:594] addPeerDevice(): add conn_id: 0, GATT role: client
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:162] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
onDisconnect
[I][BLEDevice.cpp:605] removePeerDevice(): remove: 0, GATT role client
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown

```

an-erd, 2018-11-30:
Run 3 ends here. Probably the BLE device is sleeping:

```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8412
load:0x40080400,len:5768
entry 0x40080698
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
Starting Arduino BLE Client application...
[D][BLEScan.cpp:201] start(): >> start(duration=15)
[D][BLEScan.cpp:233] start(): << start()
[D][BLEAdvertisedDevice.cpp:422] setRSSI(): - setRSSI(): rssi: -68
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0xff (), length: 9, data: 4c001005031c60943c
[D][BLEAdvertisedDevice.cpp:399] setManufacturerData(): - manufacturer data: 4c001005031c60943c
BLE Advertised Device found: Name: , Address: 55:52:7f:6f:11:4c, manufacturer data: 4c001005031c60943c
[D][BLEScan.cpp:103] handleGAPEvent(): Ignoring 55:52:7f:6f:11:4c, already seen it.
[D][BLEScan.cpp:103] handleGAPEvent(): Ignoring 55:52:7f:6f:11:4c, already seen it.
[D][BLEAdvertisedDevice.cpp:422] setRSSI(): - setRSSI(): rssi: -70
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0xff (), length: 9, data: 4c001005031c5cf1cc
[D][BLEAdvertisedDevice.cpp:399] setManufacturerData(): - manufacturer data: 4c001005031c5cf1cc
BLE Advertised Device found: Name: , Address: 7a:19:e5:31:3a:19, manufacturer data: 4c001005031c5cf1cc
[D][BLEScan.cpp:103] handleGAPEvent(): Ignoring 55:52:7f:6f:11:4c, already seen it.
[W][BLEScan.cpp:75] handleGAPEvent(): ESP_GAP_SEARCH_INQ_CMPL_EVT

```

chegewara, 2018-11-30:
Hi @an-erd it possible that in rush i might have introduce some bugs. I have code that has been working yesterday pretty good in my connecting test (1400 perfect connections then massive memory leak). Today its a mess. 
I am working on it now and i will try to find the cause and fix it.

an-erd, 2018-11-30:
Hi @chegewara Ok, just drop me a note if you want me to test!

chegewara, 2018-11-30:
Yes, today's pull. I created it today and has been merged by me-no-dev to arduino-esp32 library. 
From now on i will try to keep arduino ble library updated as long as i am working with esp32.

an-erd, 2018-12-04:
@chegewara Anything I can support you with, some testing, etc.? I updated to the last version of arduino-esp32 and can provide some logs (should I open a new issue, paste here?). 

I found that the scan() example doen't find my devices until I do a search with an iPhone app. Different values for pBLEScan->setInterval(); BLEScan->setWindow(); don't improve.

chegewara, 2018-12-04:
You can paste logs here or create gists and here paste only links. 
More tests you can do and provide more logs then i can learn more about problems, only change in your app you can do to help me is to add custom callbacks:
```
static void my_gattc_event_handler(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gattc_cb_param_t* param) {
	ESP_LOGI(LOG_TAG, "custom gattc event handler, event: %d", (uint8_t)event);
}

static void my_gatts_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gatts_cb_param_t* param) {
	ESP_LOGI(LOG_TAG, "custom gatts event handler, event: %d", (uint8_t)event);
}

static void my_gap_event_handler(esp_gap_ble_cb_event_t  event, esp_ble_gap_cb_param_t* param) {
	ESP_LOGI(LOG_TAG, "custom gap event handler, event: %d", (uint8_t)event);
}
```
and add those lines before `BLEDevice::init()`:
```
    BLEDevice::setCustomGapHandler(my_gap_event_handler);
    BLEDevice::setCustomGattsHandler(my_gatts_event_handler);
    BLEDevice::setCustomGattcHandler(my_gattc_event_handler);
```

Thanks.

an-erd, 2018-12-04:
**BLE_Scan.ino don't find BLE devices unless iPhone BLE search triggered**

Used version: arduino-esp32 1.0.1-rc incl. the 13 commits since that realease (until #af7e489)
Debug level configured
Device used: M5Stack-fire
BLE Devices are awake.

Sketch: example incl. the cb handler and different values for pBLEScan->setInterval(); and pBLEScan->setWindow();
```

/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

int scanTime = 15; //In seconds
BLEScan* pBLEScan;

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s addr type: %u\n", advertisedDevice.toString().c_str(), (uint8_t) advertisedDevice.getAddressType());
	
    }
};

static void my_gattc_event_handler(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gattc_cb_param_t* param) {
	ESP_LOGI(LOG_TAG, "custom gattc event handler, event: %d", (uint8_t)event);
}

static void my_gatts_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gatts_cb_param_t* param) {
	ESP_LOGI(LOG_TAG, "custom gatts event handler, event: %d", (uint8_t)event);
}

static void my_gap_event_handler(esp_gap_ble_cb_event_t  event, esp_ble_gap_cb_param_t* param) {
	ESP_LOGI(LOG_TAG, "custom gap event handler, event: %d", (uint8_t)event);
}

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::setCustomGapHandler(my_gap_event_handler);
  BLEDevice::setCustomGattsHandler(my_gatts_event_handler);
  BLEDevice::setCustomGattcHandler(my_gattc_event_handler);
  
  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(1349);
  pBLEScan->setWindow(449);
  //pBLEScan->setInterval(100);
  //pBLEScan->setWindow(99);  // less or equal setInterval value
}

void loop() {
  // put your main code here, to run repeatedly:
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
  pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
  delay(2000);
}

[... 217 lines cut ...]
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x09 (), length: 15, data: 4d696c6573746f6e65506f64203132
[D][BLEAdvertisedDevice.cpp:413] setName(): - setName(): name: MilestonePod 12
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x12 (), length: 4, data: 10002000
[D][BLEAdvertisedDevice.cpp:349] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x0a (), length: 1, data: 00
[D][BLEAdvertisedDevice.cpp:484] setTXPower(): - txPower: 0
Advertised Device: Name: MilestonePod 12, Address: 50:33:8b:1c:f0:12, serviceUUID: 00001814-0000-1000-8000-00805f9b34fb, txPower: 0 addr type: 0
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 6c:d5:57:c8:e1:41, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 6c:d5:57:c8:e1:41, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 56:a1:79:8f:c1:62, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 50:33:8b:1c:f0:12, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 6c:d5:57:c8:e1:41, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 6c:d5:57:c8:e1:41, already seen it.

[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 6c:d5:57:c8:e1:41, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 56:a1:79:8f:c1:62, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 6c:d5:57:c8:e1:41, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 56:a1:79:8f:c1:62, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 56:a1:79:8f:c1:62, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 6c:d5:57:c8:e1:41, already seen it.
[W][BLEScan.cpp:78] handleGAPEvent(): ESP_GAP_SEARCH_INQ_CMPL_EVT
Devices found: 4
Scan done!

```

chegewara, 2018-12-04:
Thanks @an-erd 
here is my log with your code. As you can see i have logs from custom handlers, which does not appear in your logs:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9220
load:0x40080400,len:6300
entry 0x400806a4
Scanning...
[D][BLEScan.cpp:204] start(): >> start(duration=15)
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: ScanEnd (0x3ffdf13c), owner: <N/A> for start
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: ScanEnd (0x3ffdf13c), owner: start
[D][BLEScan.cpp:236] start(): << start()
[I][sketch_dec04a.ino:30] my_gap_event_handler(): custom gap event handler, event: 2
[I][sketch_dec04a.ino:30] my_gap_event_handler(): custom gap event handler, event: 7
[D][BLEAdvertisedDevice.cpp:424] setRSSI(): - setRSSI(): rssi: -53
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x0a (), length: 1, data: 01
[D][BLEAdvertisedDevice.cpp:484] setTXPower(): - txPower: 1
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x09 (), length: 5, data: 4368656765
[D][BLEAdvertisedDevice.cpp:413] setName(): - setName(): name: Chege
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x03 (), length: 2, data: 3412
[D][BLEAdvertisedDevice.cpp:453] setServiceUUID(): - addServiceUUID(): serviceUUID: 00001234-0000-1000-8000-00805f9b34fb
Advertised Device: Name: Chege, Address: 7e:af:25:7b:dd:7d, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 1 addr type: 1
[I][sketch_dec04a.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 7e:af:25:7b:dd:7d, already seen it.
[I][sketch_dec04a.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 7e:af:25:7b:dd:7d, already seen it.
[I][sketch_dec04a.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[W][BLEScan.cpp:78] handleGAPEvent(): ESP_GAP_SEARCH_INQ_CMPL_EVT
Devices found: [I][sketch_dec04a.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
1
Scan done!
```

Now is the question why? I see you have turned on PSRAM, do you use wrover board? Do you hav setup in arduino-ide wrover board or just turned on PSRAM for devKitC? Could you switch board to devKitC and turn on/off psram, even if you are using wrover?

an-erd, 2018-12-04:
Board used: M5Stack-fire
PSRAM now deactivated
Also, I set logging level to DEBUG, which give (whyever) a different log view than logging level VERBOSE
```

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9220
load:0x40080400,len:6300
entry 0x400806a4
Scanning...
[D][BLEScan.cpp:204] start(): >> start(duration=15)
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: ScanEnd (0x3ffdf3bc), owner: <N/A> for start
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: ScanEnd (0x3ffdf3bc), owner: start
[D][BLEScan.cpp:236] start(): << start()
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 2
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 7
[D][BLEAdvertisedDevice.cpp:424] setRSSI(): - setRSSI(): rssi: -58
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0xff (), length: 9, data: 4c001005031c71537b
[D][BLEAdvertisedDevice.cpp:401] setManufacturerData(): - manufacturer data: 4c001005031c71537b
Advertised Device: Name: , Address: 45:ae:d5:15:6c:5e, manufacturer data: 4c001005031c71537b addr type: 1
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 45:ae:d5:15:6c:5e, already seen it.
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 45:ae:d5:15:6c:5e, already seen it.
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEAdvertisedDevice.cpp:424] setRSSI(): - setRSSI(): rssi: -64
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0xff (), length: 9, data: 4c001005031c3d3199
[D][BLEAdvertisedDevice.cpp:401] setManufacturerData(): - manufacturer data: 4c001005031c3d3199
Advertised Device: Name: , Address: 72:80:83:b5:63:40, manufacturer data: 4c001005031c3d3199 addr type: 1
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 45:ae:d5:15:6c:5e, already seen it.
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 45:ae:d5:15:6c:5e, already seen it.
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 72:80:83:b5:63:40, already seen it.
...
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 45:ae:d5:15:6c:5e, already seen it.
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 45:ae:d5:15:6c:5e, already seen it.
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEAdvertisedDevice.cpp:424] setRSSI(): - setRSSI(): rssi: -84
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0xff (), length: 25, data: 4c000215e523fdf24c9111e5885d50338b1ceff600000000b0
[D][BLEAdvertisedDevice.cpp:401] setManufacturerData(): - manufacturer data: 4c000215e523fdf24c9111e5885d50338b1ceff600000000b0
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x09 (), length: 15, data: 4d696c6573746f6e65506f64204636
[D][BLEAdvertisedDevice.cpp:413] setName(): - setName(): name: MilestonePod F6
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x12 (), length: 4, data: 10002000
[D][BLEAdvertisedDevice.cpp:349] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x0a (), length: 1, data: 00
[D][BLEAdvertisedDevice.cpp:484] setTXPower(): - txPower: 0
Advertised Device: Name: MilestonePod F6, Address: 50:33:8b:1c:ef:f6, manufacturer data: 4c000215e523fdf24c9111e5885d50338b1ceff600000000b0, txPower: 0 addr type: 0
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 45:ae:d5:15:6c:5e, already seen it.
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 45:ae:d5:15:6c:5e, already seen it.
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
...
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 72:80:83:b5:63:40, already seen it.
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 45:ae:d5:15:6c:5e, already seen it.
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
[W][BLEScan.cpp:78] handleGAPEvent(): ESP_GAP_SEARCH_INQ_CMPL_EVT
[I][BLE_scan.ino:30] my_gap_event_handler(): custom gap event handler, event: 3
Devices found: 3
Scan done!

```

Again, the ESP32 only sees the device when scanning with a app. I'll try to better understand, because using the BLE_client.ino example with increased scan intervalls (2000/1500 finds the device.

chegewara, 2018-12-04:
If you don't use WiFi try with interval equal window.

an-erd, 2018-12-04:
Can you please give me a hint on how to do "interval equal window"?

an-erd, 2018-12-04:
Sketch BLE_client.ino

Example sketch with changed `pBLEScan->setInterval(2000); pBLEScan->setWindow(1500);` and adjusted to my UUIDs

```
static BLEUUID serviceUUID("00001814-0000-1000-8000-00805f9b34fb");
static BLEUUID    charUUID("0000ff00-0000-1000-8000-00805f9b34fb");
```

```
/**
 * A BLE client example that is rich in capabilities.
 * There is a lot new capabilities implemented.
 * author unknown
 * updated by chegewara
 */

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("00001814-0000-1000-8000-00805f9b34fb");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("0000ff00-0000-1000-8000-00805f9b34fb");

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
    Serial.print("data: ");
    Serial.println((char*)pData);
}

class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
  }

  void onDisconnect(BLEClient* pclient) {
    connected = false;
    Serial.println("onDisconnect");
  }
};

static void my_gattc_event_handler(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gattc_cb_param_t* param) {
	ESP_LOGW(LOG_TAG, "custom gattc event handler, event: %d", (uint8_t)event);
        if(event == ESP_GATTC_DISCONNECT_EVT) {
                Serial.print("Disconnect reason: "); 
                Serial.println((int)param->disconnect.reason);
        }
}

static void my_gatts_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gatts_cb_param_t* param) {
	ESP_LOGI(LOG_TAG, "custom gatts event handler, event: %d", (uint8_t)event);
}

static void my_gap_event_handler(esp_gap_ble_cb_event_t  event, esp_ble_gap_cb_param_t* param) {
	ESP_LOGI(LOG_TAG, "custom gap event handler, event: %d", (uint8_t)event);
}

[... 321 lines cut ...]
[D][BLERemoteCharacteristic.cpp:484] registerForNotify(): << registerForNotify()
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1eff  PS      : 0x00060130  A0      : 0x80190663  A1      : 0x3ffca5b0  
A2      : 0x3ffc6314  A3      : 0x3ffc51f6  A4      : 0x00000001  A5      : 0x80000020  
A6      : 0x00000000  A7      : 0x3ffba2c4  A8      : 0x800d1eff  A9      : 0x3ffca560  
A10     : 0x00000001  A11     : 0x2923fe35  A12     : 0x2923fe35  A13     : 0x3ffc6314  
A14     : 0x3ffe2224  A15     : 0x00fb0010  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x400d1eff:0x3ffca5b0 0x40190660:0x3ffca620 0x40092ded:0x3ffca640

Rebooting...
```

Decoded stack trace
```
PC: 0x400d1eff: loop() at C:\Users\AKAEM\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h line 291
EXCVADDR: 0x00000000

Decoding stack results
0x400d1eff: loop() at C:\Users\AKAEM\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h line 291
0x40190660: loopTask(void*) at C:\Users\AKAEM\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 17
0x40092ded: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

an-erd, 2018-12-04:
If I decode the exception with the (newly) build in function of VisualMicro I got this exception decoded:
```

   Error
   : ?? ??:0
   : ?? ??:0
   : ?? ??:0
   0x400d1eff StringSumHelper
   :\Users\AKAEM\Documents\Arduino\My Examples\BLE_client/BLE_client.ino:175
   : ?? ??:0
   : ?? ??:0
   :17 (discriminator 1):::0x40190660:loopTask(void*)
   :355 (discriminator 1):::0x40092ded:vPortTaskWrapper
```

an-erd, 2018-12-04:
`0x400d1eff StringSumHelper` = `PC: 0x400d1eff: loop() at C:\Users\AKAEM\Documents\Arduino\hardware\espressif\esp32\cores\esp32/WString.h line 291`

an-erd, 2018-12-04:
I just used the BLE example BLE_client.ino and only did the adjustments as mentioned above, so there no more code or my code used. I experience the same problems with my code hen using the current version, too. But, I used an old version 0.4.9 and same arduino-esp32 release I don't know anymore before - with this versions it works without any problem for 5 hours....

main.cpp is the core esp32 code, file `C:\Users\AKAEM\Documents\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp`
Main.cpp reads

```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "Arduino.h"

#if CONFIG_AUTOSTART_ARDUINO

#if CONFIG_FREERTOS_UNICORE
#define ARDUINO_RUNNING_CORE 0
#else
#define ARDUINO_RUNNING_CORE 1
#endif

void loopTask(void *pvParameters)
{
    setup();
    for(;;) {
        loop();
    }
}

extern "C" void app_main()
{
    initArduino();
    xTaskCreatePinnedToCore(loopTask, "loopTask", 8192, NULL, 1, NULL, ARDUINO_RUNNING_CORE);
}

#endif

```
and line 17 is `loop()`

chegewara, 2018-12-04:
There is also this:
`BLE_client.ino:175`

But original BLEclient has only 160 lines.

Ok, i read logs one more time. Last message is that you are trying to register for notify. Is it possible that characteristic does not provide notifications?
Try to comment it out.

an-erd, 2018-12-04:
The characteristic is providing notifications

`
[D][BLERemoteCharacteristic.cpp:444] registerForNotify(): >> registerForNotify(): Characteristic: uuid: 0000ff00-0000-1000-8000-00805f9b34fb, handle: 28 0x1c, props: broadcast: 0, read: 0, write_nr: 0, write: 0, **notify: 1**, indicate: 0, auth: 0
`
The file got a little longer because of the cb handler you asked me to add.

After register for notify connectToServer() ends, back to loop(), the next block is

```
   // If the flag "doConnect" is true then we have scanned for and found the desired
  // BLE Server with which we wish to connect.  Now we connect to it.  Once we are 
  // connected we set the connected flag to be true.
  if (doConnect == true) {
    if (connectToServer()) {
      **Serial.println("We are now connected to the BLE Server.");**
    } else {
      **Serial.println("We have failed to connect to the server; there is nothin more we will do.");**
    }
    doConnect = false;
  }

  // If we are connected to a peer BLE Server, update the characteristic each time we are reached
  // with the current time since boot.
  if (connected) {
    String newValue = "Time since boot: " + String(millis()/1000);
    Serial.println("Setting new characteristic value to \"" + newValue + "\"");
    
    // Set the characteristic's value to be the array of bytes that is actually a string.
    pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());
  }else if(doScan){
    BLEDevice::getScan()->start(0);  // this is just eample to start scan after disconnect, most likely there is better way to do it in arduino
  }
```

The log entry **"We are now connected to the BLE Server."** is misssing in the logs, maybe not flushed yet.

Line 175 is then 
`Serial.println("Setting new characteristic value to \"" + newValue + "\"");`

Is this example working for your BLE device?

an-erd, 2018-12-04:
I commented out the register for notify lines. Resulting log attached. There's no stack trace/excception.

```

Opening port
Port open
Starting Arduino BLE Client application...
[D][BLEScan.cpp:204] start(): >> start(duration=10)
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: ScanEnd (0x3ffdf400), owner: <N/A> for start
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: ScanEnd (0x3ffdf400), owner: start
[D][BLEScan.cpp:236] start(): << start()
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 2
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 7
[D][BLEAdvertisedDevice.cpp:424] setRSSI(): - setRSSI(): rssi: -48
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0xff (), length: 9, data: 4c001005031c0ba1c9
[D][BLEAdvertisedDevice.cpp:401] setManufacturerData(): - manufacturer data: 4c001005031c0ba1c9
BLE Advertised Device found: Name: , Address: 79:3a:d0:27:3f:1e, manufacturer data: 4c001005031c0ba1c9
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEAdvertisedDevice.cpp:424] setRSSI(): - setRSSI(): rssi: -44
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0xff (), length: 9, data: 4c001005031c8bb496
[D][BLEAdvertisedDevice.cpp:401] setManufacturerData(): - manufacturer data: 4c001005031c8bb496
BLE Advertised Device found: Name: , Address: 5f:93:95:3c:d5:a6, manufacturer data: 4c001005031c8bb496
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 5f:93:95:3c:d5:a6, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 5f:93:95:3c:d5:a6, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 5f:93:95:3c:d5:a6, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 5f:93:95:3c:d5:a6, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 5f:93:95:3c:d5:a6, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 5f:93:95:3c:d5:a6, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 5f:93:95:3c:d5:a6, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 5f:93:95:3c:d5:a6, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 5f:93:95:3c:d5:a6, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 5f:93:95:3c:d5:a6, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 5f:93:95:3c:d5:a6, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[... 136 lines cut ...]
[... cut ...]
[D][BLERemoteCharacteristic.cpp:286] retrieveDescriptors(): Found a descriptor: Handle: 29, UUID: 00002902-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:300] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[D][BLERemoteCharacteristic.cpp:51] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:215] retrieveCharacteristics(): << getCharacteristics()
 - Found our characteristic
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

```

chegewara, 2018-12-04:
Ok, im running now test and i have crash too. Investigating it ...

chegewara, 2018-12-04:
@an-erd Try to comment out this instead of register for notify:
```
    if(pRemoteCharacteristic->canRead()) {
      std::string value = pRemoteCharacteristic->readValue();
      Serial.print("The characteristic value was: ");
      Serial.println(value.c_str());
    }
```

chegewara, 2018-12-04:
@an-erd I know what is wrong, it was reported in other issue. `bool connectToServer()` does not return true when connection is done. Add last one line `return true;`.

an-erd, 2018-12-04:
I reported the missing `return true;` and it's already added in the used sketch.

chegewara, 2018-12-04:
You see, i have to get rest. I slept today 2 hours, sorry.

DAKarp, 2020-08-10:
Did anyone solve this issue? I'm getting exactly the same crash at registerForNotify().

closed 2020-08-10 by chegewara

prasanna37, 2023-02-05:
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0

I'm Receving this type error can someone help me out