nkolban/esp32-snippets #737: Missing include in BLEValue.cpp

mathisloge, 2018-11-30:
Hey there, 
[https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/src/BLEValue.cpp](https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/src/BLEValue.cpp) missing #include "BLEValue.h"

chegewara, 2018-11-30:
Thanks, fixed.

closed 2018-11-30 by chegewara

hansmbakker, 2018-12-02:
This issue impacts the 1.0.0 release of https://github.com/nkolban/ESP32_BLE_Arduino - could you please release a new version there?

chegewara, 2018-12-02:
@hansmbakker its been fixed in ESP32_BLE_Arduino library, or am i missing something?

hansmbakker, 2018-12-02:
This fix is also not included in the PlatformIO 1.0.0 version for example: https://github.com/nkolban/ESP32_BLE_Arduino/commit/24d7bdf1f083ceb95c06c0393e8be80c8e0cc18a.

Maybe you can create a new release? Like 1.0.1?

chegewara, 2018-12-02:
I can do that.

chegewara, 2018-12-02:
done.

hansmbakker, 2018-12-02:
Thank you! I'll wait a bit before it shows up at https://platformio.org/lib/show/1841/ESP32%20BLE%20Arduino.

The updated library.properties is already showing on the manifest tab, but it does not yet show as a new release in the right-hand sidebar and the BLE_client.ino example is not updated yet (the doScan line is not correct)

chegewara, 2018-12-02:
Sorry for inconvenience, it was big and rush update.