nkolban/esp32-snippets #738: Is Task available in Arduino?

hansmbakker, 2018-12-02:
I'm using ESP32_BLE_Arduino via PlatformIO.

I found the tests at https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/ and was wondering whether the Task class is usable in Arduino?

It is not part of the ESP32_BLE_Arduino library I believe, is it available through another library?.

hansmbakker, 2018-12-02:
Aah, I see it is linked to FreeRTOS indeed. Is it possible at all to have a BLE client and server running at the same time in Arduino?

In the example I see it is done by having two Task subclasses; my goal is to have a BLE client listen to another device, and to emit HID reports (using BLEHIDDevice) based on the notify_callbacks from the BLE client.

chegewara, 2018-12-02:
It should be possible, this is how ive been trying to design changes in library. I didnt have too much time to test all aspects, and there is small chance BLEHIDDevice may not work with windows anymore, or maybe i am just too tired and having problem with mouse example.

Being you i would try with Keyboard example first. Just remember, that you cant do operations on ble like read or write from within callback functions.

hansmbakker, 2018-12-02:
Does that mean that I cannot emit a BLE HID report from a notify_callback that is called by a characteristic?

In that case I suppose I should do the following:
- create a kind of event queue
- in the notify_callback, append events to the event queue
- in the Arduino loop, emit the events from the event queue as HID reports

Is that correct?

chegewara, 2018-12-02:
I suppose you could try to emit HID report from notify callback, if this will cause stuck your code then no you cant.

In such case you can do something like that:
```
void hid_task(void* d){
uint8_t* data = d;
do something;

delete data;
vTaskDelete(NULL);
}

void notifycallback(uint8_t* data){
xTaskCreate(scan1, "scan", 2548, data, 5, NULL);
}
```

Its simplified just to depict how it can be done.