nkolban/esp32-snippets #740: How to properly send my HID input report struct?

hansmbakker, 2018-12-02:
I'm trying to create a Windows Wheel implementation using ESP32_BLE_Arduino and the [Windows Wheel protocol documentation](https://docs.microsoft.com/en-us/windows-hardware/design/component-guidelines/radial-controller-sample-report-descriptors).

My report descriptor:
```cpp
static const uint8_t _hidReportDescriptorSurfaceDial[] = {
    // Integrated Radial Controller TLC
    0x05, 0x01,       // USAGE_PAGE (Generic Desktop)
    0x09, 0x0e,       // USAGE (System Multi-Axis Controller)
    0xa1, 0x01,       // COLLECTION (Application)
    0x85, 1,          //   REPORT_ID (Radial Controller)
    0x05, 0x0d,       //   USAGE_PAGE (Digitizers)
    0x09, 0x21,       //   USAGE (Puck)
    0xa1, 0x00,       //   COLLECTION (Physical)
    0x05, 0x09,       //     USAGE_PAGE (Buttons)
    0x09, 0x01,       //     USAGE (Button 1)
    0x95, 0x01,       //     REPORT_COUNT (1)
    0x75, 0x01,       //     REPORT_SIZE (1)
    0x15, 0x00,       //     LOGICAL_MINIMUM (0)
    0x25, 0x01,       //     LOGICAL_MAXIMUM (1)
    0x81, 0x02,       //     INPUT (Data,Var,Abs)
    0x05, 0x01,       //     USAGE_PAGE (Generic Desktop)
    0x09, 0x37,       //     USAGE (Dial)
    0x95, 0x01,       //     REPORT_COUNT (1)
    0x75, 0x0f,       //     REPORT_SIZE (15)
    0x55, 0x0f,       //     UNIT_EXPONENT (-1)
    0x65, 0x14,       //     UNIT (Degrees, English Rotation)
    0x36, 0xf0, 0xf1, //     PHYSICAL_MINIMUM (-3600)
    0x46, 0x10, 0x0e, //     PHYSICAL_MAXIMUM (3600)
    0x16, 0xf0, 0xf1, //     LOGICAL_MINIMUM (-3600)
    0x26, 0x10, 0x0e, //     LOGICAL_MAXIMUM (3600)
    0x81, 0x06,       //     INPUT (Data,Var,Rel)
    // 0x09, 0x30,       //     USAGE (X)
    // 0x75, 0x10,       //     REPORT_SIZE (16)
    // 0x55, 0x0d,       //     UNIT_EXPONENT (-3)
    // 0x65, 0x13,       //     UNIT (Inch,EngLinear)
    // 0x35, 0x00,       //     PHYSICAL_MINIMUM (0)
    // 0x46, 0xc0, 0x5d, //     PHYSICAL_MAXIMUM (24000)
    // 0x15, 0x00,       //     LOGICAL_MINIMUM (0)
    // 0x26, 0xff, 0x7f, //     LOGICAL_MAXIMUM (32767)
    // 0x81, 0x02,       //     INPUT (Data,Var,Abs)
    // 0x09, 0x31,       //     USAGE (Y)
    // 0x46, 0xb0, 0x36, //     PHYSICAL_MAXIMUM (14000)
    // 0x81, 0x02,       //     INPUT (Data,Var,Abs)
    // 0x05, 0x0d,       //     USAGE_PAGE (Digitizers)
    // 0x09, 0x48,       //     USAGE (Width)
    // 0x36, 0xb8, 0x0b, //     PHYSICAL_MINIMUM (3000)
    // 0x46, 0xb8, 0x0b, //     PHYSICAL_MAXIMUM (3000)
    // 0x16, 0xb8, 0x0b, //     LOGICAL_MINIMUM (3000)
    // 0x26, 0xb8, 0x0b, //     LOGICAL_MAXIMUM (3000)
    // 0x81, 0x03        //     INPUT (Cnst,Var,Abs)
    0xc0, //   END_COLLECTION
    0xc0  // END_COLLECTION
};
```

My input report `struct`:
```cpp
typedef struct
{
    uint16_t button : 1;
    uint16_t rotation : 15;
    //int8_t xAxis;
    //int8_t yAxis;
} WindowsWheelReport;
```

I suppose I set the report descriptor using
```cpp
pHidDevice->reportMap((uint8_t *)_hidReportDescriptorSurfaceDial, sizeof(_hidReportDescriptorSurfaceDial));
```

I'm not sure how to create and send the input report.

I guess it is something like:
```cpp
if (pHidDevice != nullptr)
    {
        WindowsWheelReport report;

        report.button = buttonPressed;
        report.rotation = rotation;
        //report.xAxis = x;
        //report.yAxis = y;

        pHidDevice->inputReport(NULL)->setValue((uint8_t*)&report, sizeof(report));
        pHidDevice->inputReport(NULL)->notify();
    }
```
but I'm not sure about how to properly use `setValue` and how to do the type conversion from the struct to the expected `setValue` arguments.

chegewara, 2018-12-02:
http://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/

This is good example:
https://github.com/nkolban/esp32-snippets/blob/45fcc07b6d4cd65aa0ff31e3e47c9d9e458abf87/cpp_utils/tests/BLETests/SampleHIDKeyboard.cpp#L46-L48

Its going to be tough ride to implement that.

hansmbakker, 2018-12-02:
@chegewara Thank you for your quick responses!

I found that example already, but I'm new to c++ and bit-level programming and have some trouble with the type conversions.

I made a [Surface Dial-compatible class for Arduino](https://github.com/NicoHood/HID/blob/master/src/HID-APIs/SurfaceDialAPI.hpp) in the past (based on examples of others), and wonder how to apply it with the functions that ESP32_BLE_Arduino

hansmbakker, 2018-12-02:
I'll try
```cpp
    if (pHidDevice != nullptr)
    {
        WindowsWheelReport report;

        report.button = buttonPressed;
        report.rotation = rotation;
        //report.xAxis = x;
        //report.yAxis = y;

        char* reportBytes = reinterpret_cast<char*>(&report);
        
        pHidDevice->inputReport(NULL)->setValue((uint8_t*)&reportBytes, sizeof(reportBytes));
        pHidDevice->inputReport(NULL)->notify();
    }
```
based on [this StackOverflow answer](https://stackoverflow.com/a/2692431/1114918)

chegewara, 2018-12-02:
Looks like you know what you are doing:
https://en.cppreference.com/w/cpp/language/bit_field

Report map from your example is exactly what you do:
- button, 1 bit value
- dial, 15 bits value

hansmbakker, 2018-12-02:
Yes I knew that the bit field is correct :) but I was wondering how to convert the struct (`report`) into something that `setValue` would accept correctly.

chegewara, 2018-12-02:
`pHidDevice->inputReport(NULL)->setValue((uint8_t*)&report, 2);`

This is all you need.

hansmbakker, 2018-12-19:
I finally got my ESP32, and started building upon what you said.

I created a queue where I push my own events to. The goal is to emit them as HID input reports.

Windows recognizes my BLE HID device and correctly sees its battery level.

However I'm having some issues with it:
- Windows reports that the HID device did not start
- when calling `pHidDevice->inputReport(1)->notify()`, the ESP32 crashes. A short part of the logs is pasted below, but I can't see what is going wrong..

I based my code upon https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleHIDDevice.cpp; you can see my full code at https://github.com/hansmbakker/NuimoRelay.

Are there any common things that can go wrong with BLE HID?

```
notifying
[D][BLEService.cpp:229] addCharacteristic(): >> addCharacteristic()
[D][BLEService.cpp:232] addCharacteristic(): Adding characteristic: uuid=00002a4d-0000-1000-8000-00805f9b34fb to service: UUID: 00001812-0000-1000-8000-00805f9b34fb, handle: 0x37
[W][BLEService.cpp:236] addCharacteristic(): << Adding a new characteristic with the same UUID as a previous one
[D][BLEService.cpp:244] addCharacteristic(): << addCharacteristic()
[D][BLECharacteristic.cpp:73] addDescriptor(): >> addDescriptor(): Adding UUID: 00002902-0000-1000-8000-00805f9b34fb, handle: 0xffff to UUID: 00002a4d-0000-1000-8000-00805f9b34fb, handle: 0xffff Read Notify
[D][BLECharacteristic.cpp:75] addDescriptor(): << addDescriptor()
[D][BLECharacteristic.cpp:73] addDescriptor(): >> addDescriptor(): Adding UUID: 00002908-0000-1000-8000-00805f9b34fb, handle: 0xffff to UUID: 00002a4d-0000-1000-8000-00805f9b34fb, handle: 0xffff Read Notify
[D][BLECharacteristic.cpp:75] addDescriptor(): << addDescriptor()
[D][BLECharacteristic.cpp:484] notify(): >> notify: length: 0
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

hansmbakker, 2018-12-20:
I fixed the HID part of the issue. Will upload my fixes to the repo mentioned above later.

However I am having trouble connecting as a client device to a central device. Will create another issue for that.

closed 2018-12-20 by hansmbakker