nkolban/esp32-snippets #742: 'btStart' was not declared in this scope

tatsutaigu, 2018-12-03:
Hi,

I use latest ESP32_BLE_Arduino on ESP-IDF.

I found error as below.
How do I pass it?

Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/foo/esp/esp-idf/requirements.txt are satisfied.
CXX build/arduino/libraries/BLE/src//BLEDevice.o
C:/msys32/home/foo/esp/bluetoothtest/components/arduino/libraries/BLE/src/BLEDevice.cpp: In static member function 'static void BLEDevice::init(std::__cxx11::string)':
C:/msys32/home/foo/esp/bluetoothtest/components/arduino/libraries/BLE/src/BLEDevice.cpp:341:16: error: 'btStart' was not declared in this scope
   if (!btStart()) {
                ^
make[1]: *** [/home/foo/esp/esp-idf/make/component_wrapper.mk:286: libraries/BLE/src//BLEDevice.o] error 1
make: *** [C:/msys32/home/foo/esp/esp-idf/make/project.mk:505: component-arduino-build] error 2

chegewara, 2018-12-03:
You have something wrong with your arduino installation.

tatsutaigu, 2018-12-03:
I tested old version of ESP32_BLE_Arduino.
It could build BLEDevice.o without error.

File time stamp is 2018/Nov/13.

chegewara, 2018-12-03:
`btStart` is part of arduino-esp32 library not BLE library:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-bt.c#L34

Like i said, you have something wrong with arduino installation.

tatsutaigu, 2018-12-04:
I installed arduino-esp32 by below site.
[https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md](url)  

I changed BLEDevice.cpp as below.
It could build without error.

Please check it.

> #include "BLEDevice.h"
#include "BLEClient.h"
#include "BLEUtils.h"
#include "GeneralUtils.h"
/*
#if defined(ARDUINO_ARCH_ESP32) && defined(CONFIG_ARDUHAL_ESP_LOG)
#include "esp32-hal-log.h"
#include "esp32-hal-bt.h"
#define LOG_TAG ""
#else
#include "esp_log.h"
static const char* LOG_TAG = "BLEDevice";
#endif
*/
#include "esp32-hal-log.h"
#include "esp32-hal-bt.h"
#include "esp_log.h"
static const char* LOG_TAG = "BLEDevice";

chegewara, 2018-12-04:
Thats the problem. I never been trying and noone else to use this lib in arduino as component app.
As you are using arduino (even if its as component in esp-idf) preprocessor can find `ARDUINO_ARCH_ESP32` flag and according to this will try to run btStart. This is something that require changes in code. You can also try to add `#undef ARDUINO_ARCH_ESP32` in your main app file, but im not sure it is good idea.

tatsutaigu, 2018-12-04:
I check it again.
below code is ok.
So, I gusss `defined(CONFIG_ARDUHAL_ESP_LOG)` is root cause.

> #include "BLEDevice.h"
#include "BLEClient.h"
#include "BLEUtils.h"
#include "GeneralUtils.h"
/*
#if defined(ARDUINO_ARCH_ESP32) && defined(CONFIG_ARDUHAL_ESP_LOG)
#include "esp32-hal-log.h"
#include "esp32-hal-bt.h"
#define LOG_TAG ""
#else
#include "esp_log.h"
static const char* LOG_TAG = "BLEDevice";
#endif
*/
#if defined(ARDUINO_ARCH_ESP32)
#include "esp32-hal-log.h"
#include "esp32-hal-bt.h"
#include "esp_log.h"
#define LOG_TAG ""
#endif

chegewara, 2018-12-04:
Ok, we can blame @me-no-dev, he asked me to add this line:
`#if defined(ARDUINO_ARCH_ESP32) && defined(CONFIG_ARDUHAL_ESP_LOG)`

I will change it, thanks.

me-no-dev, 2018-12-04:
don't go so quick on changing this. The definitions are there so logs work in Arduino. Why `btStart` is undefined is also something that needs to be looked at! This is how Arduino clears the BT memory if BT is not used. I have no problem building in IDF or Arduino, so let's see what causes that issue now. @chegewara what have you changed other than adding the lines?
Maybe split the inclusion of esp32-hal-log.h and esp32-hal-bt.h so esp32-hal-bt.h depends only on Arduino. The Logs should depend on CONFIG_ARDUHAL_ESP_LOG

tatsutaigu, 2018-12-05:
My sdkconfig is as below.
I do not apply CONFIG_ARDUHAL_ESP_LOG at menuconfig.
"Debug Log Configuration" is default setting.

Please check it.

```
#
# Debug Log Configuration
#
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_NONE=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_ERROR=y
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_WARN=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_INFO=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_DEBUG=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_VERBOSE=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL=1
CONFIG_ARDUHAL_LOG_COLORS=
CONFIG_ARDUHAL_ESP_LOG=
CONFIG_ARDUHAL_PARTITION_SCHEME_DEFAULT=y
CONFIG_ARDUHAL_PARTITION_SCHEME_MINIMAL=
CONFIG_ARDUHAL_PARTITION_SCHEME_NO_OTA=
CONFIG_ARDUHAL_PARTITION_SCHEME_MIN_SPIFFS=
CONFIG_ARDUHAL_PARTITION_SCHEME="default"
CONFIG_AUTOCONNECT_WIFI=
CONFIG_ARDUINO_SELECTIVE_COMPILATION=
```

tatsutaigu, 2018-12-05:
I checked below code, it could build without error.
I could separate `esp32-hal-log.h` and  `esp32-hal-bt.h` by define.
however, I do not know whether it is the code & result that you expected.

```
#include "BLEDevice.h"
#include "BLEClient.h"
#include "BLEUtils.h"
#include "GeneralUtils.h"
/*
#if defined(ARDUINO_ARCH_ESP32) && defined(CONFIG_ARDUHAL_ESP_LOG)
#include "esp32-hal-log.h"
#include "esp32-hal-bt.h"
#define LOG_TAG ""
#else
#include "esp_log.h"
static const char* LOG_TAG = "BLEDevice";
#endif
*/
#if defined(ARDUINO_ARCH_ESP32)
#include "esp32-hal-bt.h"
#endif

#if defined(CONFIG_ARDUHAL_ESP_LOG)
#include "esp32-hal-log.h"
#define LOG_TAG ""
#else
#include "esp_log.h"
static const char* LOG_TAG = "BLEDevice";
#endif
```

nevercast, 2018-12-13:
Hey all, not trying to hijack. I think I've the exact same issue here. Though I am not using any esp32-snippets, just trying to build arduino libs in component form from esp-idf. Is this `esp32-snippets` specific or shall this be moved to `espressif/arduino-esp32` ?

chegewara, 2018-12-13:
@tatsutaigu I think this code looks good. Could you prepare PR and i will merge it?

tatsutaigu, 2018-12-14:
I try pull request.
Please check it.

chegewara, 2018-12-14:
Merged, thanks @tatsutaigu

tatsutaigu, 2018-12-21:
I could confirm No Error with marged code.
Thanks.

closed 2018-12-21 by tatsutaigu