nkolban/esp32-snippets #745: Scan does not stop/is restarting although characteristicUUID is found

an-erd, 2018-12-04:
**BLE_Client.ino finds correct characterisicUUID but does not stop/is restarting scanning**

Used version: arduino-esp32 1.0.1-rc incl. the 13 commits since that realease (until #af7e489)
Debug logging level configured
Device used: M5Stack-fire
PSRAM disabled
BLE Devices are awake.

Sketch BLE_Client.ino with changes in `pBLEScan->setInterval(2000); pBLEScan->setWindow(1500);` and cb functions added:

```
/**
 * A BLE client example that is rich in capabilities.
 * There is a lot new capabilities implemented.
 * author unknown
 * updated by chegewara
 */

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("00001814-0000-1000-8000-00805f9b34fb");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("0000ff00-0000-1000-8000-00805f9b34fb");

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
    Serial.print("data: ");
    Serial.println((char*)pData);
}

class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
  }

  void onDisconnect(BLEClient* pclient) {
    connected = false;
    Serial.println("onDisconnect");
  }
};

static void my_gattc_event_handler(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gattc_cb_param_t* param) {
	ESP_LOGW(LOG_TAG, "custom gattc event handler, event: %d", (uint8_t)event);
        if(event == ESP_GATTC_DISCONNECT_EVT) {
                Serial.print("Disconnect reason: "); 
                Serial.println((int)param->disconnect.reason);
        }
}

static void my_gatts_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gatts_cb_param_t* param) {
	ESP_LOGI(LOG_TAG, "custom gatts event handler, event: %d", (uint8_t)event);
}

static void my_gap_event_handler(esp_gap_ble_cb_event_t  event, esp_ble_gap_cb_param_t* param) {
	ESP_LOGI(LOG_TAG, "custom gap event handler, event: %d", (uint8_t)event);
}
[... 330 lines cut ...]
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 5f:93:95:3c:d5:a6, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
...

```

The device found provides the correct serviceUUID, so scan should stop. 
In deed, it stops, but didn't report that the correct device was found, and then it starts again with a duration of 0. The log does not end (run for 10min) always reporting 
```

[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 79:3a:d0:27:3f:1e, already seen it.

```

chegewara, 2018-12-04:
Your logs:
```
BLE Advertised Device found: Name: MilestonePod F6, Address: 50:33:8b:1c:ef:f6, serviceUUID: 00001814-0000-1000-8000-00805f9b34fb, txPower: 0
[D][BLEScan.cpp:259] stop(): >> stop()
[D][BLEScan.cpp:271] stop(): << stop()
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 3
[I][BLE_client.ino:59] my_gap_event_handler(): custom gap event handler, event: 18
[D][BLEScan.cpp:204] start(): >> start(duration=0)
```
Scan stopped, but then it started again with duration 0 (infinity).

an-erd, 2018-12-04:
Ok, then I think the example provided by the BLE library needs to be reworked. The flag `doScan` is set in the cb function when we found our device, but before we're connected. Depending where we're in loop() it may be the case that we jump in the second block  ` ... }else if(doScan){...`

```
void loop() {

  // If the flag "doConnect" is true then we have scanned for and found the desired
  // BLE Server with which we wish to connect.  Now we connect to it.  Once we are 
  // connected we set the connected flag to be true.
  if (doConnect == true) {
    if (connectToServer()) {
      Serial.println("We are now connected to the BLE Server.");
    } else {
      Serial.println("We have failed to connect to the server; there is nothin more we will do.");
    }
    doConnect = false;
  }

  // If we are connected to a peer BLE Server, update the characteristic each time we are reached
  // with the current time since boot.
  if (connected) {
    String newValue = "Time since boot: " + String(millis()/1000);
    Serial.println("Setting new characteristic value to \"" + newValue + "\"");
    
    // Set the characteristic's value to be the array of bytes that is actually a string.
    pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());
  }else if(doScan){
    BLEDevice::getScan()->start(0);  // this is just eample to start scan after disconnect, most likely there is better way to do it in arduino
  }
  
  delay(1000); // Delay a second between loops.
} // End of loop

```

Quick fix could be to change this block in `loop()` to 
```
 // If the flag "doConnect" is true then we have scanned for and found the desired
  // BLE Server with which we wish to connect.  Now we connect to it.  Once we are 
  // connected we set the connected flag to be true.
 if (doConnect == true) {
    if (connectToServer()) {
      Serial.println("We are now connected to the BLE Server.");
    } else {
      Serial.println("We have failed to connect to the server; there is nothin more we will do.");
    }
    doConnect = false;
  } else {
    // If we are connected to a peer BLE Server, update the characteristic each time we are reached
    // with the current time since boot.
    if (connected) {
      String newValue = "Time since boot: " + String(millis()/1000);
      Serial.println("Setting new characteristic value to \"" + newValue + "\"");
    
      // Set the characteristic's value to be the array of bytes that is actually a string.
      pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());
    }else if(doScan){
      BLEDevice::getScan()->start(0);  // this is just eample to start scan after disconnect, most likely there is better way to do it in arduino
    }
  }
```

an-erd, 2018-12-04:
After the correction of the code as shown above, I got this results:

Sketch
```

/**
 * A BLE client example that is rich in capabilities.
 * There is a lot new capabilities implemented.
 * author unknown
 * updated by chegewara
 */

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("00001814-0000-1000-8000-00805f9b34fb");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("0000ff00-0000-1000-8000-00805f9b34fb");

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
    Serial.print("data: ");
    Serial.println((char*)pData);
}

class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
  }

  void onDisconnect(BLEClient* pclient) {
    connected = false;
    Serial.println("onDisconnect");
  }
};

static void my_gattc_event_handler(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gattc_cb_param_t* param) {
	ESP_LOGW(LOG_TAG, "custom gattc event handler, event: %d", (uint8_t)event);
        if(event == ESP_GATTC_DISCONNECT_EVT) {
                Serial.print("Disconnect reason: "); 
                Serial.println((int)param->disconnect.reason);
        }
}

static void my_gatts_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gatts_cb_param_t* param) {
	ESP_LOGI(LOG_TAG, "custom gatts event handler, event: %d", (uint8_t)event);
}

static void my_gap_event_handler(esp_gap_ble_cb_event_t  event, esp_ble_gap_cb_param_t* param) {
	ESP_LOGI(LOG_TAG, "custom gap event handler, event: %d", (uint8_t)event);
}

bool connectToServer() {
    Serial.print("Forming a connection to ");
    Serial.println(myDevice->getAddress().toString().c_str());
    
    BLEClient*  pClient  = BLEDevice::createClient();
[... 276 lines cut ...]
Exception
```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1865  PS      : 0x00060130  A0      : 0x800d1f02  A1      : 0x3ffca580  
A2      : 0x00000001  A3      : 0x01024bc8  A4      : 0x00000009  A5      : 0x3ffc6314  
A6      : 0x3ffe2854  A7      : 0x00fb0010  A8      : 0x800d1dc4  A9      : 0x3ffca500  
A10     : 0x01024bc8  A11     : 0x01024bc8  A12     : 0x3ffc6314  A13     : 0x3f4025b9  
A14     : 0x3f402b07  A15     : 0x3ffe369c  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000001  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x400d1865:0x3ffca580 0x400d1eff:0x3ffca5b0 0x401903f0:0x3ffca620 0x40092ded:0x3ffca640
```
Decoded
```
PC: 0x400d1865: std::__cxx11::basic_string  , std::allocator  >::basic_string(char const*, std::allocator  const&) at c:\users\akaem\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/basic_string.h line 109
EXCVADDR: 0x00000001

Decoding stack results
0x400d1865: std::__cxx11::basic_string  , std::allocator  >::basic_string(char const*, std::allocator  const&) at c:\users\akaem\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/basic_string.h line 109
0x400d1eff: loop() at C:\Users\AKAEM\Documents\Arduino\My Examples\BLE_client/BLE_client.ino line 177
0x401903f0: loopTask(void*) at C:\Users\AKAEM\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 17
0x40092ded: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

```

chegewara, 2018-12-04:
Sorry for all issues with library. Before i decided to merge all changes i tested it very intensive with connect/disconnect performance tests. But looks like my tests were not comprehensive and didnt catch some issues.

I will try to fix all issues asap, but it can take few days maybe even more, since i have day job, upwork job and not so much time. I appreciate all posted issues and logs with observations which gives me knowledge what is going on and what needs to be fixed, even if not always it is obvious.

Thanks again and sorry.