nkolban/esp32-snippets #746: Fail connection when I read address from spiffs file

mpbejo, 2018-12-04:
Hi everybody,
My goal is to scan the BLE devices and when I find my device I save in a file its MAC address so I do not have to repeat the scan.
The problem is that when I try to connect using the saved MAC Address I have the following message:

Reading file: /MAC.txt
- read from file:
MAC Address from file: 24:0a:c4:9e:d4:2e
Forming a connection to 24:0a:c4:9e:d4:2e
 - Created client
 - Connecting to server
[D][BLEClient.cpp:91] connect(): >> connect(24:0a:c4:9e:d4:2e)
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:81] clearServices(): << clearServices
[E][BLEClient.cpp:101] connect(): esp_ble_gattc_app_register: rc=259 Invalid state
[D][BLEClient.cpp:133] disconnect(): >> disconnect()
[E][BLEClient.cpp:136] disconnect(): esp_ble_gattc_close: rc=259 Invalid state
 - Disconnected from connectToServer

My snippet code:
std::string MacAddress;
readFile(SPIFFS, "/MAC.txt");
    if (FileExists == false)
    { 
      Scanning(); 
    }
    else
    {  
      Serial.print("MAC Address from file: ");
      Serial.println(MacAddress.c_str());
      Address = new BLEAddress(MacAddress);
      connectToServer(*Address);

Any suggestions?
Thank You
Best Regards
Marco

mpbejo, 2018-12-04:
I have the same error if i write; 
Address = new BLEAddress("24:0a:c4:9e:d4:2e");
What  am I missing?

chegewara, 2018-12-04:
Again, i dnt know. I did test where i did:
- scan,
- start loop,
- connect,
- do something,
- disconnect,
- end loop,

This loop has been ran for few thousands times and it worked to me. I also have upwork job with arduino library where i have to find device by mac address and connect to it and it worked. 
`posPrinter.scanForPrinter(1, "66:02:af:f1:69:ee")`

mpbejo, 2018-12-05:
Sorry, I found my problem I forgot 
BLEDevice::init("");
Yesterday i looked for hours, but i did not see

chegewara, 2018-12-05:
No problem, im glad you found it.

closed 2018-12-05 by chegewara