nkolban/esp32-snippets #749: Access Name or BLE Address in NotifyHandler

espresso-if, 2018-12-11:
Hello together,

is there a way to Access the BLEAddress inside a notify Handler for a RemoteCharacteristic?
My specific case:

Multiple devices with the same Service ID connected to clients which notifies about changes, how can I distinguish them?

I haven't found an interface to access an Attribute to accomplish this.

Plattform: ESP32 with ESP-IDF

Thank you in advance

espresso-if, 2018-12-11:
Ok thank you
Do you know a work-around or something similar
Because I need it for an official project

chegewara, 2018-12-11:
Except changing library code? I could not find something quick right now.

espresso-if, 2018-12-13:
FYI

I solved it by adding the following public function to BLERemoteCharacteristic
BLEAddress  BLERemoteCharacteristic::getPeerAddress(){
	return m_pRemoteService->getClient()->getPeerAddress();
}

chegewara, 2018-12-14:
@espresso-if Like i said, change library code. Thanks for update.

espresso-if, 2018-12-14:
See #752

closed 2018-12-14 by espresso-if