nkolban/esp32-snippets #751: Problem multiple characterisic

espresso-if, 2018-12-14:
Hello

I'm trying to use multiple charactericts within one service.
But when the Client retrieves the characteristics within  BLERemoteService::retrieveCharacteristics, it fails while accessing the second characteristic:
```

	if (status == ESP_GATT_INVALID_OFFSET) {   // We have reached the end of the entries.
			ESP_LOGE(LOG_TAG, "Invalid offset in status:< %s", BLEUtils::gattStatusToString(status).c_str());
			break;
		}
```
 
Output:
D (3603) BLEClient: << getService: found the service with uuid: 0855d3e4-d847-4fc5-a06d-e3ec9629146f␛[0m
D (3623) BLERemoteService: >> getCharacteristics() for service: 0855d3e4-d847-4fc5-a06d-e3ec9629146f␛[0m
D (3633) BLERemoteService: Found a characteristic: Handle: 42, UUID: c076b13d-f8b9-46e0-975a-70080fb15b09␛[0m
␛[0;31mE (3643) BLERemoteService: Invalid offset in status: Unknown␛[0m
D (3643) BLERemoteService: << getCharacteristics()␛[0m

All characteristics seemed to be added correcty at the serverside while Debugging

Thanks in advance

chegewara, 2018-12-14:
Is this original code from this library or you added `ESP_LOGE` to have log output?
Im asking because this is code you  pionting on:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLERemoteService.cpp#L183-L185

It may be you are using old version of library or something?

espresso-if, 2018-12-14:
There are no typos
But I think the issue is that that just one RemoteCharacteristic is found while retrieving because BLERemoteService::retrieveCharacteristics, stops after one Iteration

> if(pChar == nullptr) quit;.

 I have that in my code and that's the line where my programm stops

Only one characterstic can be found in the RemoteCharacteristic - two are missing

chegewara, 2018-12-14:
Ok, thanks. Will check it.

espresso-if, 2019-01-06:
Any news on this?

chegewara, 2019-01-07:
Sorry for delay. Ive been testing it, just quick test with 2 or 3 characteristics, and i couldnt recreate it.
I have to fix one or two issues with BLEClient and update library, but still no time to do it.

wakwak-koba, 2019-01-14:
I am fixing original bugs and succeeding multiple connections.

See below for details.
https://github.com/nkolban/esp32-snippets/issues/765#issuecomment-453900444

wakwak-koba, 2019-01-16:
I am github-biginner so I do not know much about it, Is the procedure below?

1. create new repository. (fork from 'https://github.com/nkolban/esp32-snippets')
2. Fix code (and commit & push)
3. create Pull-Request

please tell me...

chegewara, 2019-01-16:
Yes @wakwak-koba

wakwak-koba, 2019-01-17:
PRed!

closed 2019-01-30 by chegewara

HoernchenJ, 2021-01-07:
Sorry for opening this Issue again, but i am having trouble to use multiple characteristics on one service.

**Here is my example code:**

`void setup() {
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);

  // Create the BLE Device
  BLEDevice::init("ESP32");

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new ServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(BLEUUID(SERVICE_UUID));

  // Create a Characteristic for the LED
  pLED = pService->createCharacteristic(
                      BLEUUID(LED_UUID),
                      BLECharacteristic::PROPERTY_NOTIFY |
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_WRITE  
                    );
                    
  // Create a Characteristic for the pan rotation
  pPan = pService->createCharacteristic(
                      BLEUUID(PAN_UUID),
                      BLECharacteristic::PROPERTY_NOTIFY |
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_WRITE  
                    );
                    
  // Create a Characteristic for the tilt rotation
  pTilt = pService->createCharacteristic(
                      BLEUUID(TILT_UUID),
                      BLECharacteristic::PROPERTY_NOTIFY |
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_WRITE  
                    );

  pLED->setCallbacks(new LEDCharacteristicCallback());
  pPan->setCallbacks(new PANCharacteristicCallback());
  pTilt->setCallbacks(new TILTCharacteristicCallback());
  
  pLED->addDescriptor(new BLE2902());
  pPan->addDescriptor(new BLE2902());
  pTilt->addDescriptor(new BLE2902());

  // Start the service
  pService->start();

  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");
}`

**This ist in the setup(). I created the UUID`s with the online generator:**
`
#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define LED_UUID            "beb5483e-36e1-4688-b7f5-ea07361b26a8"
#define PAN_UUID            "a8d5db42-d652-422c-af30-6f61c8f9ad19"
#define TILT_UUID           "c89d5cce-14b3-4992-b8a8-234a273010c3"
`

**Problem:**
I am only able to see the pLED characteristic, the other two are missing, what am i doing wrong?
Unfortunatly there is no example code for multiple ones. 

**What i have done so far:**
I cloned this repo and build it with make and added it to the arduino IDE, so there should be the latest library.

chegewara, 2021-01-07:
Code seems to be ok, what tool are you using to test?

HoernchenJ, 2021-01-07:
Thanks for your quick reply.
I am using the Android App nRF Connect and just the verbose debug messages from the serial monitor in the arduino IDE.

**This is a log from the arduino serial monitor:**

`21:53:18.609 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:53:18.609 -> configsip: 0, SPIWP:0xee
21:53:18.609 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:53:18.609 -> mode:DIO, clock div:1
21:53:18.609 -> load:0x3fff0018,len:4
21:53:18.609 -> load:0x3fff001c,len:1216
21:53:18.609 -> ho 0 tail 12 room 4
21:53:18.609 -> load:0x40078000,len:9720
21:53:18.609 -> ho 0 tail 12 room 4
21:53:18.609 -> load:0x40080400,len:6352
21:53:18.609 -> entry 0x400806b8
21:53:19.594 -> [D][BLEDevice.cpp:80] createServer(): >> createServer
21:53:19.594 -> [D][BLEServer.cpp:291] registerApp(): >> registerApp - 0
21:53:19.594 -> [D][FreeRTOS.cpp:164] take(): Semaphore taking: name: RegisterAppEvt (0x3ffddf84), owner: <N/A> for registerApp
21:53:19.641 -> [D][FreeRTOS.cpp:173] take(): Semaphore taken:  name: RegisterAppEvt (0x3ffddf84), owner: registerApp
21:53:19.641 -> [V][FreeRTOS.cpp:69] wait(): >> wait: Semaphore waiting: name: RegisterAppEvt (0x3ffddf84), owner: registerApp for registerApp
21:53:19.641 -> [D][BLEDevice.cpp:106] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_REG_EVT
21:53:19.641 -> [V][BLEUtils.cpp:1517] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_REG_EVT
21:53:19.641 -> [V][BLEUtils.cpp:1707] dumpGattServerEvent(): dumpGattServerEvent: *** NOT CODED ***
21:53:19.685 -> [D][BLEServer.cpp:154] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_REG_EVT
21:53:19.685 -> [V][FreeRTOS.cpp:119] give(): Semaphore giving: name: RegisterAppEvt (0x3ffddf84), owner: registerApp
21:53:19.685 -> [D][BLEServer.cpp:281] handleGATTServerEvent(): << handleGATTServerEvent
21:53:19.685 -> [V][FreeRTOS.cpp:85] wait(): << wait: Semaphore released: name: RegisterAppEvt (0x3ffddf84), owner: <N/A>
21:53:19.685 -> [D][BLEServer.cpp:295] registerApp(): << registerApp
21:53:19.685 -> [D][BLEDevice.cpp:87] createServer(): << createServer
**21:53:19.685 -> [D][BLEServer.cpp:77] createService(): >> createService - 4fafc201-1fb5-459e-8fcc-c5c9c331914b**
21:53:19.731 -> [D][FreeRTOS.cpp:164] take(): Semaphore taking: name: CreateEvt (0x3ffde1dc), owner: <N/A> for createService
21:53:19.731 -> [D][FreeRTOS.cpp:173] take(): Semaphore taken:  name: CreateEvt (0x3ffde1dc), owner: createService
21:53:19.731 -> [D][BLEService.cpp:67] executeCreate(): >> executeCreate() - Creating service (esp_ble_gatts_create_service) service uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b
21:53:19.731 -> [D][FreeRTOS.cpp:164] take(): Semaphore taking: name: CreateEvt (0x3ffde4a0), owner: <N/A> for executeCreate
21:53:19.778 -> [D][FreeRTOS.cpp:173] take(): Semaphore taken:  name: CreateEvt (0x3ffde4a0), owner: executeCreate
21:53:19.778 -> [V][FreeRTOS.cpp:69] wait(): >> wait: Semaphore waiting: name: CreateEvt (0x3ffde4a0), owner: executeCreate for executeCreate
21:53:19.778 -> [D][BLEDevice.cpp:106] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CREATE_EVT
21:53:19.778 -> [V][BLEUtils.cpp:1517] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CREATE_EVT
21:53:19.778 -> [V][BLEUtils.cpp:1707] dumpGattServerEvent(): dumpGattServerEvent: *** NOT CODED ***
21:53:19.825 -> [D][BLEServer.cpp:154] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CREATE_EVT
21:53:19.825 -> [V][FreeRTOS.cpp:119] give(): Semaphore giving: name: CreateEvt (0x3ffde1dc), owner: createService
21:53:19.825 -> [D][BLEService.cpp:201] setHandle(): >> setHandle - Handle=0x28, service UUID=4fafc201-1fb5-459e-8fcc-c5c9c331914b)
21:53:19.825 -> [D][BLEService.cpp:207] setHandle(): << setHandle
21:53:19.825 -> [V][FreeRTOS.cpp:119] give(): Semaphore giving: name: CreateEvt (0x3ffde4a0), owner: executeCreate
21:53:19.825 -> [D][BLEServer.cpp:281] handleGATTServerEvent(): << handleGATTServerEvent
21:53:19.873 -> [V][FreeRTOS.cpp:85] wait(): << wait: Semaphore released: name: CreateEvt (0x3ffde4a0), owner: <N/A>
21:53:19.873 -> [D][BLEService.cpp:83] executeCreate(): << executeCreate
21:53:19.873 -> [V][FreeRTOS.cpp:69] wait(): >> wait: Semaphore waiting: name: CreateEvt (0x3ffde1dc), owner: <N/A> for createService
**21:53:19.873 -> [V][FreeRTOS.cpp:85] wait(): << wait: Semaphore released: name: CreateEvt (0x3ffde1dc), owner: createService
21:53:19.873 -> [D][BLEServer.cpp:93] createService(): << createService
21:53:19.873 -> [D][BLEService.cpp:229] addCharacteristic(): >> addCharacteristic()
21:53:19.873 -> [D][BLEService.cpp:232] addCharacteristic(): Adding characteristic: uuid=beb5483e-36e1-4688-b7f5-ea07361b26a8 to service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
21:53:19.920 -> [D][BLEService.cpp:244] addCharacteristic(): << addCharacteristic()
21:53:19.920 -> [D][BLEService.cpp:229] addCharacteristic(): >> addCharacteristic()
21:53:19.920 -> [D][BLEService.cpp:232] addCharacteristic(): Adding characteristic: uuid=a8d5db42-d652-422c-af30-6f61c8f9ad19 to service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
21:53:19.920 -> [D][BLEService.cpp:244] addCharacteristic(): << addCharacteristic()
21:53:19.967 -> [D][BLEService.cpp:229] addCharacteristic(): >> addCharacteristic()
[... cut ...]
21:53:43.141 -> [D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
21:53:43.141 -> [D][BLECharacteristic.cpp:466] handleGATTServerEvent(): << handleGATTServerEvent
21:53:43.141 -> [D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
21:53:43.141 -> [D][BLECharacteristic.cpp:466] handleGATTServerEvent(): << handleGATTServerEvent
21:53:43.188 -> [D][BLEServer.cpp:281] handleGATTServerEvent(): << handleGATTServerEvent`

chegewara, 2021-01-07:
Is it possible that nRF connect remember previous data?
The thing is that android is caching services and characteristics. When you connect in top right corner you have option to refresh it.

HoernchenJ, 2021-01-07:
Well, that fixed the problem 👍 
Thank you very much and stay healthy :D

commit 810aeb81 (2019-01-30): Merge pull request #800 from chegewara/master

Few bugfixes and add functions
Fix #796 , fix #786 fix #744 fix #751