nkolban/esp32-snippets #753: BLERemoteCharacteristic::writeValue

espresso-if, 2018-12-16:
Hello,
it's me again. ✋ 
When calling   `void BLERemoteCharacteristic::writeValue(uint8_t* data, size_t length, bool response);`  the function seems to stuck at `m_semaphoreWriteCharEvt.take("writeValue");`.

When I remove the semaphore  everything works. (I kown, it's risky)

My environment: Win10 Platformio 1.6 , built-in ESP-IDF; running on ESP Dev Board

chegewara, 2018-12-17:
Welcome back.
One question, did you check you are trying to write characteristic is exposing write properties, and if it does, is it normal write or write no response?
Also do you send request with response set to true or false.

chegewara, 2018-12-18:
I have some suspicions. Could you try to add some delay in your code right after write request?
Just for test purpose 1000ms or so. Thanks.

espresso-if, 2018-12-19:
I tested it and it has no effect.
```
		pCharacteristic->writeValue(str);
		vTaskDelay(1000/portTICK_RATE_MS);
```