nkolban/esp32-snippets #754: Debugging Strange ESP32 behavior

afnan, 2018-12-17:
Hi,
I am using your BLE sketches form ESP32 device in Arduino IDE. I wrote a simple program to advertise and send random int after connection. This program works fine on one board but, if I program another board with the same code, the receiver (i believe server?) does not receive the data. Now, I am sure there is something wrong with the board as if a code works on one it should work on another too. 

Just to learn out of it I did some digging. Following are my findings.
1. GATT connect and disconnect works fine and the board detects the connection
2. On using nRFConnect app, I see the BLE being advertised and I can connect to it
3. Strangely upon connection I only see Generic Attribute not others.
4. If I write the same program to other board, I see all the attributes I am exposing. 

I understand this cannot be an easy diagnosis but I am curious why the board works fine i.e reads sensors and stuff but does not expose all services ?

any heading will be appreciated. 

Thanks

chegewara, 2018-12-17:
Hi, very intriguing. 
Lets start from beginning, what kind boards are you using?

afnan, 2018-12-21:
Hi, 
Sorry for late reply. I am using board by Peskey Products https://www.tindie.com/products/onehorse/smallest-esp32-development-board/. These boards have served me well. They work without dramas and this peculiar behavior was something that did not make any sense.

Upon connecting to the android application (works with rest of the boards) the board does connect to GATT and disconnect as well, but the data sent is never received by either nrfConnect or my device