nkolban/esp32-snippets #757: ESP32 BLE does not connect to Bluetooth LE Lamp (Magic Blue)

don41382, 2018-12-17:
**Arduino Code**

    void setup() {
      Serial.begin(115200);
      Serial.println("Starting"); 

      BLEDevice::init("ESP32");
      Serial.println("Init Bluetooth");

      BLEAddress address("ee:32:4f:a0:b4:06");
      BLEClient*  pClient  = BLEDevice::createClient();
      bool success = pClient->connect(address);
    }

**Serial Output**
```
Init Bluetooth
[D][BLEDevice.cpp:66] createClient(): >> createClient
[D][BLEDevice.cpp:72] createClient(): << createClient
[D][BLEClient.cpp:103] connect(): >> connect(ee:32:4f:a0:b4:06)
[I][BLEDevice.cpp:600] addPeerDevice(): add conn_id: 0, GATT role: client
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: RegEvt (0x3ffe7c74), owner: <N/A> for connect
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: RegEvt (0x3ffe7c74), owner: connect
[V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: RegEvt (0x3ffe7c74), owner: connect for connect
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: RegEvt (0x3ffe7c74), owner: connect
[V][FreeRTOS.cpp:86] wait(): << wait: Semaphore released: name: RegEvt (0x3ffe7c74), owner: connect
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: OpenEvt (0x3ffe7eb8), owner: <N/A> for connect
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: OpenEvt (0x3ffe7eb8), owner: connect
[V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: OpenEvt (0x3ffe7eb8), owner: connect for connect
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[I][BLEDevice.cpp:611] removePeerDevice(): remove: 0, GATT role client
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: RssiCmplEvt (0x3ffe7f78), owner: <N/A>
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: SearchCmplEvt (0x3ffe7f18), owner: <N/A>
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
```

chegewara, 2018-12-17:
Sorry for confusion about `Unknown client events`. I wanted to make binary size smaller when you are using `DEBUG` log level or less but i made some mistake in code and all debug messages are now unreadable.
client event 2 is `open_evt` and 41 is `disconnect_evt`.

So it looks like your esp32 is disconnecting even before connection is established. You can try to add this code to setup():
```
		BLEDevice::setCustomGapHandler(my_gap_event_handler);
		BLEDevice::setCustomGattcHandler(my_gattc_event_handler);
```
and those 2 functions:
```
static void my_gap_event_handler(esp_gap_ble_cb_event_t  event, esp_ble_gap_cb_param_t* param) {
	ESP_LOGW(LOG_TAG, "custom gap event handler, event: %d", (uint8_t)event);
}

static void my_gattc_event_handler(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gattc_cb_param_t* param) {
	ESP_LOGW(LOG_TAG, "custom gattc event handler, event: %d", (uint8_t)event);
}
```
I am using this code to debug every BLE code i write now. In arduino you can change ESP_LOGx with Serial.println().

don41382, 2018-12-18:
Thanks for your quick response @chegewara. I added your recommended handlers.

The new output is:

```
Init Bluetooth
[D][BLEDevice.cpp:66] createClient(): >> createClient
[D][BLEDevice.cpp:72] createClient(): << createClient
[D][BLEClient.cpp:103] connect(): >> connect(ee:32:4f:a0:b4:06)
[I][BLEDevice.cpp:600] addPeerDevice(): add conn_id: 0, GATT role: client
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: RegEvt (0x3ffe7d04), owner: <N/A> for connect
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: RegEvt (0x3ffe7d04), owner: connect
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[W][esp32-ble-device-list.ino:39] my_gattc_event_handler(): custom gattc event handler, event: 0
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: OpenEvt (0x3ffe7f48), owner: <N/A> for connect
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: OpenEvt (0x3ffe7f48), owner: connect
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[I][BLEDevice.cpp:611] removePeerDevice(): remove: 0, GATT role client
[W][esp32-ble-device-list.ino:39] my_gattc_event_handler(): custom gattc event handler, event: 41
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[W][esp32-ble-device-list.ino:39] my_gattc_event_handler(): custom gattc event handler, event: 2
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[W][esp32-ble-device-list.ino:39] my_gattc_event_handler(): custom gattc event handler, event: 1
```

Still not sure how to proceed or fix this issue. Can you please help?

chegewara, 2018-12-18:
https://github.com/espressif/esp-idf/blob/master/components/bt/bluedroid/api/include/api/esp_gattc_api.h#L27-L70

Its hard to say, without part of code or device and cant help too much.

don41382, 2018-12-18:
After investigating more time, I've found the problem.

The signature of BLEClient->connect is the [following](https://github.com/nkolban/esp32-snippets/blob/2bb21f020b1c6d82af6b47acf056adb0361c3ff9/cpp_utils/BLEClient.h#L37):
```bool connect(BLEAddress address, esp_ble_addr_type_t type = BLE_ADDR_TYPE_PUBLIC);```

So by default it uses the `BLE_ADDR_TYPE_PUBLIC`, but my BLE light bulb needed the type `BLE_ADDR_TYPE_RANDOM`.

**Working code**

    BLEClient*  pClient  = BLEDevice::createClient();
    bool success = pClient->connect(address,BLE_ADDR_TYPE_RANDOM);

Can you quickly explain, what the difference is?

closed 2018-12-18 by don41382

chegewara, 2018-12-18:
@don41382 Public address is own address that is individual for each device and its static. Random address is used mostly to avoid device tracking but the main difference is that its changing usually every time device is powered up or start new advertising. You can see that most if not all smartphones, tablets and laptops are using random addressing. Thats why i introduced this in current version of library. There should be also BLEClient->connect() function where you can just pass advertised device instead of address and device address type and all is done by library.

don41382, 2018-12-19:
Thanks for the clarification. The advertising device was actually the critical hint. My BLE light bulb worked inside of the `BLE_client` example. I discovered that the `connect(advertisedDevice)` passes the [address type](https://github.com/nkolban/esp32-snippets/blob/2bb21f020b1c6d82af6b47acf056adb0361c3ff9/cpp_utils/BLEClient.cpp#L90).

I think it would have helped me, if the connect would have been more explicit. The address type `BLE_ADDR_TYPE_PUBLIC` should not set by default.

chegewara, 2018-12-19:
Its for backward compatibility. Also in earlier version of library it was hardcoded, thats why i am very surprised there is some issue with it.

anandharaj-dotworld, 2021-07-13:
i also getting a same error but i can't know how to fix this issue. @don41382 Can you help me to fix that issue.

don41382, 2021-07-13:
@anandharaj-dotworld sorry, it's been a while and I don't have the setup anymore. But did you read my last message?

https://github.com/nkolban/esp32-snippets/issues/757#issuecomment-448336410

If this doesn't help - I've no clue. Good luck.

anandharaj-dotworld, 2021-07-13:
[D][BLEClient.cpp:180] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 0] ... Unknown
After i received this error, then process has stoped. how to resolve this.

chegewara, 2021-07-13:
Maybe its stupid question, but where do you see error in this message?
`[D]` means debug

anandharaj-dotworld, 2021-07-13:
Yes, I received this in debug[D].

anandharaj-dotworld, 2021-07-13:
@chegewara . my code perfectly receiving characteristics value  in 3 times. But 4th time I can't receive. How to solve this. And I using android application for ble notifier.

chegewara, 2021-07-13:
There is nothing wrong with that message.

anandharaj-dotworld, 2021-07-13:
But why I received esp_GATT_if: 0. I receiving 1st time in esp_GATT_if: 4 and second time esp_GATT_if: 5 and third time esp_GATT_if: 6 and fourth time is esp_GATT_if: 0.

anandharaj-dotworld, 2021-07-13:
4th time take long time to connect server.

chegewara, 2021-07-13:
I dont know why you have this problem, just like i dont know you are not related logs and expecting someone help you with issue we have no idea about.

anandharaj-dotworld, 2021-07-13:
@chegewara Can I share my code.

anandharaj-dotworld, 2021-07-13:
@chegewara 

#include "BLEDevice.h"
//#include "BLEScan.h"

static BLEUUID serviceUUID("fd05a570-274a-4b1f-a5a3-eb52e5e02b8b");
static BLEUUID    charUUID("53447ca9-1e5b-448e-ab7b-bd9438c048af");

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;
BLEClient*  pClient;

BLEScan* pBLEScan;
class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pClient) {
    Serial.println("connect client..");
  }

  void onDisconnect(BLEClient* pClient) {
    Serial.println("onDisconnect");
    connected = false;
  }
};

bool connectToServer() {
    Serial.print("Forming a connection to ");
    Serial.println(myDevice->getAddress().toString().c_str());
    pClient  = BLEDevice::createClient();
    Serial.println("Client was created");
    pClient->setClientCallbacks(new MyClientCallback());
    bool a = pClient->connect(myDevice);
    Serial.println(a);
    //pClient->connect(myDevice);
    Serial.println(myDevice->getAddress().toString().c_str());
    Serial.println("Server Created");
    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == false) {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      pClient->disconnect();
      return false;
    }
    Serial.println("Foundedservice");
    pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
    if (pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
      Serial.println(charUUID.toString().c_str());
      pClient->disconnect();
      return false;
    }
    Serial.println("Founded characteristic");
    if(pRemoteCharacteristic->canRead()) {
      std::string value = pRemoteCharacteristic->readValue();
      Serial.print("The characteristic value: ");
      Serial.println(value.c_str());
    }

    connected = true;
    return true;
    
}
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    Serial.print("BLE Advertised Device found: ");
    Serial.println(advertisedDevice.toString().c_str());
    if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID)) {
      Serial.println("Scan stoped");
      Serial.println(ESP.getFreeHeap());

      BLEDevice::getScan()->stop();
      myDevice = new BLEAdvertisedDevice(advertisedDevice); //advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID)
      doConnect = true;
      doScan = true;

    }
  } 
}; 

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");
  BLEDevice::init("");
  
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  //pBLEScan->setInterval(100000);
  //pBLEScan->setWindow(10000);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(0);
  
  
} 
void loop() {
  if (doConnect == true) {
    if (connectToServer()) {
      Serial.println("Connected to the BLE Server.");
    } else {
      Serial.println("Failed to connect to the server");
      
    }
    doConnect = false;
  }
  BLEDevice::getScan()->start(0);
}