nkolban/esp32-snippets #760: BLE: Use classes for callback in characteristic notify

MariusSchiffer, 2018-12-18:
BLERemoteCharacteristics takes a function pointer as callback. It would be nice to allow passing a class (similar to BLEClient etc.), as this allows keeping track of state easier. (Without an extra global variable or other workaround).

I have implemented such a change in my fork, but it is not cleaned up yet.

chegewara, 2018-12-18:
something like this?
https://github.com/nkolban/esp32-snippets/pull/752/files

hansmbakker, 2018-12-20:
I believe @MariusSchiffer's point is that the `notify_callback` type as used in https://github.com/nkolban/esp32-snippets/blob/c48cb19186744f5792b37060b4ae9b1c36b422df/cpp_utils/BLERemoteCharacteristic.cpp#L443 is limited to static functions currently.

It would be nice if the library allowed us to use classes instead, like https://github.com/nkolban/esp32-snippets/blob/c48cb19186744f5792b37060b4ae9b1c36b422df/cpp_utils/BLEClient.h#L53

MariusSchiffer, 2018-12-21:
That's exactly how I meant it.

chegewara, 2018-12-21:
ok, PR is always welcome.