nkolban/esp32-snippets #762: Change the ble::init after each loop in esp32 arduino

wan420, 2018-12-19:
I am trying to change the bluetooth name in  ble::init() everytime it advertises. 
this is the code I am following, its a very poor way of coding and also I am not sure whether the library is coded to make this happen.

None the less, this is my code : 
```
void loop() {
  // put your main code here, to run repeatedly:
  String name = "Minibeacon_00"+String(no);
  Serial.println(name);
  delay(100);
  BLEDevice::init(name.c_str());
  Serial.println("after ble init line");
  BLEServer *pServer = BLEDevice::createServer();
  pAdvertising = pServer->getAdvertising();
  pAdvertising->start();
  //Serial.println("Characteristic defined! Now you can read it in your phone!");
  delay(5000);
  pAdvertising->stop();
  no++;
  BLEDevice::init(name.c_str());
}
```

I want to change the value of Ble::init() everytime the loop starts, is it possible to do? 
Thanks.

chegewara, 2018-12-19:
Maybe this is easier for you:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertising.cpp

https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/examples/BLE_iBeacon/BLE_iBeacon.ino

wan420, 2018-12-19:
Sorry, I will close this , not exactly an issue

closed 2018-12-19 by wan420

chegewara, 2018-12-19:
@wan420 No problem, you were looking for help and i hope you found it.