nkolban/esp32-snippets #765: Crash in retrieveCharacteristics() in BLERemoteService

hansmbakker, 2018-12-20:
``
Arduino BLE Client application...
[D][BLEScan.cpp:204] start(): >> start(duration=30)
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: ScanEnd (0x3ffd1f0c), owner: <N/A> for start
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: ScanEnd (0x3ffd1f0c), owner: start
[D][BLEScan.cpp:236] start(): << start()
[D][BLEAdvertisedDevice.cpp:424] setRSSI(): - setRSSI(): rssi: -65
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x03 (), length: 2, data: 9ffe
[D][BLEAdvertisedDevice.cpp:453] setServiceUUID(): - addServiceUUID(): serviceUUID: 0000fe9f-0000-1000-8000-00805f9b34fb
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x16 (), length: 22, data: 9ffe0264365a5738496c7a6e4f4d00000167cd61d6da
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0xff (), length: 8, data: e000000aca74caf5
[D][BLEAdvertisedDevice.cpp:401] setManufacturerData(): - manufacturer data: e000000aca74caf5
BLE Advertised Device found: Name: , Address: 6c:d9:e3:ca:27:46, manufacturer data: e000000aca74caf5, serviceUUID: 0000fe9f-0000-1000-8000-00805f9b34fb
[D][BLEAdvertisedDevice.cpp:424] setRSSI(): - setRSSI(): rssi: -68
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0xff (), length: 27, data: 06000109210a0314b6ed9dc44445534b544f502d514c534c503250
[D][BLEAdvertisedDevice.cpp:401] setManufacturerData(): - manufacturer data: 06000109210a0314b6ed9dc44445534b544f502d514c534c503250
BLE Advertised Device found: Name: , Address: 67:fd:86:38:63:70, manufacturer data: 06000109210a0314b6ed9dc44445534b544f502d514c534c503250
[D][BLEAdvertisedDevice.cpp:424] setRSSI(): - setRSSI(): rssi: -89
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0xff (), length: 9, data: 4c0010050b1cddbb9a
[D][BLEAdvertisedDevice.cpp:401] setManufacturerData(): - manufacturer data: 4c0010050b1cddbb9a
BLE Advertised Device found: Name: , Address: 7c:93:06:63:17:c1, manufacturer data: 4c0010050b1cddbb9a
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 67:fd:86:38:63:70, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 67:fd:86:38:63:70, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 6c:d9:e3:ca:27:46, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 67:fd:86:38:63:70, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 7c:93:06:63:17:c1, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 67:fd:86:38:63:70, already seen it.
[D][BLEScan.cpp:106] handleGAPEvent(): Ignoring 67:fd:86:38:63:70, already seen it.
[D][BLEAdvertisedDevice.cpp:424] setRSSI(): - setRSSI(): rssi: -75
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x09 (), length: 5, data: 4e75696d6f
[D][BLEAdvertisedDevice.cpp:413] setName(): - setName(): name: Nuimo
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x19 (), length: 2, data: 8001
[D][BLEAdvertisedDevice.cpp:389] setAppearance(): - appearance: 384
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:253] parseAdvertisement(): Type: 0x03 (), length: 4, data: 0f180a18
[D][BLEAdvertisedDevice.cpp:453] setServiceUUID(): - addServiceUUID(): serviceUUID: 0000180f-0000-1000-8000-00805f9b34fb
[D][BLEAdvertisedDevice.cpp:453] setServiceUUID(): - addServiceUUID(): serviceUUID: 0000180a-0000-1000-8000-00805f9b34fb
BLE Advertised Device found: Name: Nuimo, Address: d3:ce:97:9d:b3:a7, appearance: 384, serviceUUID: 0000180f-0000-1000-8000-00805f9b34fb
Found our device!  address: [D][BLEScan.cpp:259] stop(): >> stop()
[D][BLEScan.cpp:271] stop(): << stop()
Forming a connection to d3:ce:97:9d:b3:a7
[D][BLEDevice.cpp:62] createClient(): >> createClient
[D][BLEDevice.cpp:68] createClient(): << createClient
 - Created client
[D][BLEClient.cpp:103] connect(): >> connect(d3:ce:97:9d:b3:a7)
[I][BLEDevice.cpp:596] addPeerDevice(): add conn_id: 0, GATT role: client
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: RegEvt (0x3ffe82b8), owner: <N/A> for connect
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: RegEvt (0x3ffe82b8), owner: connect
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: OpenEvt (0x3ffe8668), owner: <N/A> for connect
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: OpenEvt (0x3ffe8668), owner: connect
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:580] updatePeerDevice(): update conn_id: 4, GATT role: client
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:136] connect(): << connect(), rc=1
 - Connected to server
[D][BLEClient.cpp:383] getService(): >> getService: uuid: f29b1525-cb19-40f3-be5c-7241ecb82fd2
[D][BLEClient.cpp:419] getServices(): >> getServices
[D][BLEClient.cpp:78] clearServices(): >> clearServices
[D][BLEClient.cpp:85] clearServices(): << clearServices
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: SearchCmplEvt (0x3ffe8a2c), owner: <N/A> for getServices
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: SearchCmplEvt (0x3ffe8a2c), owner: getServices
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLERemoteService.cpp:32] BLERemoteService(): >> BLERemoteService()
[D][BLERemoteService.cpp:40] BLERemoteService(): << BLERemoteService()
[... 16 lines cut ...]
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLERemoteService.cpp:32] BLERemoteService(): >> BLERemoteService()
[D][BLERemoteService.cpp:40] BLERemoteService(): << BLERemoteService()
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:436] getServices(): << getServices
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:396] getService(): << getService: found the service with uuid: f29b1525-cb19-40f3-be5c-7241ecb82fd2
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
 - Found our service
[D][BLERemoteService.cpp:168] retrieveCharacteristics(): >> getCharacteristics() for service: f29b1525-cb19-40f3-be5c-7241ecb82fd2
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40081028  PS      : 0x00060b30  A0      : 0x800d566d  A1      : 0x3ffd5c40
A2      : 0x1940f3be  A3      : 0x00000000  A4      : 0x0000001b  A5      : 0x0000ffff
A6      : 0x3ffd5cb2  A7      : 0x3ffd5c9c  A8      : 0x1940f3be  A9      : 0x3ffd5c00
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00000000
A14     : 0x3ffd5c9c  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x1940f3be  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

Backtrace: 0x40081028:0x3ffd5c40 0x400d566a:0x3ffd5c60 0x400d5a0e:0x3ffd5d20 0x400d1a2f:0xf3be5c72

Rebooting...
ets Jun  8 2016 00:22:57

hansmbakker, 2018-12-23:
The issue seems to be in this call: https://github.com/nkolban/esp32-snippets/blob/c48cb19186744f5792b37060b4ae9b1c36b422df/cpp_utils/BLERemoteService.cpp#L173-L181

The code runs into the invalid offset statement at https://github.com/nkolban/esp32-snippets/blob/c48cb19186744f5792b37060b4ae9b1c36b422df/cpp_utils/BLERemoteService.cpp#L183-L185 so it calls `break;`, then the code causes a `Stack smashing protect failure!` before https://github.com/nkolban/esp32-snippets/blob/c48cb19186744f5792b37060b4ae9b1c36b422df/cpp_utils/BLERemoteService.cpp#L155

This issue is not present in arduino-esp32 1.0.0 (non-Platform.IO, just using Arduino IDE directly with esp32 1.0.0 from board manager).

chegewara, 2018-12-29:
Hi @hansmbakker 
sorry for late reply. I cant reproduce this with arduino-ide and your test repository, with both ble v1.0.0 and ble v1.0.1. I am not working with PlatformIO, sorry.

wakwak-koba, 2019-01-14:
https://github.com/nkolban/esp32-snippets/blob/c48cb19186744f5792b37060b4ae9b1c36b422df/cpp_utils/BLERemoteService.cpp#L170-L172

was (maybe) wrong.
If you want to do this, you'd need to 
```
esp_gattc_char_elem_t result[10];
```

For example...
https://pbs.twimg.com/media/Dwv3pnaVAAAk3kk.jpg:large

Some bugs I found were fixed.
https://github.com/wakwak-koba/ESP32_BLE_Arduino
https://github.com/nkolban/ESP32_BLE_Arduino/compare/master...wakwak-koba:master

chegewara, 2019-01-14:
Hi @wakwak-koba 
thanks for finding this. If you can make PR in this repository i will merge it.

DeqingSun, 2019-01-25:
> [esp32-snippets/cpp_utils/BLERemoteService.cpp](https://github.com/nkolban/esp32-snippets/blob/c48cb19186744f5792b37060b4ae9b1c36b422df/cpp_utils/BLERemoteService.cpp#L170-L172)
> 
> Lines 170 to 172 in [c48cb19](/nkolban/esp32-snippets/commit/c48cb19186744f5792b37060b4ae9b1c36b422df)
> 
>  esp_gattc_char_elem_t result; 
>  while (true) { 
>  	uint16_t count = 10;  // this value is used as in parameter that allows to search max 10 chars with the same uuid 
> was (maybe) wrong.
> If you want to do this, you'd need to
> 
> ```
> esp_gattc_char_elem_t result[10];
> ```
> For example...
> https://pbs.twimg.com/media/Dwv3pnaVAAAk3kk.jpg:large
> 
> Some bugs I found were fixed.
> https://github.com/wakwak-koba/ESP32_BLE_Arduino
> [nkolban/ESP32_BLE_Arduino@master...wakwak-koba:master](https://github.com/nkolban/ESP32_BLE_Arduino/compare/master...wakwak-koba:master)

I triggered a similar bug when I try to connect a Surface Dial.
In BLERemoteService::getCharacteristic, the uuid got modified after retrieveCharacteristics() is called (then cpu crash). This seems an out of bounds problem in retrieveCharacteristics()

this commit seems fixes it.
https://github.com/wakwak-koba/ESP32_BLE_Arduino/commit/b1ef06eee327062351a05c4159576ba0d66ca380#diff-95400a504c71c5204a517b0ed85fcd1c

chegewara, 2019-01-25:
@DeqingSun this PR has been merged to this library, just not pushed to Arduino yet.
I have to fix few more issues before i will do it.