nkolban/esp32-snippets #767: Missed GATTC_OPEN_EVT causes BLEClient::connect() to wait indefinitely

m0x72, 2018-12-21:
After a `esp_ble_gattc_open()` request in `BLEClient::connect()`, the GATTC_CONNECT_EVT is fired but the GATTC_OPEN_EVT is occasionally not fired. This causes BLEClient::connect() to wait indefinitely for the semaphore `m_semaphoreOpenEvt` to clear in `BLEClient::gattClientEventHandler()`.

Would suggest to replace `m_semaphoreOpenEvt.wait("connect")` with a semaphore with timeout and return false after timeout (resp. `m_isConnected state`). A user can then try to reconnect with the client.

This behavior is rare and happens approximately after reconnecting ~100 times. 

Similar to #744, but not caused due to a GATTC_DISCONNECT_EVT. The Server get's its expected GATTS_CONNECT_EVT and even a subsequent GATTS_MTU_EVT.

Client Log (Expected w/ open evt)
```
13:38:30.246 -> [D][BLEClient.cpp:119] connect(): >> connect(3c:71:bf:96:f6:b6)
13:38:30.246 -> [V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:38:30.246 -> [D][FreeRTOS.cpp:165] take(): Semaphore taking: name: OpenEvt (0x3ffe7f48), owner: <N/A> for connect
13:38:30.246 -> [D][BLEClient.cpp:470] handleGAPEvent(): BLEClient ... handling GAP event!
13:38:30.246 -> [D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: OpenEvt (0x3ffe7f48), owner: connect
13:38:30.279 -> [V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 3 0x03
13:38:30.279 -> [V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: OpenEvt (0x3ffe7f48), owner: connect for connect
13:38:30.279 -> [V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
13:38:30.279 -> [V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:38:30.279 -> [D][BLEClient.cpp:470] handleGAPEvent(): BLEClient ... handling GAP event!
13:38:30.279 -> [V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 3 0x03
13:38:30.279 -> [V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
13:38:30.279 -> [V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:38:30.279 -> [D][BLEClient.cpp:470] handleGAPEvent(): BLEClient ... handling GAP event!
13:38:30.279 -> [V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 18 0x12
13:38:30.279 -> [V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
13:38:30.279 -> [V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:38:30.279 -> [D][BLEClient.cpp:470] handleGAPEvent(): BLEClient ... handling GAP event!
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 40
13:38:30.345 -> [D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
13:38:30.345 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 40
13:38:30.345 -> [V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
13:38:30.345 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 40
13:38:30.345 -> [D][BLEClient.cpp:167] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
13:38:30.345 -> [D][BLEDevice.cpp:580] updatePeerDevice(): update conn_id: 4, GATT role: client
13:38:30.345 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2
13:38:30.345 -> [D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
13:38:30.345 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2
13:38:30.378 -> [V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
13:38:30.378 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2
13:38:30.378 -> [D][BLEClient.cpp:167] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
13:38:30.378 -> [V][FreeRTOS.cpp:120] give(): Semaphore giving: name: OpenEvt (0x3ffe7f48), owner: connect
13:38:30.378 -> [V][FreeRTOS.cpp:86] wait(): << wait: Semaphore released: name: OpenEvt (0x3ffe7f48), owner: connect
13:38:30.378 -> [D][BLEClient.cpp:137] connect(): << connect(), rc=1
```

Client log (Unexpected - no open evt)
```
13:38:35.125 -> [D][BLEClient.cpp:119] connect(): >> connect(3c:71:bf:96:f6:b6)
13:38:35.125 -> [V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:38:35.125 -> [D][FreeRTOS.cpp:165] take(): Semaphore taking: name: OpenEvt (0x3ffe7f48), owner: <N/A> for connect
13:38:35.125 -> [D][BLEClient.cpp:470] handleGAPEvent(): BLEClient ... handling GAP event!
13:38:35.125 -> [D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: OpenEvt (0x3ffe7f48), owner: connect
13:38:35.125 -> [V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 18 0x12
13:38:35.125 -> [V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: OpenEvt (0x3ffe7f48), owner: connect for connect
13:38:35.125 -> [V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
13:38:35.159 -> [V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:38:35.159 -> [D][BLEClient.cpp:470] handleGAPEvent(): BLEClient ... handling GAP event!
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 40
13:38:35.225 -> [D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
13:38:35.225 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 40
13:38:35.225 -> [V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
13:38:35.225 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 40
13:38:35.225 -> [D][BLEClient.cpp:167] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
13:38:35.225 -> [D][BLEDevice.cpp:580] updatePeerDevice(): update conn_id: 4, GATT role: client
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 18
13:38:35.656 -> [D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
13:38:35.656 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 18
13:38:35.689 -> [V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
13:38:35.689 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 18
[... 17 lines cut ...]
13:38:35.225 -> [D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
13:38:35.225 -> [D][BLECharacteristic.cpp:459] handleGATTServerEvent(): << handleGATTServerEvent
13:38:35.225 -> [D][BLEServer.cpp:280] handleGATTServerEvent(): << handleGATTServerEvent
[V][BLEUtils.cpp:1018] gattServerEventTypeToString(): Unknown GATT Server event type: 4
13:38:35.656 -> [D][BLEDevice.cpp:104] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
13:38:35.656 -> [V][BLEUtils.cpp:1018] gattServerEventTypeToString(): Unknown GATT Server event type: 4
13:38:35.656 -> [V][BLEUtils.cpp:1532] dumpGattServerEvent(): GATT ServerEvent: Unknown
13:38:35.656 -> [V][BLEUtils.cpp:1722] dumpGattServerEvent(): dumpGattServerEvent: *** NOT CODED ***
13:38:35.656 -> [V][BLEUtils.cpp:1018] gattServerEventTypeToString(): Unknown GATT Server event type: 4
13:38:35.656 -> [D][BLEServer.cpp:153] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
13:38:35.656 -> [V][BLEUtils.cpp:1018] gattServerEventTypeToString(): Unknown GATT Server event type: 4
13:38:35.656 -> [D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
13:38:35.656 -> [D][BLECharacteristic.cpp:459] handleGATTServerEvent(): << handleGATTServerEvent
13:38:35.656 -> [D][BLEServer.cpp:280] handleGATTServerEvent(): << handleGATTServerEvent
```
For reference: 
GATTC_CONNECT_EVT - 40
GATTC_OPEN_EVT - 2
GATTC_CFG_MTU_EVT - 18

GATTS_CONNECT_EVT - 14
GATTS_MTU_EVT - 4

(Looking at the esp-idf source, I'd expect the GATTC_OPEN_EVT to always be fired (w/ error code), but that does not seem to be the case. Maybe there's an issue.
Calls to `bta_gattc_send_open_cback` in [bta_gattc_act.c](https://github.com/espressif/esp-idf/blob/594dcd525ba745a399849e5df0ea858c21a6836f/components/bt/bluedroid/bta/gatt/bta_gattc_act.c#L333)

wakwak-koba, 2019-01-29:
Try specifying the second argument of connect () without omitting it.

chegewara, 2019-01-30:
What i found is that sometimes peer device is disconnected before real connection is established. Usually its with `0x3E Connection Failed to be Established` reason.

When this happen its very important to not continue connection, instead we need to delete BLEClient to close connection in bt stack and unregister app. This is part of example app in arduino:
```
    bool _connected = pClient->connect(myDevice);  // if you pass BLEAdvertisedDevice instead of address, it will be recognized type of peer device address (public or private)
    if(!_connected){
      delete pClient;
      delete myDevice;
      return false;
    }
```