nkolban/esp32-snippets #768: BLE diconnects every second

i4things, 2018-12-24:
the module is derived from the example advertise a characteristic and set value every two second when connected 

the mobile from the other side is running on iPhone and just dump the data sent 

this behavior is very visible when compiled with esp32 stable build 1.0.0 if compiled with latest  eps32 from github the disconnects are a lot less - but then the latest esp32 development build have other stability issues e.g. I'm kind of in catch 22  

08:46:35: BLE: CONNECTED : START NOTIFY  
08:46:35: BLE: DISCONNECTED : START ADVERTIZING  
08:46:37: BLE: SEND[ House@12.76v# ]  
08:46:39: BLE: CONNECTED : START NOTIFY  
08:46:39: BLE: SEND[ House@12.76v# ]  
08:46:41: BLE: DISCONNECTED : START ADVERTIZING  
08:46:43: MON[ 2880 ] MIN[ 15 ]  
08:46:44: BLE: SEND[ House@12.76v# ]  
08:46:51: BLE: CONNECTED : START NOTIFY  
08:46:51: BLE: SEND[ House@12.76v# ]  
08:46:53: MON[ 2688 ] MIN[ 14 ]  
08:46:54: BLE: SEND[ House@12.76v# ]  
08:46:55: BLE: DISCONNECTED : START ADVERTIZING  
08:46:57: MEM: [ 24392 ]  
08:47:04: MON[ 2688 ] MIN[ 14 ]  
08:47:04: BLE: SEND[ House@12.76v# ]  
08:47:05: BLE: CONNECTED : START NOTIFY  
08:47:05: BLE: SEND[ House@12.76v# ]  
08:47:07: BLE: DISCONNECTED : START ADVERTIZING  
08:47:09: BLE: SEND[ House@12.76v# ]  
08:47:11: BLE: CONNECTED : START NOTIFY  
08:47:11: BLE: DISCONNECTED : START ADVERTIZING  

code for ble module in ESP32 side

#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLEServer* BLE_SERVER;
BLECharacteristic* BLE_CHARACTERISTIC;

bool BLE_CONNECTED;
bool BLE_OLD_CONNECTED;

#define BLE_TIMEOUT  2000
uint32_t BLE_LAST_EXECUTE;

#define BLE_SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define BLE_CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

#define BLE_NAME "BOAT-VITALS "

#define BLE_DISPALY_DELIMITER String("@")
#define BLE_DISPALY_ELEMENT_END  String("#")

#define BLE_DISPLAY_HOUSE String("House")
#define BLE_DISPLAY_START String("Start")
#define BLE_DISPLAY_TEMP String("Temp")
#define BLE_DISPLAY_HUMIDITY String("Humidity")
#define BLE_DISPLAY_WATER String("Bilge")
#define BLE_DISPLAY_LAT String("Lat")
#define BLE_DISPLAY_LON String("Lon")
#define BLE_DISPLAY_SAT String("Sat")
#define BLE_DISPLAY_WIFI String("WiFi")
#define BLE_DISPLAY_RADIO String("Radio")
#define BLE_DISPLAY_MOBILE String("Mobile")
#define BLE_DISPLAY_LAST String("Last")
#define BLE_DISPLAY_NEXT String("Next")

#define BLE_INDEX_COUNT 14
uint8_t BLE_INDEX;

[... 165 lines cut ...]
    }

    // disconnecting
    if (!BLE_CONNECTED && BLE_OLD_CONNECTED)
    {
      BLE_INDEX = 0;

      BLE_SERVER->startAdvertising(); // restart advertising

      LOG64_SET(F("BLE: DISCONNECTED : START ADVERTIZING"));
      LOG64_NEW_LINE;

      BLE_OLD_CONNECTED = BLE_CONNECTED;
    }
    // connecting
    if (BLE_CONNECTED && !BLE_OLD_CONNECTED)
    {
      BLE_INDEX = 0;

      LOG64_SET(F("BLE: CONNECTED : START NOTIFY"));
      LOG64_NEW_LINE;
      BLE_OLD_CONNECTED = BLE_CONNECTED;
    }
  }
}

chegewara, 2018-12-24:
Try with this
https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/examples/BLE_server/BLE_server.ino#L36-L37

I will give you some more info after xmas

i4things, 2018-12-24:
as far as I understand the suggestion is to use 

setMinPreferred(0x06);  which overrides the default value or 0x20-will give it a try ...

I'm a bit puzzled btw - why on the next line in the example you actually call the same function with

setMinPreferred(0x12);

which wipes out the 0x06 value....

chegewara, 2018-12-25:
Typo, should be `setMaxPreferred` of course.

i4things, 2018-12-25:
yep seems to help :)