nkolban/esp32-snippets #774: Arduino: BLE client can't receive response after writer with indication

noukee, 2019-01-02:
I can to connect to peripheral and its remote characteristic but when I try to write something, after to enable notify/indication, **is disconnected and I don't know the reason**.

I've tried to enable pairing mode on pheripheral and it's the same. I don't know if the issue is for  write with indication, I'm puzzled.

My code is:
```
#include "MyBLE.h"

static const char* LOG_TAG="MyBLE";

// Keyturner Pairing Service
const BLEUUID    Pairing_serviceUUID("a92ee100-5501-11e4-916c-0800200c9a66");
const BLEUUID    Pairing_GDIO_charUUID("a92ee101-5501-11e4-916c-0800200c9a66"); // GDIO characteristic. Propierties: WRITE
// Keyturner Service
const BLEUUID    Keyturner_serviceUUID("a92ee200-5501-11e4-916c-0800200c9a66");
const BLEUUID    Keyturner_GDIO_charUUID("a92ee201-5501-11e4-916c-0800200c9a66"); // GDIO characteristic. Propierties: WRITE
const BLEUUID    Keyturner_USDIO_charUUID("a92ee202-5501-11e4-916c-0800200c9a66");// USDIO characteristic. Propierties: WRITE
// Variables
static BLEUUID serviceUUID;
static BLEUUID charUUID;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAddress *pServerAddress;
static BLEScan* pBLEScan;
// nuki
static Nuki nuki;
static Expected_Message message_Nuki;
static uint16_t lock_action_to_execute;
// state attributes
static boolean is_connected;
static bool is_paired;
static bool doPair;
static bool doConnect;
static bool doAction;

static uint8_t send_buffer[200];
std::vector<uint8_t> response(200);

void ble_setup() {

  ESP_LOGD(LOG_TAG,"Starting Arduino BLE Client application...");
  //esp_log_level_set("*", ESP_LOG_DEBUG);

  doConnect = false;
  doPair = false;
  doAction = false;
  is_connected = false;
  is_paired = false;

  message_Nuki= IND_NONE;
  response.clear();

  lock_action_to_execute = NONE;

 	//BLEDevice::setMTU(200);
	BLEDevice::init("NoukeeFob");
  /*// Security
  BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT_MITM); // Implemented simplest security
  BLEDevice::setSecurityCallbacks(new MySecurity()); // Required in authentication process
  BLESecurity *pSecurity = new BLESecurity();
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND); // Bonding, a binary value indicating whether the peripheral supports bonding and wishes to bond.
  pSecurity->setCapability(ESP_IO_CAP_NONE); // IO, a numeric value from 0 to 4 used to determine the pairing method.
  //pSecurity->setInitEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);
  pSecurity->setRespEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK); */
  // Scanner
  pBLEScan = BLEDevice::getScan(); // Retrieve a Scanner and set the callback we want to use to be informed when we
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true);  // have detected a new device.  Specify that we want active scanning and start the

} // End of setup.

[... 541 lines cut ...]
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 4
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 4
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: WriteCharEvt (0x3ffec584), owner: writeValue
[V][[V]eRLOS.ips.c6] 95it g: <CliantE enmapporeStrlea()d: nkmownWriTTChliEvt evxnt eyp84)41ownDr: LEDA>vic[.c[BL15emogeChClicteEvsnic.ndl:r(7: gritClalutE): tHawrlte [lue
gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[I][BLEDevice.cpp:607] removePeerDevice(): remove: 0, GATT role client
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: RssiCmplEvt (0x3ffe8044), owner: <N/A>
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: SearchCmplEvt (0x3ffe7fe4), owner: <N/A>
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 5
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 5
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[D][myBLE.cpp:117] ble_loop(): Waiting for a response
[D][myBLE.cpp:117] ble_loop(): Waiting for a response
```

noukee, 2019-01-11:
Thanks @copercini with these changes now is working:

if(pRemoteCharacteristic->canIndicate()){
      const uint8_t indicationOn[] = {0x2,0x0};
      //const uint8_t indicationOff[] = {0x0,0x0};
      pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)indicationOn,2,true);
      //pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notifyOff,2,true);
      pRemoteCharacteristic->registerForNotify(notifyCallback, false); //false
      delay(200);
    }

closed 2019-01-11 by noukee

SBrunelle, 2019-02-06:
@noukee Great post ! It's exactly the one I was looking for. Having the same issue. Is it possible to know what is your MyBLE.h include ? Thanks in advance.