nkolban/esp32-snippets #776: Manufacturer Data  and  disconnect

wanqbabay, 2019-01-09:
Arduino 1.8.8
ESP32
the latest version arduino library

The UART demo in arduino ble library.

I just want to advertise the ble address in the  manufacture data.

 BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  BLEAdvertisementData  pAdvertisingdata = BLEAdvertisementData();
  std::string m_data="00";
  std::string  pdata=BLEDevice::getAddress().toString().c_str();
  for(int i=1;i<=pdata.length();i++){
    if((i%3) !=0) {
      m_data+=pdata[i-1];
    }
  }  
  pAdvertisingdata.setManufacturerData(m_data);
  pAdvertising->setAdvertisementData(pAdvertisingdata);

 
then nrf in the android (7.0) can not connect. 
[image]
the type i find is classical and ble.
But android (more than 7.0, like 7.12 or 8.0) and ios12 connect 

If i delete the code ,then everything is ok
[image]
then nrf in the android (7.0) can  connect. But the BLE can not pair the esp32.

Thanks

wanqbabay, 2019-01-22:
Hello! 
The question i have met may be solved.But i dont kown the reason.
First:

[image]

We can see that the type is classic and ble, and the nrf in android 7.0(android 7.12 is ok) can not connect.

Second:

[image]

I use the BT in the phone to scan the ESP32 ble "MyESP32" and pair. 

Third:

[image]

I failed. 

Fourth:
[image]

But i use the NRF in the phone san the ble. The type changed: BLE only.

Fifth:
[image]
I can get the data via the nrf and the manufacturer data  is ok.

@chegewara   could you tell me the reason ? I test 3 esp32 devices(the question ), and ok.
But other esp32 is ok. That is strange.

Thanks

wanqbabay, 2019-01-22:
But some time later; the nrf in android 7.0 can not caonnect the esp32 ble and the type is classic and ble.
That is strange.
[image]

chegewara, 2019-01-22:
Sorry i didnt respond it earlier, but i just dont have any reasonable response. I have no idea what setting is causing that esp32 is advertised as BLE only or classic + BLE.

wanqbabay, 2019-01-23:
> Sorry i didnt respond it earlier, but i just dont have any reasonable response. I have no idea what setting is causing that esp32 is advertised as BLE only or classic + BLE.

I have 5 esp32 devices, and i use the same arduino demo to test.

Android 6.0: All can can be connected,and the type is BLE only. 
Android 7.0: Only two can be connected and the type is BLE only; the rest not and the type is classic and ble.
Andriod 7.1.2:All can can be connected,and the type is BLE only.
Andriod 8.0:All can can be connected,and the type is BLE only.
IOS12::All can can be connected,

So i doubt that  the cause may be android 7.0, but some can be connected. I will do more tests.

wanqbabay, 2019-01-23:
I have tested two other android 7.0 phone; and its ok. But I use the BLE address as the BLE name to link.
std::string  pdata=BLEDevice::getAddress().toString().c_str();
  for(int i=1;i<=pdata.length();i++){
    if((i%3) !=0) {
      m_data+=pdata[i-1];
    }
  } 
 // pAdvertisingdata.setManufacturerData(m_data);
  pAdvertisingdata.setName(m_data);
  pAdvertising->setScanResponseData(pAdvertisingdata);

If i use the manufacturer data to advertise the BLE address, The problem appears: nrf (android 7.12 and 7.0) cannt connect.

chegewara, 2019-01-23:
I have idea. Could you use new function with value 0:
https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/examples/BLE_server/BLE_server.ino#L36

wanqbabay, 2019-01-24:
Thanks @chegewara 
All dont work.  0ne is ok,but the rest not and the type is classic and ble.
More tests will be done.