nkolban/esp32-snippets #784: Delete pClient crashes everything

copercini, 2019-01-17:
Hi all,

I'm trying to free pClient memory before create another one calling `BLEClient*  pClient  = BLEDevice::createClient();` , but when the destructor is called a crash occurs 

Code:
```cpp
class MyClientCallback : public BLEClientCallbacks {
    void onConnect(BLEClient* pClient) {
    }

    void onDisconnect(BLEClient* pClient) {
        ESP_LOGI(__func__, "onDisconnect");
        pClient->disconnect();
        delete pClient; //crash
        ESP_LOGI(__func__, "Free heap after disconnect %u", xPortGetFreeHeapSize());
    }
};
```

Logs:
```
I (9588) onDisconnect: onDisconnect
CORRUPT HEAP: multi_heap.c:474 detected at 0x3ffd5a9c
abort() was called at PC 0x400912f9 on core 0
0x400912f9: multi_heap_assert at C:/msys32/home/note/esp/esp-idf/components/heap/multi_heap.c:377
 (inlined by) multi_heap_free_impl at C:/msys32/home/note/esp/esp-idf/components/heap/multi_heap.c:474

Backtrace: 0x4008c288:0x3ffcb260 0x4008c4b5:0x3ffcb280 0x400912f9:0x3ffcb2a0 0x4008271a:0x3ffcb2c0 0x40082be1:0x3ffcb2e0 0x40
00bec7:0x3ffcb300 0x400902eb:0x3ffcb320 0x40102c9e:0x3ffcb340 0x40100a77:0x3ffcb360 0x401040a4:0x3ffcb380 0x401040de:0x3ffcb3
a0 0x401040fe:0x3ffcb3c0 0x400ff57e:0x3ffcb3e0 0x400d3f2a:0x3ffcb400 0x400ff8ae:0x3ffcb420 0x401052e1:0x3ffcb4e0 0x4011dc41:0
x3ffcb530 0x4011ac86:0x3ffcb570 0x4008f7ed:0x3ffcb5a0
0x4008c288: invoke_abort at C:/msys32/home/note/esp/esp-idf/components/esp32/panic.c:707

0x4008c4b5: abort at C:/msys32/home/note/esp/esp-idf/components/esp32/panic.c:707

0x400912f9: multi_heap_assert at C:/msys32/home/note/esp/esp-idf/components/heap/multi_heap.c:377
 (inlined by) multi_heap_free_impl at C:/msys32/home/note/esp/esp-idf/components/heap/multi_heap.c:474

0x4008271a: heap_caps_free at C:/msys32/home/note/esp/esp-idf/components/heap/heap_caps.c:231

0x40082be1: _free_r at C:/msys32/home/note/esp/esp-idf/components/newlib/syscalls.c:42

0x400902eb: vQueueDelete at C:/msys32/home/note/esp/esp-idf/components/freertos/queue.c:2037

0x40102c9e: FreeRTOS::Semaphore::~Semaphore() at c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/char_
traits.h:243

0x40100a77: BLERemoteCharacteristic::~BLERemoteCharacteristic() at C:/msys32/home/Note/esp/DFUupdates/components/cpp_utils/BL
ERemoteCharacteristic.cpp:309

0x401040a4: BLERemoteService::removeCharacteristics() at C:/msys32/home/Note/esp/DFUupdates/components/cpp_utils/BLERemoteSer
vice.cpp:235

0x401040de: BLERemoteService::~BLERemoteService() at C:/msys32/home/Note/esp/DFUupdates/components/cpp_utils/BLERemoteService
.cpp:235

0x401040fe: BLERemoteService::~BLERemoteService() at C:/msys32/home/Note/esp/DFUupdates/components/cpp_utils/BLERemoteService
.cpp:235

0x400ff57e: BLEClient::~BLEClient() at C:/msys32/home/Note/esp/DFUupdates/components/cpp_utils/BLEClient.cpp:449

0x400d3f2a: MyClientCallback::onDisconnect(BLEClient*) at C:/msys32/home/Note/esp/DFUupdates/main/hello_world_main.cpp:122

0x400ff8ae: BLEClient::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:/msys32/ho
me/Note/esp/DFUupdates/components/cpp_utils/BLEClient.cpp:449

0x401052e1: BLEDevice::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:/msys32/ho
me/Note/esp/DFUupdates/components/cpp_utils/BLEDevice.cpp:167

0x4011dc41: btc_gattc_cb_to_app at C:/msys32/home/note/esp/esp-idf/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.c:3
1
 (inlined by) btc_gattc_cb_handler at C:/msys32/home/note/esp/esp-idf/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.
c:922

0x4011ac86: btc_task at C:/msys32/home/note/esp/esp-idf/components/bt/bluedroid/btc/core/btc_task.c:110

0x4008f7ed: vPortTaskWrapper at C:/msys32/home/note/esp/esp-idf/components/freertos/port.c:403
```

chegewara, 2019-01-17:
Hi @copercini 
its because you are doing it from inside onDsiconnect() callback. As you can see here
https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/src/BLEClient.cpp#L192-L195
after onDisconnect has been called pClient needs to be also deleted from connected clients list here
https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/src/BLEDevice.cpp#L610

chegewara, 2019-01-17:
Ok, i remember now. I changed a bit BLEClient.cpp but had no chance to create PR yet. Here is code you can try:
```
/*
 * BLEDevice.cpp
 *
 *  Created on: Mar 22, 2017
 *      Author: kolban
 */
#include "sdkconfig.h"
#if defined(CONFIG_BT_ENABLED)
#include <esp_bt.h>
#include <esp_bt_main.h>
#include <esp_gap_ble_api.h>
#include <esp_gattc_api.h>
#include "BLEClient.h"
#include "BLEUtils.h"
#include "BLEService.h"
#include "GeneralUtils.h"
#include <string>
#include <sstream>
#include <unordered_set>
#include "BLEDevice.h"
#if defined(ARDUINO_ARCH_ESP32) && defined(CONFIG_ARDUHAL_ESP_LOG)
#include "esp32-hal-log.h"
#define LOG_TAG ""
#else
#include "esp_log.h"
static const char* LOG_TAG = "BLEClient";
#endif

/*
 * Design
 * ------
 * When we perform a searchService() requests, we are asking the BLE server to return each of the services
 * that it exposes.  For each service, we received an ESP_GATTC_SEARCH_RES_EVT event which contains details
 * of the exposed service including its UUID.
 *
 * The objects we will invent for a BLEClient will be as follows:
 * * BLERemoteService - A model of a remote service.
 * * BLERemoteCharacteristic - A model of a remote characteristic
 * * BLERemoteDescriptor - A model of a remote descriptor.
 *
 * Since there is a hierarchical relationship here, we will have the idea that from a BLERemoteService will own
 * zero or more remote characteristics and a BLERemoteCharacteristic will own zero or more remote BLEDescriptors.
 *
 * We will assume that a BLERemoteService contains a map that maps BLEUUIDs to the set of owned characteristics
 * and that a BLECharacteristic contains a map that maps BLEUUIDs to the set of owned descriptors.
 *
 *
 */

BLEClient::BLEClient() {
	m_pClientCallbacks = nullptr;
	m_conn_id          = ESP_GATT_IF_NONE;
	m_gattc_if         = ESP_GATT_IF_NONE;
	m_haveServices     = false;
	m_isConnected      = false;  // Initially, we are flagged as not connected.

	m_appId = BLEDevice::m_appId++;
	m_appId = m_appId%100;
	BLEDevice::addPeerDevice(this, true, m_appId);
	m_semaphoreRegEvt.take("connect");

	esp_err_t errRc = ::esp_ble_gattc_app_register(m_appId);
	if (errRc != ESP_OK) {
		ESP_LOGE(LOG_TAG, "esp_ble_gattc_app_register: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
		return;
	}

	m_semaphoreRegEvt.wait("connect");
[... 437 lines cut ...]
	getService(serviceUUID)->getCharacteristic(characteristicUUID)->writeValue(value);
	ESP_LOGD(LOG_TAG, "<< setValue");
} // setValue

uint16_t BLEClient::getMTU() {
	return m_mtu;
}

/**
 * @brief Return a string representation of this client.
 * @return A string representation of this client.
 */
std::string BLEClient::toString() {
	std::ostringstream ss;
	ss << "peer address: " << m_peerAddress.toString();
	ss << "\nServices:\n";
	for (auto &myPair : m_servicesMap) {
		ss << myPair.second->toString() << "\n";
	  // myPair.second is the value
	}
	return ss.str();
} // toString

#endif // CONFIG_BT_ENABLED
```

copercini, 2019-01-17:
@chegewara Thanks for your quick reply and code, this crash seems related with this line: https://github.com/nkolban/esp32-snippets/blob/c8dea9ec9bcacdbada2b655be0621cb618eca3e8/cpp_utils/BLERemoteService.cpp#L296 just commenting this out, no more crashes occur, only a small memory leak

chegewara, 2019-01-17:
Yes, it is also possible, because my internal changes looks like this:
```
void BLERemoteService::removeCharacteristics() {
	for (auto &myPair : m_characteristicMap) {
	   delete myPair.second;
	   //m_characteristicMap.erase(myPair.first);  // Should be no need to delete as it will be deleted by the clear
	}
	m_characteristicMap.clear();   // Clear the map
} // removeCharacteristics
```

copercini, 2019-01-17:
it's my fault actually *facepam*

I was using a global `static BLERemoteCharacteristic* myCharacteristic;` then it was crashing on some internal free of destructor

closed 2019-01-17 by copercini

copercini, 2019-01-17:
Additionally, using this PR: https://github.com/nkolban/esp32-snippets/pull/783 there is no problem of have global characteristics