nkolban/esp32-snippets #785: ESP32 does not boot with external power after BLEDevice::init

naaaas, 2019-01-17:
HI.
 i'm newbie with esp32. I designed a simple BLE scanner wichs works fine when connected to usb. The problem is that if i disconnect board from usb and connect it to power (5v) device does not boot. Just red led is On and even if i push boot or en buttons...it never boot again. Any ideas?

Thank you

chegewara, 2019-01-17:
Find better power source? This one may be not suficient.

naaaas, 2019-01-17:
I tried a lot of things. Even board samomes like "wifi scanner" and happends the same. But what do you mean with...better power source?

chegewara, 2019-01-17:
https://github.com/nkolban/esp32-snippets/issues/168

try to search `esp32 brownout`

mitchjs, 2019-01-17:
> tried "WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);" with no success. I suppose it is a HW issue? So this is increasing complexity of using this board.

Try this:

 

//////////////////////////////////////////////////////////

    // DISABLE BROWNOUT DETECTOR (BT Radio power up would trigger under USB power only)

    //////////////////////////////////////////////////////////

    REG_CLR_BIT(RTC_CNTL_BROWN_OUT_REG, RTC_CNTL_BROWN_OUT_ENA);

 

    //////////////////////////////////////////////////////////

            // SET UP BLE

            //////////////////////////////////////////////////////////

            // create BLE service

            BLEDevice::init(MODEL);

            

 

    //////////////////////////////////////////////////////////

    // ENABLE BROWNOUT DETECTOR

    //////////////////////////////////////////////////////////

    REG_SET_BIT(RTC_CNTL_BROWN_OUT_REG, RTC_CNTL_BROWN_OUT_ENA);

 

 

I have a Switching supply that supplies 300ma, and when the radio intiailizes

It doesn’t draw more that 300ma, but it spikes, and maybe the “smart” switching supply

LM2841Y, detects it and stops its output

 

 

mitch

naaaas, 2019-01-17:
thank you but...still wrong.  this time red light is very very low when connecting to VIN. im quite lost

closed 2019-01-18 by naaaas

PyTechMike, 2020-06-22:
Hi! Have the same issue. I use Arduino IDE, if it matters. Could you please help me a bit, don't really understood the solution. Thanks!

chegewara, 2020-06-22:
https://www.google.com/search?q=esp32+brownout&oq=esp32+brownout&aqs=chrome..69i57.14248j0j1&sourceid=chrome&ie=UTF-8