nkolban/esp32-snippets #787: BLE client stuck on m_semaphoreWriteCharEvt.wait("writeValue")

noukee, 2019-01-21:
The BLE client, developed with Platformio IDE, gets to connect to server (peripheral) and the characteristic with enable write and indication.

The first time I can write with response and receive the correspondent indication however when I try again to write with response is stuck in the semaphore m_semaphoreWriteCharEvt.wait("writeValue") 

pRemoteCharacteristic->writeValue(send_buffer,length,true);

Does anybody know what happens?

noukee, 2019-01-25:
@chegewara, can you check the logs? To find out what I'm doing wrong. Thank you!

chegewara, 2019-01-25:
Hi @noukee 
I cant say too much from logs, but i think you are trying to write remote characteristic from within notify callback which will cause code stuck because of semaphore lock.

noukee, 2019-02-04:
Hi @chegewara , 

Then, to check if I understand you well, I can't call processMessage within notifycall because there I'm writing the remote characteristic, is it right?

```
void MyBLE::notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {

    ESP_LOGD(LOG_TAG,"******************** Notify callback for characteristic ");
    ESP_LOGD(LOG_TAG,"%s",pBLERemoteCharacteristic->getUUID().toString().c_str());
    ESP_LOGD(LOG_TAG," of data length %d\n", length);

    processMessage(pData,length,pBLERemoteCharacteristic);

} // end notifyCallback
```

Thank you!

sami1369, 2020-06-24:
I have same problem 
What is the solution?

chegewara, 2020-06-24:
Try to set flag/variable like `canProccessMsg` and call this function from loop if flag is true.