nkolban/esp32-snippets #788: Ble scan result update  rate only every one second with the fastest

Sketh, 2019-01-21:
I want to use esp32 as client to scan mybeacon device RSSI for proximity detection that broadcast signal with 30hz update rate, but esp32 scan rate  only with 1hz update rate that resulting  RSSI  fluctuation is large, i try to change function start parameter in ""BLEScanResults results = scan->start(0.1);", but it can not working that only support second unit parameter, so how to solve it?

chegewara, 2019-01-21:
Hi, 
you cant scan shorter than 1 second, its esp-idf limitation. Well, its not its true only if you use procedure start scan => wait till scan ends, but you could use procedure start scan => stop scan. I think its not what you want/need. Maybe instead using scan this way `BLEScanResults results = scan->start(0.1);` try with scan result callback function.

Sketh, 2019-01-22:
Sorry my english is not goof  i don't  realy understand  your mean, you mean  ble scan update rate once per second is faster in esp32 did that right?

chegewara, 2019-01-22:
Yes, you cant scan shorter than 1 second.