nkolban/esp32-snippets #790: How to stop the BLE server

rado99, 2019-01-22:
Hello,
I have the following use-case: The user can choose in which mode to use the ESP32 - in BLE, Wi-Fi client or Wi-Fi AP. I went to this decision after several tries to run Wi-Fi and BLE simultaneously, but without success - BLE was very laggy and I often got wdt resets. So, I know how to stop the Wi-Fi, but what about the Bluetooth? Does with only stopping the services and the advertising I can run without problem the Wi-FI mode?
Thank you in advance.

chegewara, 2019-01-22:
Try this
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEDevice.cpp#L620

rado99, 2019-01-22:
Hmm, is this file different from the included one in esp32 arduino package version 1.0.0? Because I have the following error:
"deinit is not a member of BLEDevice"