nkolban/esp32-snippets #795: failed to connect two ESP32 with BLE Communication for arduino ide

lukascjh, 2019-01-25:
with Arduino BLE Examples  BLE_server and BLE_client dosen't work, i Need help, thx so much

lukascjh, 2019-01-25:
@chegewara the examples in Arduino BLE_server and BLE_client  are not for two esp32 Communication for each other , but for the connection between esp and mobile phone for example

chegewara, 2019-01-26:
Yes, they require a little changes to connect each other.

mpbejo, 2019-02-04:
Hi chegewara,
I was trying to use the BLE Examples BLE_server and BLE_client, but the client has an error and restart.
I am using Esp32 1.0.1 Board library and ESP32 BLE Arduino 1.0.1 the error:
- Found our characteristic
[D][BLERemoteCharacteristic.cpp:404] readValue(): >> readValue(): uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8, handle: 42 0x2a
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: ReadCharEvt (0x3ffe3b90), owner: <N/A> for readValue
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: ReadCharEvt (0x3ffe3b90), owner: readValue
[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c", line 268, function: heap_caps_free
abort() was called at PC 0x4014be87 on core 0

Decoding stack results
0x4008d908: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008db39: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x4014be87: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x4008454f: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 268
0x4008a289: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/syscalls.c line 42
0x400d561a: BLERemoteCharacteristic::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:\Users\Nicola\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLERemoteCharacteristic.cpp line 192
0x400d64d9: BLERemoteService::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:\Users\Nicola\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLERemoteService.cpp line 120
0x400d42f9: BLEClient::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:\Users\Nicola\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp line 314
0x400d5065: BLEDevice::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:\Users\Nicola\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEDevice.cpp line 169
0x400e5095: btc_gattc_cb_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.c line 31
0x400e1082: btc_task at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/core/btc_task.c line 110
0x4009257d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

mpbejo, 2019-02-05:
Rolling back to Esp32 1.0.0 Board library the samples works.
Am I the only to have these problems?!?

mpbejo, 2019-02-06:
Downloading esp32-snippets/cpp_utils
and replacing the Esp32 BLE 1.0.1  library files I solved the problem

closed 2019-02-06 by chegewara