nkolban/esp32-snippets #797: BLEAdvertising stop doesn't stop advertising

tpitman, 2019-01-29:
I have a simple application that I want to be able to start and stop advertising so that other devices are prevented from connecting when I don't want them too. This needs to be turned on and off at runtime.

Here is my startup code:

```BLEDevice::init("MyName");
  
  pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  pServer->setCallbacks(new BTServerCallbacks());

  pCharacteristicCommand = pService->createCharacteristic(
                             COMMAND_UUID,
                             BLECharacteristic::PROPERTY_READ |
                             BLECharacteristic::PROPERTY_WRITE |
                             BLECharacteristic::PROPERTY_NOTIFY
                           );

  pCharacteristicCommand->setCallbacks(new BTCallbacks());
  pCharacteristicCommand->setValue("");
  pCharacteristicCommand->addDescriptor(new BLE2902());

  pService->start();
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->start();
```

This works great and my service advertises and I am able to connect with it and process things.

Then when I want my device to become unavailable I do this:

```
        BLEAdvertising *pAdvertising = pServer->getAdvertising();
        pAdvertising->stop();
```

It calls it and doesn't give any errors or exceptions, but my device is still visible to my iPhone and I can still connect to it.

How do I disable advertising and connecting during runtime?

chegewara, 2019-01-29:
Are you sure you run new scan, or is esp32 visible from old BLE scan?
I dont know how iPhone works, but for example on android nRF connect scan lasts few seconds and then stops. All discovered devices are still visible in nRF connect list and can be connected even if you stop advertising on them, unless peer device will disappear or not allow to be connected.

Azriel-Samson, 2019-02-12:
I think this is because start() is called in ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT in cpp_utils/BLEAdvertising.cpp Line 496.

onesbasket, 2019-02-12:
I have same issue.
And as Azrie-Samson mentioned, I tried with successfully.

artin-phares, 2019-06-04:
any known workaround for this?

Gabriel-Gardin, 2019-08-11:
Facing the same problem, I start BLE advertising and after 300 ms I try to stop with 
pAdvertising->stop(). Doesnt work.

chegewara, 2019-08-11:
Just delete this line:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertising.cpp#L496

taichunmin, 2019-09-06:
Fixed already with https://github.com/espressif/arduino-esp32/pull/3034

GKCHDEV, 2019-11-05:
> Just delete this line:
> https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertising.cpp#L496

I commected Still it is discovering to my PC

i added stop in void onDisconnect(BLEServer* pServer) function
 BLEAdvertising *pAdvertising = pServer->getAdvertising();
    //pAdvertising->start();
    pAdvertising->stop();

and commented start() function but still advertising to my PC so what i have to do please help me on this to stop advertising.
case ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT: {
			
			//start();
			break;
		}

chegewara, 2019-11-05:
If you comment out this line (start()) and use advertising->stop(), then esp32 should stop advertising.

GKCHDEV, 2019-11-05:
I added stop() function like below ,i am putting my client in sleep mode so stopping the advertising now and LOG(ESP_LOGI(LOG_TAG, "STOP advertising"); )is printing continuously,how can i reconnect automatically when client is wakes from sleep.

case ESP_GAP_BLE_ADV_STOP_COMPLETE_EVT: {
			ESP_LOGI(LOG_TAG, "STOP advertising");
			//start();
            stop();
			break;
		}

marcidelux, 2021-04-27:
I got the same problem, even with the fixed Arduino version when.
`start();` was commented out in `void BLEAdvertising::handleGAPEvent{}`.

I found an intresting syntax problem in `BLEAadvertising.cpp` at line 248:
`esp_err_t errRc = ::esp_ble_gap_stop_advertising();` There is two coma for a scope of nothing :D 
So I just removed the comas: `esp_err_t errRc = esp_ble_gap_stop_advertising();`

And magic happens, it works !

N4poLion, 2021-04-28:
In older versions of Arduino framework for ESP32 at BLEServer.cpp check the `BLEServer::handleGATTServerEvent` function at : case of ESP_GATTS_DISCONNECT_EVT.  there is a line: `startAdvertising(); //- do this with some delay from the loop()` what causes the restart of BLE after calling the stop succesfully.

The recommended solution is to update your Arduino Framework. In the current 1.0.6 version it is fixed, or you can just comment this line out.