nkolban/esp32-snippets #798: Architecture Issue: Respecting different MTU for notifications of large data

nsamala, 2019-01-30:
I've re-implemented the Espressif vendor profile implementation of UART over BLE (to be able to send larger amounts of data via notifications) to be able to send MsgPack data over BLE.

If I have two different clients (say one has an MTU of 23 and the other has an MTU of 128), I'd need to chunk up my data per the MTU for each one and send a notify. In the current implementation, I'm limited to whichever device has the lowest MTU because I can't chunk up the data per device in the notify function.

Any thoughts on how to implement this other than re-implementing the `notify` function myself?

chegewara, 2019-01-30:
If you have 2 clients with 2 different MTU then each client is receiving notifications with its own size. If you send notification value longer than client's MTU then it will be truncated.

What you can do is to find a way to find client's MTU and split message according to it, but you will have to count for each client how many bytes have you sent with notification etc, or send notifications with smallest MTU length. You can also simplify it even more and send notifications with 20 bytes length, which is longest length for MTU 23.

Probably i dont understand your problem and maybe you could try to explain it to me.

chegewara, 2019-01-30:
This is how i am chunking characteristic value for different MTU size:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLECharacteristic.cpp#L363

nsamala, 2019-01-30:
So that works for reading (I have similar code for reading with different MTU size). But my question is regarding *sending* data with clients that have different MTU

chegewara, 2019-01-30:
Yes, you can use that part of code in your app to get client MTU and then split message according to it:
`getService()->getServer()->getPeerMTU(conn_id) `
Only value that you need is conn_id, which can be retrieved from here:
`getService()->getServer()->getConnId()`.

So, after connecting all pieces together you will end up with something like this:
```
size_t packetSize = txCharacteristic->getService()->getServer()->getPeerMTU(txCharacteristic->getService()->getServer()->getConnId());
```

This library and code i am proposing is for both, esp-idf and arduino-ide.

chegewara, 2019-01-30:
Its not pretty code, but in theory should works.

nsamala, 2019-01-30:
Won't that only work for a single client though? In the example above, if my `packetSize` was determined to be 128 from `getConnId`, but there was a previous client that was connected with MTU 23, the data would get truncated wouldn't it?

chegewara, 2019-01-30:
You are right, this code is only partially correct. You have to store in your code connID for every connected client and use in this code:
```
for() { // every connected client
size_t packetSize = txCharacteristic->getService()->getServer()->getPeerMTU(clientConnID);
// calculate and send notify as long as you didnt sent all data
}
```

nsamala, 2019-01-30:
So doesn't that mean the Arduino implementation in BLECharacteristic is limited? You basically have to re-implement the `notify` function to do that (instead of the setValue and notify pattern)

chegewara, 2019-01-30:
Yes, library is designed to truncate value if its longer than MTU rather than sending few notifications to send all data.