nkolban/esp32-snippets #805: Wifi Reconnect - FreeRTOS: Semaphore NOT taken

rweller59, 2019-02-05:
Started to add some error checking to the wifi connection. Wrote simple code to test reconnect. I connect to wifi then cycle the router. The even handler catches staDisconnected but then I get this error when trying to reconnect.

   'FreeRTOS: Semaphore NOT taken:  name: ConnectFinished (0x3ffafdc4), owner: connectAP'. 

When the router is active again, the wifi reconnects - not sure how since the event handler log messages are no longer displayed so I'm assuming the handler is not what recreates the connection.
The error message continues to get printed every few seconds.
```// wifi event handler
const int CONNECTED_BIT = BIT0;
class wifiEventHandler: public WiFiEventHandler {

	esp_err_t staStart() { ESP_LOGI(TAG, "staStart"); return ESP_OK; }

	esp_err_t staGotIp(system_event_sta_got_ip_t info) {
		ESP_LOGI(TAG, "staGotIp");
		wifi_connected = true;
		xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
		return ESP_OK;
	}
	esp_err_t staDisconnected(system_event_sta_disconnected_t info) {
		ESP_LOGI(TAG, "staDisconnected");
		wifi_connected = false;
	        xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
	        wifiService::wifiConnect();
		return ESP_OK;
	  }
	esp_err_t staConnected(system_event_sta_connected_t info) { ESP_LOGI(TAG, "staConnected"); return ESP_OK; }
	esp_err_t staStop() { ESP_LOGI(TAG, "staStop"); return ESP_OK; }
	esp_err_t wifiReady() { ESP_LOGI(TAG, "wifiReady"); return ESP_OK; }

};

class task_init_wifi: public Task {
  void run(void *data) {
	  while(true) {
			delay(1000);
	  }
}};

wifiEventHandler *wifiEventHandler_m = new wifiEventHandler();
task_init_wifi* initWifiTask = new task_init_wifi();

// **************** beginning of namespace
namespace wifiService {

void begin(bool connect) {
	temp();
        wifi_on = connect;
	if(setup.device.network && connect) {
                ESP_LOGI(TAG, "Initializing Wifi Service: %d", ap_connected);
	        wifi_event_group = xEventGroupCreate();
                wifi_m.setWifiEventHandler(wifiEventHandler_m);
		wifiConnect();
                initWifiTask->setName("initWifiTask");
		initWifiTask->setStackSize(8096);
		initWifiTask->start();
		return;
  }
  if(connect) ESP_LOGI(TAG, "Network service turned off");
}

void wifiConnect() {
	uint8_t ap_index = 0;
	std::string ssid = setup.network.wifi[ap_index].ssid;
	std::string password = setup.network.wifi[ap_index].pwd;
	ESP_LOGI(TAG, "trying ap: %d, %s - %s",ap_index, ssid.c_str(), password.c_str());
	uint8_t x = wifi_m.connectAP(ssid, password, true);
	ESP_LOGI(TAG, "exiting wifiConnect, %d", x); 
}

} // ******************* end of namespace
```

HanVertegaal, 2019-02-05:
The error message is generated through line 206 in WiFi.cpp.
Sometimes the connect call will not generate an event, and the semaphore will never be released. Line 206 retries the connect if there is no connection within 5 seconds. You could change the timeout value, or add a maximum number of retries.

rweller59, 2019-02-05:
Sometimes the connect call will not generate an event? 
I've only been working on this for a few hours but, for me, it happens every time so far. I've also done some testing without cpp_utils using straight c and espidf - I do not see the issue there, i.e., disconnect and connect seem to always trigger related events.

rweller59, 2019-02-05:
I played a bit with the code you mentioned...
 - that definitely controls the timing of the error message but doesn't impact the process
 - even after a connection is re-established, I continue to get the error
a bit more testing... starting with the router powered down
 - I don't get the error when calling connectAP the first time
 - after getting staDisconnected and calling connectAP again, I start getting the error

Is there something that should be re-initialized before repeating the connectAP call again?

HanVertegaal, 2019-02-05:
That is interesting. It isn't clear from the documentation that a connect call cannot be made from within the event handler. On the contrary, it suggests the opposite: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/wifi.html#wi-fi-reconnect.

rweller59, 2019-02-05:
I'm guessing it has something to do with cpp_utils, or my use of them. When I tested with c code, I was calling esp_wifi_connect() in the event handler without issues.

rweller59, 2019-02-06:
Wondering if this have something to do with the scope of the wifi object. I declared it as static but that did not make any difference.
 
Once the Semaphore error starts logging when I call wifiConnect from the event handler, events are no longer triggered although I can cycle the router and it will reconnect.

chegewara, 2019-02-06:
It has something to do with semaphores.

rweller59, 2019-02-09:
Any ideas on what I can look at to resolve this? I've not used semaphores before so a bit lost.

I can work around the issue but always nice to understand without workarounds.