nkolban/esp32-snippets #806: BLE Server Security Advice

mitchjs, 2019-02-07:
I have a BLE device (server) with 1 service that has 5 characteristics...
all is working perfectly thanks to neil work and most especially chegewara continued work
I'm doing ESP32 IDF (no need to mess with Arduino overhead)

I want to add very basic security, the device has no keyboard or display
so I want the client to simply provide a passkey then access to the service is allowed
or something like that

do we need to be paired for security to work?
or can a client connect, and respond with a passkey?
(actually would like the client sw to respond, and show a dialog asking user for pin (android app))

thanks!
mitch

mitchjs, 2019-02-08:
im making some progress on my own (yea!!!)

using:
```
// set up security
#ifdef _USE_SECURITY_
BLESecurity *pSecurity = new BLESecurity();
pSecurity->setStaticPIN(123456);
#endif
```

when I added the setAccessPermissions to the 0x2902 descriptor (notification)
```
// add notifcation
BLE2902             *MyLEDStatusNotificationDescriptor = new BLE2902();
#ifdef _USE_SECURITY_
MyLEDStatusNotificationDescriptor->setAccessPermissions(ESP_GATT_PERM_READ_ENCRYPTED | ESP_GATT_PERM_WRITE_ENCRYPTED);
#endif
MyLEDStatusCharacteristic->addDescriptor(MyLEDStatusNotificationDescriptor);
```

I tell nrfConnect to subscribe
I get from the ESP logging:
E (114481) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION

and that's after I put in the proper passkey (using the static key)
the ones with out this line... work fine of course
if I do a read of the characteristic, I get the value...

so I do a read:
E (170811) BT_GATT: GATT_INSUF_AUTHENTICATION

I (170891) BLEDevice: ESP_GAP_BLE_SEC_REQ_EVT
W (170891) BT_SMP: Non bonding: No keys will be exchanged
I (170891) BLEDevice: ESP_GAP_BLE_PASSKEY_NOTIF_EVT
I (170891) BLEDevice: passKey = 123456
I (177561) BLEDevice: ESP_GAP_BLE_AUTH_CMPL_EVT

then try to subscribe to the notifications:
E (219551) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION
E (219621) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION

thanks
mitch

mitchjs, 2019-02-09:
I have more information, I develop in PlatformIO, and they only use "released" versions of ESP-IDF SDK, so its at 3.1.2

not sure if that makes a difference.... one thing I did notice...
in Bluedroid in 3.1.2 ESP_GATTS_SEND_SERVICE_CHANGE_EVT is not defined
so its use in BLEUtils.cpp I had to add and #Ifdef ESP_GATTS_SEND_SERVICE_CHANGE_EVT

here is a super simple test code

https://gist.github.com/mitchjs/eabd9ecd63bcca0a8008436b1df5e6cb

mitchjs, 2019-02-09:
oops mis-clicked on close :)

chegewara, 2019-02-09:
Hi,
this may be the issue. I dont know about PlatformIO, how its setup, sorry.
Because security is very popular topic in last few days i will try to look at it closer.

mitchjs, 2019-02-09:
the security on the characteristic works perfectly 
https://gist.github.com/mitchjs/eabd9ecd63bcca0a8008436b1df5e6cb#file-main-cpp-L63

its the one on the descriptor 0x2902 (notifications)
https://gist.github.com/mitchjs/eabd9ecd63bcca0a8008436b1df5e6cb#file-main-cpp-L68

that seems to be the bugger! 

also... I can read the descriptor itself in nrfconnect (of course after putting in passkey)
it just seems the WRITE isn't working... as pushing the subscribe button in nrfconnect causes
the "BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION"

chegewara, 2019-02-09:
Hi @mitchjs 
i have no problem with connecting esp32 with static PIN and `ESP_GATT_PERM_READ_ENCRYPTED | ESP_GATT_PERM_WRITE_ENCRYPTED`. I am working now on exactly this example and my android smartphone is connecting, pairing and reading characteristic. Write also works.
Well, at the moment i have slightly changed example code, but im sure it shouldnt change behavior:
```
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class MySecurity : public BLESecurityCallbacks {

	uint32_t onPassKeyRequest(){
    ESP_LOGI(LOG_TAG, "PassKeyRequest");
		return 123456;
	}
	void onPassKeyNotify(uint32_t pass_key){
    ESP_LOGI(LOG_TAG, "The passkey Notify number:%d", pass_key);
	}
	bool onConfirmPIN(uint32_t pass_key){
    ESP_LOGI(LOG_TAG, "The passkey YES/NO number:%d", pass_key);
		return true;
	}
	bool onSecurityRequest(){
    ESP_LOGI(LOG_TAG, "SecurityRequest");
		return true;
	}

	void onAuthenticationComplete(esp_ble_auth_cmpl_t cmpl){
		ESP_LOGI(LOG_TAG, "Starting BLE work!");
    esp_ble_gap_update_whitelist(true, cmpl.bd_addr);
	}
};

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("ESP32");
  BLEDevice::setSecurityCallbacks(new MySecurity());
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID, BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_WRITE);
  pCharacteristic->setAccessPermissions(ESP_GATT_PERM_READ_ENCRYPTED | ESP_GATT_PERM_WRITE_ENCRYPTED);
  pCharacteristic->setValue("Hello World says Neil");
  pService->start();
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  // pAdvertising->setScanFilter(false, true);
  pAdvertising->setScanResponse(true);
  pAdvertising->start();

  BLESecurity *pSecurity = new BLESecurity();
  pSecurity->setStaticPIN(123456); 
  
  //set static passkey
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

mitchjs, 2019-02-09:
and I too have NO problem with security on the characteristic

**its the NOTIFICATION CONTROL (0x2902)**

```
BLE2902* p2902Descriptor = new BLE2902();
//p2902Descriptor->setNotifications(true);
p2902Descriptor->setAccessPermissions(ESP_GATT_PERM_READ_ENCRYPTED | ESP_GATT_PERM_WRITE_ENCRYPTED);
pCharacteristic->addDescriptor(p2902Descriptor);
```

try that in your code, and in the test software... try to subscribe to the notifications

chegewara, 2019-02-09:
Still works with staticPIN example and i can register for notify.
When i do it before i am paired then i am asked to input PIN and all works.

mitchjs, 2019-02-09:
Hmm do you know what version of esp-idf you testing on?
I’m stuck with 3.1.2 release

chegewara, 2019-02-09:
I am on master esp-idf, but i am testing with arduino-esp32 v1.0.1-release.

mitchjs, 2019-02-09:
Ok I bet the master idf is the key
Anyway you can try release 3.1?
I tried switching to the master and got stuck
I also tried just updating the bt folder and enough changed and moved that it didn’t compile 

Thanks

mitchjs, 2019-02-11:
I spent all day to get the master into PlatformIO
what a pita

and still error when I try to subscribe in nrf
␛[0;31mE (92306) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION␛[0m

even tested it with LightBlue
I press subscribe... it pops up asking for the passkey
I put it in...
␛[0;31mE (251420) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION␛[0m
it does like the passkey... cause if I do a read of the characteristic, it gives me the "hello world"
so... the passkey was accepted

I so frustrated... as I so close to a complete program...
I can put security on the notifications enable/disable :(

mitchjs, 2019-02-11:
ok, this is a bigger bug and maybe in the IDF

I did a simple write to the characteristic, and put in the passkey
and get the same error
tracing down into the bluedroid

its this 
`if ((perm & GATT_WRITE_AUTH_REQUIRED ) && !(sec_flag & GATT_SEC_FLAG_LKEY_UNAUTHED)) {
                    status = GATT_INSUF_AUTHENTICATION;
                    GATT_TRACE_ERROR( "!!!gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION");

`
this is suspect
(sec_flag & GATT_SEC_FLAG_LKEY_UNAUTHED)

its always 0, even after I put in a valid passkey!!

ok, for a read!

`if ((perm & GATT_READ_AUTH_REQUIRED ) && !(sec_flag & GATT_SEC_FLAG_LKEY_UNAUTHED) &&
            !(sec_flag & BTM_SEC_FLAG_ENCRYPTED)) {
        GATT_TRACE_ERROR( "GATT_INSUF_AUTHENTICATION\n");
        return GATT_INSUF_AUTHENTICATION;
    }`

␛[0;31mE (61240) BT_GATT: value of (sec_flag & GATT_SEC_FLAG_LKEY_UNAUTHED) 0␛[0m
␛[0;31mE (61240) BT_GATT: value of (sec_flag & BTM_SEC_FLAG_ENCRYPTED) 4␛[0m

so... in this case... we are encrypted... 
but the WRITE only checks GATT_SEC_FLAG_LKEY_UNAUTHED

is that right... or do we have a bit of missing code in the IDF?

mitchjs, 2019-02-11:
ok, looking at the code... I bet its missing
&& !(sec_flag & BTM_SEC_FLAG_ENCRYPTED)

here is the code (and I added that to the 1st IF)
and of course it works now...
but am I right? don't know enough about BLE security

```
/* require authentication, but not been authenticated */
else if ((perm & GATT_WRITE_AUTH_REQUIRED ) && !(sec_flag & GATT_SEC_FLAG_LKEY_UNAUTHED) && !(sec_flag & BTM_SEC_FLAG_ENCRYPTED)) {
    status = GATT_INSUF_AUTHENTICATION;
    GATT_TRACE_ERROR( "gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION");
} else if ((perm & GATT_WRITE_MITM_REQUIRED ) && !(sec_flag & GATT_SEC_FLAG_LKEY_AUTHED)) {
    status = GATT_INSUF_AUTHENTICATION;
    GATT_TRACE_ERROR( "gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION: MITM required");
} else if ((perm & GATT_WRITE_ENCRYPTED_PERM ) && !(sec_flag & GATT_SEC_FLAG_ENCRYPTED)) {
    status = GATT_INSUF_ENCRYPTION;
    GATT_TRACE_ERROR( "gatts_write_attr_perm_check - GATT_INSUF_ENCRYPTION");
} else if ((perm & GATT_WRITE_ENCRYPTED_PERM ) && (sec_flag & GATT_SEC_FLAG_ENCRYPTED) && (key_size < min_key_size)) {
    status = GATT_INSUF_KEY_SIZE;
    GATT_TRACE_ERROR( "gatts_write_attr_perm_check - GATT_INSUF_KEY_SIZE");
}
```

mitchjs, 2019-02-11:
ok, looking at bluedroid's source on android... 
course I knew it when I touched the IF
that shouldn't be there

so.. just need to be AUTHENTICATED to write
so I added this to my main code
pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND);

```
BLESecurity *pSecurity = new BLESecurity(); 
pSecurity->setStaticPIN(123456); 
pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND);
```

so.. maybe one needs to be bonded vs whats in setStaticPIN()
ESP_LE_AUTH_REQ_SC_ONLY

BINGO!

no changes to IDF (bluedroid)

and now WRITES work....

chegewara, 2019-02-11:
Hi,
yes, it is possible that some peer devices can require bonding. Because setup in staticPIN is picked by me as default does not mean it is only or even the correct one, it worked for me and passed test, thats all. For the same reason i left this description in code:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLESecurity.cpp#L65

Im glad you found the reason why its not working for you. Thanks.

mitchjs, 2019-02-11:
ah... I saw that comment... late... but didn't know what you meant exactly,,,

so.. strange, I would have thought it would behave the same for you
and prevent writes... I mean it has to... or it should...

ESP_LE_AUTH_REQ_SC_ONLY, would not ask for a BOND assuming BOND(pairing)=AUTHENTICATION
and it seems it does
Security Level 3: Authenticated pairing with encryption

and the code in bluedroid clearly says !(sec_flag & GATT_SEC_FLAG_LKEY_UNAUTHED)

if ((perm & GATT_WRITE_AUTH_REQUIRED ) && !(sec_flag & GATT_SEC_FLAG_LKEY_UNAUTHED)) {
status = GATT_INSUF_AUTHENTICATION;
GATT_TRACE_ERROR( "!!!gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION");
}

and GATT_SEC_FLAG_LKEY_UNAUTHED = BTM_SEC_FLAG_LKEY_KNOWN /* 0x10 */

so that if says, if sec_flag does NOT contain the LKEY...
word UNAUTHED makes it a little confusing...

yesh...

mitchjs, 2019-02-11:
oh I had a thought, could your device be bonded with the ESP32 your testing with?
maybe clear all bonded devices
(prob should add that function you posted few days ago to BLEDevice.c)

chegewara, 2019-02-11:
From code above you quoted we can see that write permission can be with ENCRYPTED only connection, with AUTHentication or with ManInTheMiddle(MITM) security:
`GATT_WRITE_AUTH_REQUIRED`
`GATT_WRITE_MITM_REQUIRED`
`GATT_WRITE_ENCRYPTED_PERM`

And those are flags you can pass to setAccessPermissions():
```
#define    ESP_GATT_PERM_READ                  (1 << 0)   /* bit 0 -  0x0001 */    /* relate to BTA_GATT_PERM_READ in bta_gatt_api.h */
#define    ESP_GATT_PERM_READ_ENCRYPTED        (1 << 1)   /* bit 1 -  0x0002 */    /* relate to BTA_GATT_PERM_READ_ENCRYPTED in bta_gatt_api.h */
#define    ESP_GATT_PERM_READ_ENC_MITM         (1 << 2)   /* bit 2 -  0x0004 */    /* relate to BTA_GATT_PERM_READ_ENC_MITM in bta_gatt_api.h */
#define    ESP_GATT_PERM_WRITE                 (1 << 4)   /* bit 4 -  0x0010 */    /* relate to BTA_GATT_PERM_WRITE in bta_gatt_api.h */
#define    ESP_GATT_PERM_WRITE_ENCRYPTED       (1 << 5)   /* bit 5 -  0x0020 */    /* relate to BTA_GATT_PERM_WRITE_ENCRYPTED in bta_gatt_api.h */
#define    ESP_GATT_PERM_WRITE_ENC_MITM        (1 << 6)   /* bit 6 -  0x0040 */    /* relate to BTA_GATT_PERM_WRITE_ENC_MITM in bta_gatt_api.h */
#define    ESP_GATT_PERM_WRITE_SIGNED          (1 << 7)   /* bit 7 -  0x0080 */    /* relate to BTA_GATT_PERM_WRITE_SIGNED in bta_gatt_api.h */
#define    ESP_GATT_PERM_WRITE_SIGNED_MITM     (1 << 8)   /* bit 8 -  0x0100 */    /* relate to BTA_GATT_PERM_WRITE_SIGNED_MITM in bta_gatt_api.h */
```

mitchjs, 2019-02-12:
I think your missing my point/discovery
you cannot write to a characteristic with encryption unless you are AUTHORIZED
you can read with encryption with OUT Authorization

its clear that way in the bluedroid firmware..  the if says has to have the LTK (bonded)
but you said yours works...
so im thinking maybe your test device was bonded to the ESP32
then it would allow it to work...

mitchjs, 2019-02-17:
this is really solved... to get WRITES to work, using "ESP_GATT_PERM_WRITE_ENCRYPTED"
you need to have device bonded too "pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND);"

closed 2019-02-17 by mitchjs