nkolban/esp32-snippets #808: Arduino: how can BLE Client detecting RSSI in Connecting with server,not only for scan?

lukascjh, 2019-02-08:
I connect two esp32s through the BLE server to the Client, when i  use 

int ble_rssi = myDevice->getRSSI();
 Serial.println(ble_rssi); 

in the loop function, become the same value which got  in the Scan 

I want the Client to always test the signal strength of the server. Is this possible?

thank you in advance

chegewara, 2019-02-08:
what is myDevice? BLEClient, BLEServer, BLEAdvertisedDevice?

lukascjh, 2019-02-17:
@chegewara hello, it's BLEAdertisedDevice.

myDevice = new BLEAdvertisedDevice(advertisedDevice);

I want the client to keep detecting RSSI of the server while it is connected, but I failed.
Is this possible?

thank you in advance

chegewara, 2019-02-18:
https://github.com/nkolban/esp32-snippets/blob/810aeb810a15ff6cf8742080ea5a413447cd0b92/cpp_utils/BLEClient.cpp#L366

You need to use it on connected client. 
`myDevice = new BLEAdvertisedDevice(advertisedDevice);` is static structure of data, never updated.

lukascjh, 2019-02-18:
I am using  ble_client   in the example. It uses BLEDevice. [https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/examples/BLE_client/BLE_client.ino](url)
How can this Code be cloned from BLEClient to BLEDevice. I am a beginner and im not very clear how to do it. Can someone help me. thanks

lukascjh, 2019-02-18:
@chegewara Thank you for your reply

lukascjh, 2019-02-18:
i want to try it in loop();

BLEClient*  pClient  = BLEDevice::createClient();
int rssi= pClient->getRssi();
Serial.println("RSSI Value is :" + rssi);

chegewara, 2019-02-20:
Use only this in loop:
```
int rssi= pClient->getRssi();
Serial.println("RSSI Value is :" + rssi);
```

lukascjh, 2019-02-21:
@chegewara  Thank you I was successful
I defined it globally   BLEClient* pClient, and use in loop:
if (connected) {

    int rssi = pClient->getRssi();
    Serial.println(rssi);
}

DeJason-lsng, 2020-01-30:
hey im doing a senior design and im in the same boat i need my rssi to read consistent values. What do you mean you globally defined BLEClient*pClient = BLEDevice::createClient();

jrgodin07, 2022-06-07:
@DeJason-lsng What he meant with globally defined is he declared the object outside of the loop and setup. This enables a global access for every call/use on it.
Ps. Sorry, Im not so good on what they are called.