nkolban/esp32-snippets #809: Not value expected BLECharacteristicCallbacks onWrite getvalue

mpbejo, 2019-02-08:
Hi all, and above all chegewara!
I have this problem:
class Callbacks_Evento: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pChar_Evento) {
      Value_Evento = pChar_Evento->getValue();
      Serial.print("Callbacks_Evento: ");
      Serial.println(Value_Evento.c_str());

With nRFConnect I write on the characteristics but the firs time the Serial.println(Value_Evento.c_str()) is empty and then show me the previous value, instead if read the value from nRFConnect the value is correct.

Here the simple sketch I do not understand if it is a bug or i am missing something:

#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLEServer* pServer = NULL;
BLECharacteristic* pChar_Evento = NULL;                           
BLEAdvertising *pAdvertising = NULL;

std::string Value_Evento;  

int DevConn = 0; 

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      
      DevConn++;
      Serial.print("DevConn: ");
      Serial.println(DevConn);
      
    };

    void onDisconnect(BLEServer* pServer) {
      
      DevConn--;
      Serial.print("DevConn: ");
      Serial.println(DevConn);
    }
};

class Callbacks_Evento: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pChar_Evento) {
      Value_Evento = pChar_Evento->getValue();
      Serial.print("Callbacks_Evento: ");
      Serial.println(Value_Evento.c_str());
      
    }
};

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHAR_UUID_EVENTO    "beb5483e-36e1-4688-b7f5-ea07361b26a8"
void setup() {
  Serial.begin(115200);
// Create the BLE Device
  BLEDevice::init("ESP32");
  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic Evento
  pChar_Evento = pService->createCharacteristic(
                  CHAR_UUID_EVENTO,
                  BLECharacteristic::PROPERTY_READ   |
                  BLECharacteristic::PROPERTY_WRITE);

  pChar_Evento->addDescriptor(new BLE2902());
  
  pChar_Evento->setCallbacks(new Callbacks_Evento());

  // Start the service
  pService->start();

  // Start advertising
  pAdvertising = pServer->getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->start();

}
void loop() {
  // put your main code here, to run repeatedly:

}

Thank You 
Best Regards
Marco

chegewara, 2019-02-08:
Should be fixed by this:
https://github.com/nkolban/esp32-snippets/pull/807

mpbejo, 2019-02-09:
Tested, problem fixed!
Thank You very much.
Nice weekend
Marco

closed 2019-02-09 by chegewara