nkolban/esp32-snippets #810: BLEClient crashes

michaelu123, 2019-02-11:
Since several days I do not get past a problem in BLEClient. I am trying to connect an ESP32 to a Steamcontroller as a HID device. From an Android tool I know that the steam controller offers these services/characteristics:

```
Device Name: SteamController
Device Address: C9:C5:C1:01:B3:DC  // changes every time

Services:--------------------------
GAP (00001800-0000-1000-8000-00805f9b34fb)
Device Name (00002a00-0000-1000-8000-00805f9b34fb)
Appearance (00002a01-0000-1000-8000-00805f9b34fb)
Peripheral Preferred Connection Parameters (00002a04-0000-1000-8000-00805f9b34fb)
GATT (00001801-0000-1000-8000-00805f9b34fb)

Device Information (0000180a-0000-1000-8000-00805f9b34fb)
Manufacturer Name String (00002a29-0000-1000-8000-00805f9b34fb)
PNPID (00002a50-0000-1000-8000-00805f9b34fb)

Unknown service (00001812-0000-1000-8000-00805f9b34fb) 			#define ESP_GATT_UUID_HID_SVC                       0x1812 HID Service
Unknown characteristic (00002a4a-0000-1000-8000-00805f9b34fb)	#define ESP_GATT_UUID_HID_INFORMATION               0x2A4A
							-
Unknown characteristic (00002a4b-0000-1000-8000-00805f9b34fb)	#define ESP_GATT_UUID_HID_REPORT_MAP                0x2A4B
							-
Unknown characteristic (00002a4c-0000-1000-8000-00805f9b34fb)	#define ESP_GATT_UUID_HID_CONTROL_POINT             0x2A4C
Unknown characteristic (00002a4d-0000-1000-8000-00805f9b34fb)	#define ESP_GATT_UUID_HID_REPORT                    0x2A4D
							2902, 2908

Unknown service (100f6c32-1735-4313-b402-38567131e5f3)
Unknown characteristic (100f6c33-1735-4313-b402-38567131e5f3)
Unknown characteristic (100f6c34-1735-4313-b402-38567131e5f3)
```
I am concentrating on service 1812 and characteristic 2a4d, which I configured in BLEClient.ino as serviceUUID and charUUID. In BLERemoteService::retrieveCharacteristics descriptors 2902 and 2908 are found. But either the return from BLERemoteService::getCharacteristic crashes, or the code behind the call in BLEClient.ino. This depends mainly  on the esp_gattc_char_elem_t result variable in  BLERemoteService::retrieveCharacteristics(). When I make it static, the program crashes differently than if I leave it on the stack. It seems that sometimes the stack, sometimes the heap is corrupt. As I do not have the source code of esp_ble_gattc_get_all_char(), I don't know how to debug this further. Something must be wrong around the call of 

```
		esp_gatt_status_t status = ::esp_ble_gattc_get_all_char(
			getClient()->getGattcIf(),
			getClient()->getConnId(),
			m_startHandle,
			m_endHandle,
			&result,
			&count,
			offset
		);

```
count is 8 on return, offset goes from 0 to 7. I noticed once that the charUUID variable was overwritten after a call to esp_ble_gattc_get_all_char(), so that the characteristic was not found.

chegewara, 2019-02-11:
Hi @michaelu123 
in current arduino version of this library we have few bugs. Most of it has been fixed, i hope, in this repository. Could you replace ble library in arduino with this one?
Here is description how to do it:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#replacing-the-version-that-comes-with-arduino-esp32

michaelu123, 2019-02-12:
Hi chegewara, thank you for responding. I am using Sloeber, and I obtained BLE from the current master branch in https://github.com/nkolban/ESP32_BLE_Arduino. I just double checked that this code is identical to mine except for debugging prints I added in the meantime. I wonder if my problems have to do with the fact that the Steamcontroller advertises more services/characteristics than other devices?
I can not add too many debugging prints, though, because then I get an interrupt timeout.
Do you happen to know if there is a malloc library that has more checks for heap corruption, or gcc options that emit code to check stack corruption? Is there a way to obtain the source code for esp_ble_gattc_get_all_char?

chegewara, 2019-02-12:
The issue is in this line:
https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/src/BLERemoteService.cpp#L175

michaelu123, 2019-02-12:
Yes, that is part of it. The "result" value should be an array of count elements. The first call to esp_ble_gattc_get_all_char with offset 0 fills the first n <= count elements of the array. result[0] is the first characteristic. The second call with offset 1 fills the first n-1 elements, and result[0] is the second characteristic, and so on. So one solution is

```
void BLERemoteService::retrieveCharacteristics() {
	ESP_LOGE(LOG_TAG, ">> getCharacteristics() for service: %s", getUUID().toString().c_str());

	removeCharacteristics(); // Forget any previous characteristics.

#define MAX_CHAR 10
	uint16_t offset = 0;
	esp_gattc_char_elem_t result[MAX_CHAR];
	while (true) {
		uint16_t count = MAX_CHAR;  // this value is used as in parameter that allows to search max 10 chars with the same uuid
		esp_gatt_status_t status = ::esp_ble_gattc_get_all_char(
			getClient()->getGattcIf(),
			getClient()->getConnId(),
			m_startHandle,
			m_endHandle,
			result,
			&count,
			offset
		);

...

		ESP_LOGE(LOG_TAG, "Found a characteristic: Handle: %d, UUID: %s", result[0].char_handle, BLEUUID(result[0].uuid).toString().c_str());

		// We now have a new characteristic ... let us add that to our set of known characteristics
		BLERemoteCharacteristic *pNewRemoteCharacteristic = new BLERemoteCharacteristic(
			result[0].char_handle,
			BLEUUID(result[0].uuid),
			result[0].properties,
			this
		);

		m_characteristicMap.insert(std::pair<std::string, BLERemoteCharacteristic*>(pNewRemoteCharacteristic->getUUID().toString(), pNewRemoteCharacteristic));
		m_characteristicMapByHandle.insert(std::pair<uint16_t, BLERemoteCharacteristic*>(result[0].char_handle, pNewRemoteCharacteristic));
		offset++;   // Increment our count of number of descriptors found.
	} // Loop forever (until we break inside the loop).

```

I still have to find out if esp_ble_gattc_get_all_descr has the same behaviour.