nkolban/esp32-snippets #812: Maintainer acts like a child that cannot accept being proven wrong

LaurentLouf, 2019-02-15:
Hi @chegewara ,

I wonder why you keep blocking my issues and PR on this repo and on ESP32_BLE_Arduino ? It really is a shame that being a maintainer of this repo, you cannot accept being proven wrong. Yes, when you copy-paste code from one repository without doing this commit by commit, you loose history. This is not history https://github.com/nkolban/ESP32_BLE_Arduino/commits/master . This is garbage. This is what I would consider inadequate even for a junior dev.

I hope you realize, some day, that being able to say "I'm wrong" is a sign of intelligence, not stupidity. And that code duplication is bad. And that having to copy-paste code between two repos is bad. And that you are unable to justify your choice by any reasonable argument (Kolban used this architecture 5 years ago is not a reasonable argument).

I wish you all the best and will happily continue on my forks, where one commit equals one modification, not 457 modifications to 457 different topics (for your own sake, please google "git commit atomicity"). If you need help learning what a submodule is, I'll also be glad to help.

chegewara, 2019-02-16:
Thanks to @LaurentLouf and @me-no-dev for showing me that i am not the right guy to maintain this repository and to show me what is really important in life. Its not wasting time on discussions with morons who does not understand simple word NO. I can now spend time im wasting here on playing with my autistic 3 years old nephew and having fun.

Good luck to all.

mitchjs, 2019-02-16:
im not sure of any back story between @LaurentLouf  and @me-no-dev …
but I know this without @chegewara the BLE code that @nkolban started years ago
would be no where...  I have followed this repository for while and post after post I see
@chegewara  helping people... and me too

@chegewara  don't let (I guess its 2 people) ruin it for all, I think you enjoy working on this code
as much as helping others... and I know this in in your "free" time... 
I only use the straight ESP-IDF, never bothered with the Arduino fork...
oh and I myself don't fully understand GIT.... I see no reason why you cant commit multiple "fixes/changes" in one commit... 

@chegewara THANK YOU FOR YOUR WORK ON the BLE code from niel's cpp+ work
without it, I prob wouldn't be using a ESP32... as the straight IDF C implementation of BLE
looks so complicated :)

mitch

chegewara, 2019-02-16:
To be straight, @me-no-dev did nothing wrong, just helped me to understand some things.

mitchjs, 2019-02-16:
ok… just that you deserve better than what I read by @LaurentLouf

me-no-dev, 2019-02-16:
Not sure what I helped you understand, given the message, it seems i did not do a good job :D 
It's internet. People arguing is "normal". Do not take things too serious ;) Do not throw away what you have achieved because someone did not like it.

LaurentLouf, 2019-02-16:
Let's play a game called "facts", shall we ? So I opened on the other repo a PR a while ago fixing some bugs, because I'm using the repo arduino-esp32 that includes the other repo as a submodule. I've got a reply saying that no, my PR should go to this repo, because all the code is done here, then copy-pasted on the other repo.

I received one argument for that : the code are different and we don't want to break arduino-esp32 by including changes to the BLE submodule. So first the code was no different at all, except for an include guard. Then, if we don't want to break arduino-esp32, that's absolutely a no-brainer. And even if you're not used to using submodules, just work on branches, that works pretty damn well too.

So I explained at length why having code duplicated between two repos is pretty bad. You can find the reasons here : https://github.com/nkolban/esp32-snippets/issues/764 . Note that discussion has been made impossible by @chegewara that decided that between adults, when we disagree, we don't talk, we ignore each other.

While I received absolutely no response about my arguments, I received one response : the code is like that, I'm not gonna change even if 100 people tell me it's bad. The Creator shall decide, not me. Is this a reasonable argument ? No. Has the Creator of this repo has more or less disappeared for 3 months ? Yes. How convenient !

@mitchjs if you're new to Git and don't get why squashing hundreds of commits into one is bad, then trust people having more experience that tell you it is. Quick example. I use arduino-esp32. I update the BLE submodule to the latest commit because I like to keep things up to date. My BLE code stops working. As an experienced Git user, you use the `git bisect` feature to pin-point what commit did break the functionality. And guess what, it's the commit where hundreds of commits from another repo. Then you're fucked, because that mean that you would have to change your code to include not ble arduino repo, but this one. 

So I'd gladly talk reasonably, just note that I don't accept "it's always been like that" as an argument.

mitchjs, 2019-02-16:
@LaurentLouf is your main issue is that the BLE code is in  "ESP32-Snippets/cpp_utils" and (somewhat) duplicated in another repository "ESP32_BLE_Arduino"

I will guess that @chegewara had nothing to do with that, it all came from @nkolban 
im also guessing that @chegewara started working on the BLE code, because he needed it for a project, and found out that @nkolban didn't have the time to work on it.  as its not originally @chegewara's  work, we have to be thankful he working on it all.

I would agree the "ESP32_BLE_Arduino" should be "linked" to the original code if that's to stay in "ESP32-Snippets/cpp_utils" so one change syncs to both... but I couldn't begin to tell you how to do that in GIT.
the thing is the cpp_utils is much more than the BLE code, yet I would say 90% of the people out there are only using the BLE part :)

really the method that @nkolban created to turn the ESP32-Snippets/cpp_utils BLE Code into the Arduino LIB is prob the fault of all this
myself I would have just done a separate repository for ESP32 BLE cpp code... Arduino prob wouldn't even be a factor (and maybe it wasn't originally for @nkolban )

@LaurentLouf im not sure what your PR on the other repo to fix bugs are... and why he said no or ignored... 

one thing, I never blindly use an updated master commit to my WORKING project...
if anything I will compare every committed change, line by line and see if it makes sense
and I make a quick local copy of the files involved...so if something is broken, I can roll back in a jiffy...

LaurentLouf, 2019-02-16:
It's not somewhat duplicated, it is duplicated, only with some delay. Just add the "with delay" to the list of complaints. Why on Earth should I wait a month for a bugfix made on this repo to magically appear on the BLE arduino repo, when a maintainer on the BLE arduino decides to do a manual copy-paste of the code ? Just "manual" and "copy-paste" should in itself trigger a dozen of red flags to any software developer out there. 

I couldn't care less whose fault it is the repositories are how they are now. I've made mistakes in the past as we all do. The only thing that I'm saying is that no, we don't have to wait for God to weigh in on this issue. And that saying "I won't change it even if 100 people say it's bad" is completely unreasonable. Especially if "God made it like that" is your one and only "argument" if I even may call it that.

If you don't know how to integrate a repo into another one, then it's time to check what a git submodule is. It's used pretty much everywhere. MQTT in esp-idf is included as a submodule for example, same goes for coap, mbedtls, the bt lib etc etc. 

As for your process to update some code on a working project, I'm sorry but I stopped at the copy-paste thing.

mitchjs, 2019-02-17:
well... don't blame @chegewara its not his repository (not yet at least :) )
not everyone knows GIT as well as you do....  
the instructions to build the cpp_utls BLE into Arduino code is given and created by @nkolban 
I guess he created the "ESP32_BLE_Arduino" repository to allow easy of linking to the ESP32 Arduino respository…

it is what it is... it may be wrong, but nothing forces you to use the code...
im grateful @nkolban made this even public, else we would struggle to do BLE using ESP-IDF c code
btw...you got your own fork, fix the bugs as you see fit in it...
and use it in a project... if you see a bug fixed in the cpp_utils, your gonna have to just
copy it into your fork... when @chegewara  feels comfortable with the code, he can put it in the "ESP32_BLE_Arduino" repository... lucky we got him working on this at all!!!

if at some point @chegewara takes ownership of this repository… then maybe it can be reorganized
and have only 1 copy of the BLE code and that would be linked to the ESP32 Arduino repository
I assume you read the post from @nkolban... again like to thank him for his work

LaurentLouf, 2019-02-17:
I don't know how to put it simpler : I don't blame @chegewara for the state of the repo when he began to work on it, I blame him for saying "even if 100 people tell me that the current state is bad, I won't ever change it". 

Whith "it is what it is", you would still be living in a cave with no fire, we would die from the flu, we wouldn't have electricity. I seriously hope you don't consider this as an argument. 

Have I ever said I'm not grateful for the work done ? No. So please stop using this as an argument, it's not one. I'm grateful the work has been done but that doesn't prevent me from thinking that the situation could be greatly improved. 

I don't see how "you can work on your own" is an argument either. I know that. That's not the point of this discussion. Yes, if I see a bug I can fix it on my fork, I can even do that on a private repository. Theoretically, if I discover a major security breach in the code, would you be happy if I can keep on a private repo ? I guess not. Yes I can do an other way. Did I ignore that ? No. Is this a desirable solution having people fixing bugs on their own and not sharing it with the rest of the community ? No. 

You can also discard all of the arguments including something like "copy-paste". That defeats the purpose of Git to copy-paste things around. 

Last, you don't have to be the owner of the repo to change its structure.

mitchjs, 2019-02-17:
@LaurentLouf id like to close this conversation… its not going to get you anywhere..
since you have such strong feelings about this... contact @nkolban and ask him about
taking over the repository. then you can reorganize it and maintain it. myself not good enough C++ programmer and for sure not good enough with GIT to even think about doing any kind of maintaining...
I just didn't want @chegewara publicly abused... and now that I know he didn't even jump into the BLE code because he needed but because he just loved programming the ESP32... have to thank again his work.... and how fast he gets back to people and willingness to test code...

@LaurentLouf  lets just agree to disagree... please close this "issue".

LaurentLouf, 2019-02-17:
So what exactly do you disagree on ? What argument do you refute ? Is code duplication a good thing, no matter what decades of good practices say ? Or is copy-pasting good when you have a tool that prevents this ? Or maybe it's the fact that pull requests are scattered on two different repositories that you find good ? Or the fact the fixes and improvements made here takes between a day or three months to reach the other repo, is that good ?

copercini, 2019-02-26:
@LaurentLouf there are 2 repos because an Arduino library requires a specific structure to work, as you can see [here](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification), and this (main) repo also need to work as an IDF component, so if you have any better idea that works in both cases without duplicate the code I'm curious to hear

MikhailNatalenko, 2019-03-19:
Laurent Louf is right. What the heck? "I don't know git very well, so it's your problem now, don't abuse me!"

Repository must be maintainable! This is the main (and the only i suppose) reason to use GitHub, but now repository is a trash. Stop acting like a little baby, let's fix it together as professionals. We will help you as much as we can. 
And of course Mr Kolban should be responsible, not @chegewara. @nkolban could you solve this argue? We could make straight, clear, obvious repository.

Relationships should be vice-versa: BLE is a submodule that is used in Arduino Repository and in this repository. 
Also it should have it's own Issues PR and everything. You show a very bad practice to novices! >_<

me-no-dev, 2019-03-20:
Situation is that this repository is now unsupported. I have talked to @nkolban and will branch out the Arduino part directly in the Arduino core repo and work on it that way. Not sure what will happen with this one...

MikhailNatalenko, 2019-03-20:
@me-no-dev 

i think it's a good solution.